dresden-elektronik/deconz-rest-plugin #4589: No switch event x003 after long press - Aqara/Opple

klaasnicolaas, 2021-03-16:
## Describe the question or issue you are having

I've noticed that I don't get an x003 event from my 6-way opple switch after a long press of more than 5 seconds. I can see this would have been fixed in the past, but could there be a new bug now? Or possibly this has already been resolved in the current beta?

#3412

## Screenshots

[image]

## Environment

 - Host system: Intel NUC
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.09.03
 - Device: ConBee II
 - Do you use an USB extension cable: no

## deCONZ Logs

## Additional context

Mimiix, 2021-03-16:
To add:

@SAOPP tested this for me on 2.10.2 and he got this response:
[image]

I'll test this later today when i update my setup to the latest beta.

SAOPP, 2021-03-16:
I did not have this problem on the version you specified, now my gateways work on versions 2.10.00, 2.10.01 and 2.10.02, and it is on the latest version (specified by me) that the opple switch works, and it works fine, I just paired it yesterday.

SAOPP, 2021-03-16:
'til yesterday opple worked for me on conbee2 with firmware `264A0700`, today it works on conbee2 with firmware `26580700`.

klaasnicolaas, 2021-03-17:
But do you have this problem if you are running version `2.9.3`? Unfortunately, I cannot easily run a beta version myself to see if it has been solved for me 😢 Unfortunately, I cannot easily run a beta version myself to see if it has been solved for me. But I think the next stable release is coming soon 🤔

SAOPP, 2021-03-18:
> But do you have the problem if you are run version `2.9.3`?

Nope.

SirChickenLittle, 2021-03-20:
Hi, I have this problem on all my OPPLE switches connected via Deconz. I have OPPLE switch 1-gang WXCJKG11LM, OPPLE switch 2-gang WXCJKG12LM and OPPLE switch 3-gang WXCJKG13LM. All of them give me a x003 command back if I hold less then around 5s, but if I hold longer I only get a x001. Have you tried holding longer than 5s and got a x003 command back? I am also on 2.09.03.

SAOPP, 2021-03-22:
Hm guys, I think u 'r mean control of such elements as curtains or blinds... I tested it again, and yes, more than 5 sec is not back x003 event, only x001.

SAOPP, 2021-03-22:
x003 and x001 is back only for 3 sec long press, more than 1 sec and less than 3, seems for strange behaviour if we 'r talking 'bout long press actions... Right now it's just an simple action, like single click and souble click, it's not possible to use it for blinds, if u press it for 5 sec, u didn't get event of release, stop action and etc... Seems an issue.

I didn't pay attention to your messages about the duration of the press, and it seemed to me that you didn't receive long press events at all.

klaasnicolaas, 2021-03-22:
We use this action to dim our lamps in Home Assistant, to my knowledge this has worked in the past.

SAOPP, 2021-03-22:
yes, absolutely any task where u need to press once and hold until the result is achieved

SAOPP, 2021-03-22:
at the moment it is not clear whether this is the device itself or this is our favorite software, it is likely that we need to wait for the developers' response

Mimiix, 2021-03-22:
I tested it on 2.7.0: It doesn't send a event after 5 seconds.

ebaauw, 2021-03-22:
> I've noticed that I don't get an x003 event from my 6-way opple switch after a long press of more than 5 seconds

From what I can tell, this is a "feature" of the Opple.  Check the _Cluster Info_ panel in the GUI, for the Opple's _Multistate Input_ cluster (0x0012) on the endpoint corresponding to the button (check the endpoint number, they might not display in order!).  You see the _Present Value_ (0x0055) changing, as deCONZ receives messages from the switch: 0 for hold, 1 for press/release, 2 for double press, 3 for triple press, 255 for release (after hold).  When holding for more than 5 seconds the switch no longer sends a message, and the value remains 0.

> I can see this would have been fixed in the past

No, it wouldn't.  The fix from the other issue was deCONZ treating the 255 as signed char (i.e. -1) on some platforms, causing the entry in the button map (for 255) not to match on these platforms.

As I mentioned in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3459#issuecomment-713325242, the Opple can be configured as controller or as sensor (but only one at a time).  We opted for sensor when integrating it in deCONZ, because of the long press on the Off and On buttons and the double and triple press on all buttons, at the expense of controlling lights directly.

The configuration is in attribute 0x0009 (aptly named _Opple switch mode_) of the _Lumi_ cluster 0xFCC0.  If you write the value `2` to that attribute, it switches to controller mode.  Wake the Opple by briefly pressing the button on the back.  Write value `1` to revert to sensor mode.  Note that, by default, it controls all lights; you need to bind the _On/Off_ cluster to a group to change that.  Also note that the API plugin will probably reset the mode when deCONZ is restarted.

I've added the commands sent in that mode to the button map, but as you can see (or find out experimenting), in that mode, it only supports `buttonevent` values 1002, 2002, 3001, 3002, 3003, 4001, 4002, 4003, 5001, 5002, 5003, 6001, 6002, 6003.  And guess what: it won't send the _Stop_ command after holding the DimUp, DimDown, TempWarm, or TempCool button for more than 5 seconds either.  I suppose it figures it will have moved all the way after 5 seconds, so a _Stop_ would no longer be needed.

klaasnicolaas, 2021-04-15:
Let's keep this open until the stable version is released 😄

closed 2021-05-14 by github-actions[bot]