dresden-elektronik/deconz-rest-plugin #4590: Moes House Switch Zigbee 3.0 ZTS-EU3-W - TS0601 / _TZE200_tz32mtza

andys73, 2021-03-16:
## Device

- Product name: Switch 3 gang Zigbee 3.0 
- Manufacturer: Moes House TS0601
- Model identifier: _TZE200_tz32mtza
- Device type : 
  - Switch

## Screenshots
Screenshots 
[image]
[image]

It does not pair fully - it can be seen in the VNC connection only.

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2021-03-16:
It s a real end device ? Battery device ? not a fake one ?

andys73, 2021-03-16:
It should be as is main power and require a neutral wire.Thank you

Smanar, 2021-03-17:
Ha yes I remember you, you are from another issue.
So yes this device is a little more complicated because of the Mac adress range, he is not in tuya hack, but If I add your device as it, it can break other device with same Mac adress range.

Need to find a way to reconize your device.

Smanar, 2021-03-17:
I think I will try something as the device already exist as plug.
I will make a code to test tomorow, but if you have time, can you try to re-include the device (without deleting it), I think you will have in log (just "info") some line looking 

>LightNode XXX set node XXXXX

andys73, 2021-03-17:
many thanks for the follow up. Sorry I do not understand what you mean by re-include the device...

andys73, 2021-03-17:
I have enabled the log info and below are the lines related with that MAC address:

21:35:04:610 binding for cluster 0x0006 of 0x60A423FFFE0E120F exists (verified by reporting)
21:35:04:611 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x60A423FFFE0E120F (seems to be active)

does this help?

Smanar, 2021-03-18:
If you can try this version (untested on my side, so can have error on compilation)

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you

```
sudo apt install deconz-dev
git clone --branch test_code_switch https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Log are realy important, with "info" and "info_l2"

LouisHerber, 2021-03-19:
Sorry, don't want to "hijack" this topic but I think the device I mentioned on Discord is fairly similar to this one. It's also a Moes House wall switch, no-neutral version, and only 1-button version. In Deconz it's detected as "Smart Plug" which would be fine. However, it's not found in Phoscon or the API, neither in Home Assistant. I'm using Home Assistant OS / plugin so I cannot use any 10.x / beta version just yet.

Here is the basic cluster info:

[image]

Smanar, 2021-03-19:
@LouisHerber Can you show the Cluster list with the node title and the manufacture number ? Visible on "node info"

@andys73 
>I could not see the cluster info

You mean no cluster list in deconz ?
Can you remake the same test, with the today code ? Take care It will list all your devices at every new device, I hope you haven't too much devices ...

andys73, 2021-03-19:
I will try this evening if I can. I have about 50 devices :)

LouisHerber, 2021-03-19:
> @LouisHerber Can you show the Cluster list with the node title and the manufacture number ? Visible on "node info"
> 

sure
[image]
:

andys73, 2021-03-19:
it did not detected as light but it creates it and I can see in VNC as many devices.. hope you can see it in the extensive logging .
[20210319.txt.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6174113/20210319.txt.gz)

Smanar, 2021-03-20:
@LouisHerber So yes, same issue, you are using same MAC adress range.

@andys73 you are not using docker ?
Because I miss some logs starting by "Tuya debug 44"

Deconz have 2 part, one for sensor, and one for light.
On the light part, I m checking the manufacture number ( = 1002), the device type (smart plug) and 3 mac adress range

and after that I m checking all your devices, and make logs for all of them, but nothing appear on your logs.

andys73, 2021-03-20:
@Smanar I have compiled all on a raspberry pi, no docker, just using the app link added in the programming section.

Smanar, 2021-03-20:
Ok so have added 2 more debug line if you can try the same branch again pls ?

```
cd deconz-rest-plugin
git pull
git checkout test_code_switch 
got pull
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

andys73, 2021-03-20:
[20210320.txt.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6176373/20210320.txt.gz)
fresh pair of logs. I have started the logs as I started the pairing process and then read few bits and bobs in VNC on the device.
You are a hero working all these hours :)

Smanar, 2021-03-20:
Ok, so my code works fine, but is useless ^^.
It check all your devices, but I forget something, your device is not reconized as router, so he don't have entry created at all, I can search where I want, I will never see it.
And he don't have battery, so I can't create a battery sensor, like I m doing for some covering.

I m sorry, but need a new test again

```
cd deconz-rest-plugin
git pull
git checkout test_code_switch 
got pull
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Try again with the deconz trick
- set phoscon in permit join = add new light
- read again basic attribute in deconz, 2 times (need to fill at least the manufacture name) ?

If I m right you will have 1 entry (you will still missing 2 gangs)

andys73, 2021-03-20:
[20210320B.txt.gz](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6176487/20210320B.txt.gz)
Attached. No gang yet sorry

Smanar, 2021-03-21:
I don't understand

>21:50:47:490 Tuya debug 44 : manu find
>21:50:47:492 No Tuya productId entry found for manufacturername: _TZE200_tz32mtza, modelId: _TZE200_tz32mtza
>21:50:47:494 sql exec SELECT * FROM nodes WHERE mac='60:a4:23:ff:fe:d5:d2:c1-01' COLLATE NOCASE AND state != 'deleted'
>21:50:47:500 sql exec SELECT * FROM nodes WHERE mac='0x60a423fffed5d2c1' COLLATE NOCASE AND state != 'deleted'

So there is a moment where the code have the manufacture to check it, but why the rest of code not working ...

andys73, 2021-03-21:
I forgot to have the debug on after the first pairing so I have then added debug, removed all the remaining of device from VNC and pair it again maybe that will explain what you see in the log...

andys73, 2021-04-12:
hello, any action on this? @Smanar

LouisHerber, 2021-04-12:
> hello, any action on this? @Smanar

Where I understand that these devices (e.g. on AliExpress) are multiplying themselves like crazy, new ID's are identified everyday, I realized to be better off by using another Zigbee system. So I went to Zigbee2mqtt which -until now- identified all devices nicely. This might be due to another approach and a wide development community, but to me it worked out. Some time go I had some stability issues and technical approach from z2m but all has been improved since.
This might be different for everyone so I hope you'll find a solution here soon.

andys73, 2021-04-12:
Yes indeed and all are ready to go on Alexa but not on deconz. I can see as well that all of those new devices are advertised as working with zigbee2mqtt. I was trying to stay away from mqtt... 

maybe deconz needs to take a more dynamic approach of devices and stuff.

Mimiix, 2021-04-12:
@LouisHerber @andys73 The biggest difference with Z2M and deCONZ is that deCONZ tries to do a more user friendly approach. Meaning that it is easy to use for the average user and adding devices is very user friendly. If you , however, want to add everything and are technically skilled, Z2M is a better fit.  That being said, the devices not being compliant to the Zigbee spec is making things harder. 

So it is a hard pick on that. 

If you want to have a discussion on this: feel free to open a feature request/ user question issue or join our Discord. But lets keep this one on-topic :)!

andys73, 2021-04-12:
fair enough @Mimiix I agree.
So what do I do with this device...they claim it works out of the box with Alexa (I do not have an alexa or that I want an Alexa to test it out :) ).

Mimiix, 2021-04-12:
@andys73 Who is "They"? 

WE need to wait for a dev to add it. Until then, i can't do anything...
@Smanar What do we need?

andys73, 2021-04-12:
The guys that are selling all those Tuya zigbee 3.0 devices in this case for my device https://www.moeshouse.com/collections/zigbee/products/zigbee-wall-touch-smart-light-switch-with-neutral-wire-no-neutral-wire-no-capacitor-needed-smart-life-tuya-2-3-way-muilti-control-association-hub-required-2-gang-white-%E7%9A%84%E5%89%AF%E6%9C%AC-2

_"ZigBee Wall Touch Smart Light Switch With Neutral Wire/No Neutral Wire,No Capacitor Needed Smart Life/Tuya 2/3 Way Muilti-Control Association Remote Control Works with Alexa,Google Home Hub Required"_

If you look at Moes for example all their Zigbee 3.0 are advertised to work with Alexa so at least I think they have tested with the zigbee hub in Alexa or maybe they are misleading us with Hub Required you need to buy their Tuya hub... and Alexa and Google home connects to Tuya..  That is very possible!

Mimiix, 2021-04-12:
Yep. It is. Because they say "Hub Required".

Smanar, 2021-04-12:
The more usefull for me will be the future "descriptor update".
Without it deconz can't reconize the device, and I can't use same hacks I m using for others.

andys73, 2021-12-21:
any chance I can add this device using the new ddf functionality...

Smanar, 2021-12-22:
I think yes, there is already some tuya switch working with DDF.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

Just need to drag and drop a "on/off light" 
This device don't use report, don't use bind.

Mimiix, 2022-08-24:
Re-opened.

ipmad, 2022-08-24:
Thanks for your support, already installed deconz-dev and was trying to adapt some DDFs with no luck before opening ticket but for sure here is more info and will try with this cheat-sheet.
Thx & regards

Smanar, 2022-08-24:
It s the wired or battery version you have ?

ipmad, 2022-08-25:
Hi Dresden Team, thanks for your support

 

It’s wired version, one curtain, contains electrical switch to manage curtain motor power directly.

 

I can test and help too to develop DDF, I’m in holidays and can spend some time playing with hardware, indeed have more ZB hardware that seems not to be well recognized, like wired PIR sensor and so, probably next days will get more info and let you know. I’m starting a home installation with everything ZB, and have to gateways from you, one USB to test with windows laptop, and one more for raspberry, that will do real work.

 

Thanks a lot and kind regards

 

Iván

Enviado el: miércoles, 24 de agosto de 2022 18:19
Para: dresden-elektronik/deconz-rest-plugin ***@***.***>
CC: ipmad ***@***.***>; Comment ***@***.***>

Smanar, 2022-08-25:
Np, I have take a fast look the device is standard (if we can say that for tuya) so just editing 

```
  "manufacturername": "_TZE200_e3oitdyu",
  "modelid": "TS0601",
```

Can be enought.

closed 2022-09-24 by github-actions[bot]

BoBiene, 2023-01-19:
Hi, i think i have this very same device. The device is discovered correctly in phoscon there came three new lights. 
But if i enable one of them all three gets ON and if i switch on of them OFF all three goes OFF.

[image]
[image]
[image]

BoBiene, 2023-01-19:
@Smanar can you reopen this issue, or should i post a new one?

Smanar, 2023-01-20:
Better to open a new one, you have another model, yours is _TZE3000_pfc7i3kt

But the bad news is your device need the tuya unlock sequence, and this trick is not present at all in c++ code.
So to add it need to make a c++ modification.

BoBiene, 2023-01-20:
> Better to open a new one, you have another model, yours is _TZE3000_pfc7i3kt
> 
> But the bad news is your device need the tuya unlock sequence, and this trick is not present at all in c++ code. So to add it need to make a c++ modification.

Ok, did that: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6682
Is it necessary, that the c++ modification is done in deconz or is it possible to use some tooling for that?

Smanar, 2023-01-20:
>Is it necessary, that the c++ modification is done in deconz or is it possible to use some tooling for that?

Nope the only way is editing the c++ code. and find an user with a linux machine and full OS able to compile the code to test it.