dresden-elektronik/deconz-rest-plugin #4594: Request initial support for Paulmann Gent color remote control

christian-fau, 2021-03-17:
- Product name: Smart Home ZigBee remote control Gent
- Manufacturer: Paulmann Licht GmbH
- Model identifier: RemoteControl
- Device type : 
   - Remote

## Screenshots

Mimiix, 2021-03-17:
Missing screenshots.

closed 2021-03-17 by Mimiix

christian-fau, 2021-03-19:
> "https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4594#event-4471981234",
> "url":
> "https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4594#event-4471981234",
> "name": "View Issue" }, "description": "View this Issue on GitHub",
> "https://github.com" } } ]

Hi Dennis!
My apologize for the screenshots out of clipboard which did not work :-(
I posted a new request with all the required pictures

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4604

Best regards
Christian

--- Original-Nachricht ---
Von: Dennis D
Betreff: Re: [dresden-elektronik/deconz-rest-plugin] Request initial 
support for Paulmann Gent color remote control (#4594)
Datum: 17. März 2021, 19:42
An: dresden-elektronik/deconz-rest-plugin
Cc: christian-fau, Author

Closed #4594.

—

[ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage",