dresden-elektronik/deconz-rest-plugin #4595: Tuya HY368 problem to get an entitie in HASSIO

paule1120, 2021-03-17:
## Describe the question or issue you are having

 Hello,
i have a problem. I cannot get an entitie of my new Tuya HY368 thermostat.
I have one since some times and it works. I bought 3 new one, but i can´t use them in Home Assistant.
I don´t know if it is a hassio or a deconz problem. I hope you can help.
I tried: reconnect, reset the device, delete the node, search new sensor and add the new device, restart the plugin, restart hassio
but nothing happens. Only the name of the new device changed always.

## Screenshots
This one works since some times
[image]

This one is a new one
[image]

## Environment

 - Host system: Raspberry Pi 4 4GB 
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.09.03
 - Device: ConBee II 
 - Do you use an USB extension cable: yes 3m
 - Is there any other USB or serial devices connected to the host system? If so: USB to SATA (SSD)

## deCONZ Logs

## Additional context

Mimiix, 2021-03-17:
It's a new version I guess. @Smanar

Smanar, 2021-03-17:
Lol yes ^^, seriously, like pokemons.

But the device is already in the code, at least a part, you can't try with more recent version ?

paule1120, 2021-03-22:
Hello
@Smanar what you mean with recent version?
Should i update the thermostat? If yes can you tell me how?

Smanar, 2021-03-22:
No, update deconz, if you want to use only stable version the new one will be in some days I think.

paule1120, 2021-03-22:
OK. I will wait for the new stable version and test it again and notice you

paule1120, 2021-03-24:
Hello, I updated today deconz to the new version 6.8.0 and now it works. Thank you

closed 2021-03-24 by paule1120

ruipalhinhas, 2022-12-06:
Hello @Smanar,

Today I received a new HY368 valve but when adding to deconz it seems to be yet again a new model... This time it's __TZE200_cpmgn2cf_
[image]
[image]

Should I create a new issue for this? Let me know please - thanks!

Smanar, 2022-12-07:
For the moment not possible adding new tuya TRV, new device need to be added using DDF, and the PR to support TUYA TRV is still in waiting list.