dresden-elektronik/deconz-rest-plugin #4597: No Calibration possible with Tuya Curtain switch SM-SW101-ZC (TS0302) Device name

Pedder007, 2021-03-17:
## Describe the bug
Hi all, I'm trying to calibrate the shutter device Tuya TS0302, but fail with it.
A while ago the device has been implemented (#2999) in deconz and in general it works by 'open', 'stop' and 'close' propperly.

## Steps to reproduce the behavior
Looking into the 'Window Covering Cluster' there seems to be multiple options to start a calibration:
0x0017 Mode
0xF001 Calibration
But always when trying to use this attributes the gui says 'writing failed' allthough the attributes are shown as 'writable' in the respective attribute editor:

## Expected behavior
The attributes should be usable to set for calibration.

## Screenshots
[image]
[image]
[image]

[image]

## Environment
    Firmware version: (26660700)
    deCONZ version: (2.10.02)
    Device: ConBee II
    Raspbian Buster on PI 4B
    USB extension cable: yes - but anything else works
    No other USB or serial devices connected
    normally using headless and steer/access via HO Pimatic

## deCONZ Logs
not available - don't know how to get

## Additional context

Smanar, 2021-04-09:
Hello, 
Can you try the "read" button, to hcek if the attribute is present or not ?
There a procedure described on the manual ?

I m seing to this device have a special part in code to disable set level command (not supported by the device)

Pedder007, 2021-04-09:
Hi, thanks for your reaction.
Trying to read 0x0017 Mode gives back 'unsupported atrribute', resp. regarding the values in the bottom 'failed UNSUP_GENERAL_COMMAND'
Trying to read 0xF001 Calibration gives back the same answers.

Having them, after restarting deCONZ gui, one time read, no second time is possible.

Smanar, 2021-04-10:
Yes it s normal, it mean the device don't support then, so deconz disable them.
It s for that, i m asking or procedure from the manual, there are using application, special button or other to set it ?

But like i have say previouly, this device seem not support set level command, so why do you want to use a calibration feature, for up and down limit ?

Pedder007, 2021-04-10:
I do not have any docu, having been delivered with it or to be found in the net. I only found these two lines in the respective cluster and was unsure, how far the integration in the rest-api/deCONZ should/would work.
The reason to change the respective value is to extend the roller-time in general, as currently it is too short for my door shutters. The device triggers itself too early back to the stop position.

But meanwhile I got another SM-SW101ZC delivered, which doesn’t have the TS0302, but the TS130F in it. This is really wondering, but from outside you can not find any differences btw. the two devices (exact the same case and printings on it), but it is discovered by Phoscon as TS130F.

At the TS130F I was able to enlarge the roller time. I now ordered another one and looking forward how this will be discovered...

Regarding the adjustment of the TS302: If that isn’t foreseen by deCONZ, I fear I have to live with it, or maybe buy a Tuya Gateway for such special one-time activities. It isn’t that expense any longer and thus I was already thinking about ...

Smanar, 2021-04-10:
And the "clicking-procedure" is not possible on the first one ?

Pedder007, 2021-04-11:
No, that doesn't work. I took this procedure from a small booklet, having been delivered with a LoraTap curtain switch, also discovered as a TS130F.
It says:
- move to 100% close
- set calibration on start
- move to 100% open
- set calibration on stop
After that the roller time was set to a new length. It has been described for the respective app there, but also works via the deCONZ gui, but only for the TS130F type.

But as the different sellers seems buying and re-selling this switches regardless what is in (TS0302/TS130F), I fear a bit, that it will always become like a lottery, if a device might work or not accordingly to what I think it should do ...

Pedder007, 2021-05-03:
HiHo, ok no further ideas out there?
If not I will close the request...

oberschwabe, 2021-05-15:
@Pedder007 : i have the same issue here. my switch is reported as: 
Vendor _TZ3000_ltiqubue
Model TS130F 

discovered as a TS130F.
> It says:
>
> * move to 100% close
> * set calibration on start
> * move to 100% open
> * set calibration on stop
>   After that the roller time was set to a new length. It has been described for the respective [...]

How did you get the curtain 100% down before you started calibration?

Pedder007, 2021-05-15:
@oberschwab Hi, what I explained above was mend in reference to the switch, not the shutter. I clicked on down an waited until the switch toggled itself back to stop. After that I did the adjustment as explained. At that time the switch wasn’t connected to the shutter.
I in general test and adjust the switches before connecting them to the shutters which additionally all do have their own end positions (open and close). The rolling time I configure ~2 seconds longer than the shutter needs.

Pedder007, 2021-05-16:
Hi again, sorry just discovered a mistake in what I wrote above.
I edited/switched: open/close - sorry now it is correct as explained above.

navterro, 2021-06-03:
@oberschwabe it is possible with reverse motor trick, I described this workaround here:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3134#issuecomment-851685394

Unfortunately I'm still unable to calibrate the module.

closed 2021-07-04 by github-actions[bot]