dresden-elektronik/deconz-rest-plugin #4598: Linkind Button Wireless Switch (ZBT-DIMSwitch-D0001)

SirGc, 2021-03-17:
## Describe the question or issue you are having
This wireless switch is a single button switch. Right now, DeCONZ reads it as multiple buttons. It should support a single tap and a tap and hold.

Amazon link: https://www.amazon.com/gp/product/B07V5VLRNL

## Screenshots
https://d.pr/i/nn6xQF
https://d.pr/i/XhujfD
https://d.pr/i/O3ilLM

## Environment
Commercial Version -  2.09.03
Build with Qt 5.11.3
Node limit 200 (max. 200)
Commit b55b2ea7f535bb5d59878c9e6eab75ae10f1a8ef

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: (26xxyy00)
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? NO

## deCONZ Logs
This is for one tap

> [INFO] - Button 2002 - ZBT-DIMSwitch-D0001, broadcast to: 0x400A, endpoint: 0x01, cluster: ONOFF (0x0006), action: Off, payload: None, zclSeq: 77

This is for the next one tap

>[INFO] - Button 1002 - ZBT-DIMSwitch-D0001, broadcast to: 0x400A, endpoint: 0x01, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 98

It alternates between the Button 2002 and 1002 when the only one button is pressed.

This is a tap and hold

>17:15:07:802 [INFO] - Button 1001 - ZBT-DIMSwitch-D0001, broadcast to: 0x400A, endpoint: 0x01, cluster: >LEVEL_CONTROL (0x0008), action: Move up, payload: 007F0000, zclSeq: 114
>17:15:07:948 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
>17:15:08:948 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
>17:15:09:948 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
>17:15:10:623 [INFO] - Button 1003 - ZBT-DIMSwitch-D0001, broadcast to: 0x400A, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), action: Stop, payload: 0000, zclSeq: 115

SirGc, 2021-03-18:
This is an issue previously with this device: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2321

SirGc, 2021-04-10:
still an issue

Mimiix, 2021-04-12:
@ebaauw Can you add and explain why it's a confirmed bug? Also: what would we be the next steps?