dresden-elektronik/deconz-rest-plugin #4599: SQL injection from improperly escaped node names

kyletpugh, 2021-03-18:
## Describe the bug

When a node name includes an apostrophe SQL fails because the strings are not properly escaped.

## Steps to reproduce the behavior

Name a node with an `'` character in the name.

## Expected behavior

This may explain some issues but also can be used to attack the hub.

## Screenshots

[image]

## Environment

Version 2.10.03 / 3/14/2021
Firmware 26660700
v2.10.3-beta
Qt: 5.11.3
Commit: 3c6860
Firmware: 0x26660700

 - Host system: PC
 - Running method: Debian
 - Firmware version: 26660700
 - deCONZ version: 2.10.3-beta
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Smanar, 2021-03-18:
Hello, perhaps same issue than https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1697

kyletpugh, 2021-03-18:
Yes, that appears to be part of the issue but the names will save sometimes and I edited the database myself to correct the names. It happens anytime SQL code is executed, I am pretty sure that there is no escaping of passed strings.

closed 2021-12-18 by manup