dresden-elektronik/deconz-rest-plugin #4600: Philips hue dimmer switch does not generate events

kyletpugh, 2021-03-18:
## Describe the bug

The phillips hue dimmer switch does not generate any events or control any lights after pairing to deCONZ network.

## Steps to reproduce the behavior

Add Philips hue dimmer switch through phoscon. The switch is there but tries to create a group that does not exist after leaving the page and does not control lights. The behavior looks similar to how the Sengled E1E-G7F was not working because it didn't have a button mapping. This is supposedly supported switch but I'm not seeing anything in the log related to it when everything is checked and the buttons are pressed. No event is generated in Home Assistant and I have tried repairing the switch several times.

## Expected behavior

The switch generates events that can be used for automation.

## Screenshots

Not currently applicable.

## Environment

 - Host system: PC
 - Running method: Debian
 - Firmware version: 26660700
 - deCONZ version: 2.10.3-beta
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

There's nothing in the logs related to this device at all.

## Additional context

This switch was linked to some hue bulbs, I switched the bulbs to be part of the network and now the dimmer switch does nothing after pairing with deCONZ.

kyletpugh, 2021-03-18:
Okay, so I'm super frustrated by this. I got the switch sending events but now it keeps creating new groups...

incognit-o, 2021-03-21:
Hey, I'm having the same problem with my Lutron Aurora dimmer (#4620), it does not generate any events that deCONZ could use to automate my lights.

Can you please tell me what you did to get your switch sending events, @kyletpugh?

kyletpugh, 2021-03-21:
I had to keep deleting and repairing it until it was connected. Then it was not fully connected and the name was wrong. I had to edit zll.db to correct the group and sensors,b stopping and starting the deconz service before and after. Then I went to the node opened up the cluster panel selected each cluster and read all the available information individually until it populated. Now, I get events but they are crazy so they are hard to determine something is happening. If you are using HA then you can just select the device as a trigger and select turn on pressed out whatever. Got it working like it originally did cycling through scenes. The dimming works this way as well but I've noticed that the dim amount is based on the light so if you have different lights, they most likely will brighten/dim at different rates. Still not 100% working though....

incognit-o, 2021-03-21:
I see. Can you clarify "deleting and repairing" part little more? Did you mean your deCONZ or deleting/pairing the switch in deCONZ repeatedly until the switch started responding with events? How can you tell if a switch is "fully connected"?

kyletpugh, 2021-03-21:
I just removed and readded it using both the GUI and phoscon. Stopped the service, edited zll.db, restarted the service. Then I used the GUI to read each cluster configuration, when all the clusters are successfully read them you will have a working switch. You can also watch the log with info and debug both levels to see if you're having success but it's quite a lot to sift through.

kyletpugh, 2021-03-21:
I should point out that it doesn't work amazing, the switch has lost some functionality and some is unreliable. Pretty much every switch I have has an issue and I'm noticing that groups are not working correctly in the current version...

kyletpugh, 2021-03-23:
Restart and sometimes have the opposite problem. Can't set both as triggers because sometimes they both work, lol...

closed 2021-04-21 by github-actions[bot]