dresden-elektronik/deconz-rest-plugin #4601: ROBB Smarrt 4-button Transmitter (ROB_200-008-0)

DieterKoblenz, 2021-03-18:
## Device

- Product name: ROB_200-008-0
- Manufacturer: ROBB Smarrt
- Model identifier: ROB_200-008
- Device type : Please remove all unrelated device types. 
  - Remote

## Screenshots

[image]
[image]
[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
Power:
[image]

I hope this is sufficient. Please let me know otherwise.

SwoopX, 2021-03-26:
From what I see here, the device seems to be paired correctly and at least one seneor node got created. Is the device usable?

DieterKoblenz, 2021-03-26:
It's not added in phoscon and/or home assistant

SwoopX, 2021-03-27:
Deconz != Phoscon, so that's expected. Not popping up somehow in HA is an indication it's not fully integrated yet and a look at the code confirms this.

Can you please provide the [INFO] messages on the button presses from the debug output? You can do that from deconz GUI, level INFO and INFO_L2 should be sufficient.

DieterKoblenz, 2021-03-27:
[image]

Button Top Left (I) = On
Button Top Right (O) = Off
Button Bottom Left (I) = On
Button Bottom Right (O) = Off

DieterKoblenz, 2021-04-06:
Is there a way to add a button map manually?

DieterKoblenz, 2021-05-03:
Please, I'd appreciate some input or guidance to help add this device.

DieterKoblenz, 2021-06-11:
Still active issue.

closed 2021-06-13 by manup