dresden-elektronik/deconz-rest-plugin #4603: Device name

stuart-c, 2021-03-18:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: Aurora
- Model identifier: Smart16ARelay51AU
- Device type : 
  - Sensor
  - Switch

## Screenshots

[image]
[image]
[image]
[image]
[image]

### Basic

### Identify
[image]

### Alarms
[image]

### Device Temperature
[image]

### Groups
[image]

### Scenes
[image]

### On/Off
[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

stuart-c, 2021-03-24:
Is there anything else needed?

SwoopX, 2021-03-26:
Nope

closed 2021-04-18 by manup