dresden-elektronik/deconz-rest-plugin #4604: Paulmann RGBW remote control Gent

christian-fau, 2021-03-18:
I appologize for the first request where all the screenshots from the clipboard didn't work :-(

## Device

- Product name: Paulmann ZigBee remote Control RGB GENT
- Manufacturer: Paulmann Licht GmbH
- Model identifier: RemoteControl
- Device type : Please remove all unrelated device types. 
  - Remote
 
## Screenshots

  Required screenshots:
  - Endpoints and clusters of the node
  - Node Info panel
[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]
[image]

### On/Off
[image]
[image]

### Level Control
[image]
[image]

### Color Control
[image]
[image]
[image]
[image]
[image]

### ZZL Commisioning
[image]

### OTAU
[image]

elronzo, 2021-03-24:
Support for this device would be great!

christian-fau, 2021-04-22:
Please reopen this request since it isn't solved yet.
Regards Christian

--- Original-Nachricht ---
Von: github-actions[bot]
Betreff: Re: [dresden-elektronik/deconz-rest-plugin] Paulmann RGBW remote 
control Gent (#4604)
Datum: 22. April 2021, 3:53
An: dresden-elektronik/deconz-rest-plugin
Cc: christian-fau, Author

As there has not been any response in 28 days, this issue will be closed. 
@ OP: If this issue is solved post what fixed it for you. If it is not 
solved, request to get this opened again.

elronzo, 2021-04-22:
+1 for reopen.

christian-fau, 2021-04-22:
+1

--- Original-Nachricht ---
Von: elronzo
Betreff: Re: [dresden-elektronik/deconz-rest-plugin] Paulmann RGBW remote 
control Gent (#4604)
Datum: 22. April 2021, 9:21
An: dresden-elektronik/deconz-rest-plugin
Cc: christian-fau, Author

+1 for reopen.

Mimiix, 2021-04-22:
Please make sure to bump next time.

christian-fau, 2021-05-15:
how to bump and why for a request? (Newbee -question)
Is there anybody who can take or will take an assignment?

Smanar, 2021-05-15:
I think, there is one with free time for this one ATM.
But you can try yourself, this device seem classic, you just need to mimic a similar one and changing some values, like the mac adress or the model id

christian-fau, 2021-05-15:
No idea how :-(
What would be a similar one? What file(s) to adopt?

--- Original-Nachricht ---
Von: Smanar
Betreff: Re: [dresden-elektronik/deconz-rest-plugin] Paulmann RGBW remote 
control Gent (#4604)
Datum: 15. Mai 2021, 19:25
An: dresden-elektronik/deconz-rest-plugin
Cc: christian-fau, Author

I think, there is one with free time for this one ATM.
But you can try yourself, this device seem classic, you just need to mimic 
a similar one and changing some values, like the mac adress or the model id

Smanar, 2021-05-16:
Ok so have started something
You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch paulmann https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry,it can freeze during compilation, my pi 3 do, so remove "-j2"

This code just permit to the device be included in the api, so you will have a ZHAswitch but not working
Now you need to enable log, with "info", press a button look at the message and edit the file /usr/share/deCONZ/devices/button_maps.json according to result.

It s  a simple json file, but you need to restart deconz after a modification.

closed 2021-06-15 by github-actions[bot]

Daeny85, 2021-09-10:
Hello
I would also be interested in integrating that Paulmann Remote.

mat-l, 2022-03-20:
Are the mentioned Paulmann coins working already or not yet?

mat-l, 2022-03-23:
@Smanar I am planning to order some of the devices RGBW and dimmable light from Paulmann. Is there an tutorial or procedure, that I can follow to get them working with Deconz and Home Assistant? I am willing to contribute although I don't know the workflows yet.