dresden-elektronik/deconz-rest-plugin #4610: TS0041, _TZ3000_peszejy7 websocket response

RolandMegens, 2021-03-19:
## Describe the bug

After succesfull pairing Lonsonho Tuya Zigbee Switch (TS0041, _TZ3000_peszejy7) I'm trying to switch multiple lights by domoticz with Deconz plugin. I couldn't find any action or event to trigger the light switch. I almost gave up but finally I think that I've found the solution / problem, connecting with a websocket client to my Conbee II / Deconz I see this response after short, long or double press on the switch:

 ` {"config":{"battery":89,"on":true,"reachable":true},"e":"changed","id":"13","r":"sensors","t":"event","uniqueid":"5c:02:72:ff:fe:ae:71:43-01-0006"}
`

and there is data missing?

`{
    "e": "changed",
    "id": "13",
    "r": "sensors",
    "state": {
        "buttonevent": 2002,
        "lastupdated": "2019-03-15T20:16:30"
    },
    "t": "event",
    "uniqueid": "5c:02:72:ff:fe:ae:71:43-01-0006"
}`
(source: https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/websocket/#sensor-button-event-example)

I think 'state' response is missing / incorrect.

In domoticz with logging of plugin set to all, this is the response after button press:
`2021-03-19 19:07:55.986 (Conbee 2) ### WebSocket Data : {'config': {'battery': 95, 'on': True, 'reachable': True}, 'e': 'changed', 'id': '13', 'r': 'sensors', 't': 'event', 'uniqueid': '5c:02:72:ff:fe:ae:71:43-01-0006'}
2021-03-19 19:07:55.987 (Conbee 2) ### Update device ($Wandschakelaar Washok) : {'BatteryLevel': 95, 'nValue': 0, 'sValue': 'Off'}, IGNORED , no changes !
`

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?
I have:
Raspberry Pi 3 headless
Conbee II with extension cable
firmware: 26660700
version: 2.10.02

## deCONZ Logs

## Additional context

Smanar, 2021-03-20:
Effectively there is a problem, it s a battery powered one ?

Can you share deconz log (with "info") when pressing a button ?

RolandMegens, 2021-03-20:
Yes, it is a battery powered one

Tried to see what happens in log, but nothing happens?

is this the right way to retrieve a log:
sudo systemctl disable deconz
sudo systemctl stop deconz
deCONZ --dbg-info=2 --dbg-http=1 --http-port=81 -platform minimal | tee output.txt

I've attached a output.txt the switch is called "Wandschakelaar Washok" en the ID is 13

Smanar, 2021-03-20:
If you can use the GUI, just use deconz / help / log with "info" and "info_l2"
I don't see the output.txt.

RolandMegens, 2021-03-20:
This is the info:
14:51:03:577 [INFO] - No button map for: TS0041, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: LIDL (0xFD), payload: 00, zclSeq: 66
14:51:04:592 [INFO] - No button map for: TS0041, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: LIDL (0xFD), payload: 00, zclSeq: 66

RolandMegens, 2021-03-20:
I'm going to replace the local button_maps.json with the button_maps.json from github.

RolandMegens, 2021-03-20:
New log info:
15:25:18:901 [INFO] - No button handler for: TS0041, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0001, command: 0x0A, payload: 210020B2, zclSeq: 78
15:25:18:905 ZCL attribute report 0x5C0272FFFEAE7143 for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
15:25:19:207 [INFO] - Button 1002 - TS0041, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B1 short, payload: 00, zclSeq: 79
15:25:20:218 [INFO] - Button 1002 - TS0041, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: B1 short, payload: 00, zclSeq: 79

Smanar, 2021-03-20:
I think it s strange you haven't the last version on deconz 2.10.2 ... But there is so much versions, I can just say, this one is present on the version  v2.10.3-beta
But yes if you are compiling yourself the lib, you need too edit the xml and the json file to have all files updated.

RolandMegens, 2021-03-20:
When beta was on v2.10.2 is I have compiled the lib myself because I knew the switch wasn't supported in the Raspberry Pi OS version (v2.09.3), , when v2.10.3 was released, I didn't do the update.

Which files should be manually be updated? button_maps.json and other files? xml?

Smanar, 2021-03-20:
general.xml too.
But yes you can too use beta file on this URL, you can too switch to beta branch with package command

RolandMegens, 2021-03-20:
THX! Just the info that I needed

closed 2021-03-20 by RolandMegens