dresden-elektronik/deconz-rest-plugin #4615: Cant pair anything on 02.09.03

popy2k14, 2021-03-20:
## Describe the bug
Was on 02.07.01 since end of december and it was running great.
Also have paired a lot devices with it (philips HUE, aqara motion + door sensors, paulmann led driver, ...) without an issue.
Last weekend i updated to 02.09.03 stable and my network was running with all devices,
Yesterday i want to pair "YSR-MINI-01" led stripe driver, Without success!
Also during the pairing attempts, an ikea remote (yeah ikea again) loses its connection ant went offline.
Also this remote i cant repair with 02.09.03!
The YSR-MINI-01 i was able to add to an HUE bridge without an issue (so device is not faulty, that what i was thinink first).
Here is the conversation on that: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4562#issuecomment-802190238

**After manually downgrading deconz to 02.07.01 pairing of the two devices was instant and without an issue!**

## Steps to reproduce the behavior

1. open phoscon
2. goto settings
3.. goto lights/switches/sensors
4. press "pair"
5. initiate the pairing poccess on the device you want to add

with 02.07.01 it works, but not on 02.09.03

## Expected behavior
Should be able to pair also on current stable

## Screenshots

## Environment

 - Host system: Raspberry Pi 4b
 - Running method: Raspbian
 - Firmware version: conbeeII 26680700
 - deCONZ version: 2.07.01 and 02.09.03 (see above)
 - Device: ConBee II
 - Do you use an USB extension cable: yes, but nothing changed, just the deconz version!
 - Is there any other USB or serial devices connected to the host system? yes, Argon One with M2 SSD, but nothing changed.

## deCONZ Logs
Want to provide an debug log but which debug flags to use?
Tried it with:
`deCONZ -platform minimal --auto-connect=1 --http-port=8080 --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1`

But thats way to much debug output.
**Any hints which debug flags are recommended to just get pairing issues logged?**

## Additional context
66 routers
138 end points (sensors)

Smanar, 2021-03-20:
Hello, the more important flag to start are --dbg-info=2 and --dbg-error=2
But it  strange it work with an older version and not a recent ....

popy2k14, 2021-03-20:
Did the upgrade just to get the debug log. Removed the light from deconz, reseted it and tried again in 02.09.03.
This time it paired without an issue!?
Sadly cant reproduce the issue.
Will keep an eye on it.

easybeat, 2021-03-21:
Hi
maybe try to unplug the SSD and try again. Could be related to a USB-Power problem.
Thanks

popy2k14, 2021-03-21:
Today, one hue lamp left the network just after a short power down. And I couldn't reconnect. After some time it worked but not reliable.

Will now go back to an micro sd card and also replace the usb extention cable

Will see how it goes.

Mimiix, 2021-03-22:
Can you also provide the logs :)?

popy2k14, 2021-03-23:
Want to address all the possible hardware issues and if i face the issue again i will post an log.
Currently replaced the SSD with an microsd (which should lower the current needed from the power supply and prevents maybe unexpected voltage drops) and replace the 5m extension calbe (which worked reliable over a year) with an 3m one, which should be sufficient for my needs.

popy2k14, 2021-04-14:
Will get another two stripes these days and try to pair it.
If i have issues i will update this issue with an log.

popy2k14, 2021-05-03:
Pairing works now, don't know what was causing the issue. Closing this issue now.

closed 2021-05-03 by popy2k14