dresden-elektronik/deconz-rest-plugin #4619: Nexentro Zigbee Jalousieaktor Mini (Window Covering Device)

janikmu, 2021-03-20:
## Device

- Product name: Zigbee Jalousieaktor Mini
- Manufacturer: Insta GmbH
- Model identifier: Generic UP Device
- Device type : Please remove all unrelated device types. 
  - Switch
  - Other: Blinds

Nexentro has released multiple Zigbee actors, I bought this one for blinds, which also has input lines for conventional switches:
https://www.insta.de/produktloesungen/nexentro/produktdetail/zigbee_jalousieaktor_mini

[image]

It shows up in deconz after pairing it through the Nexentro Config App (initial config through Bluetooth, then pair with gateway/bridge) and has multiple endpoints, two home automation endpoints and a third I don't recognize. I can execute both lift and tilt value and percentage, though tilt seems to be off, but this can be a calibration matter.

As of now, the REST API has meaningful values for lift and open, but not tilt:
`{"etag":"968a3eef2800909a974a540715683fb9","hascolor":false,"lastannounced":null,"lastseen":"2021-03-20T18:12Z","manufacturername":"Insta GmbH","modelid":"Generic UP Device","name":"Mittleres Rollo","state":{"bri":63,"lift":25,"on":true,"open":true,"reachable":true},"swversion":"00.47.00","type":"Window covering device","uniqueid":"84:2e:14:ff:fe:a1:cd:5f-06"}}`

The first endpoint is the actual device, the second one a controller. The first one has the Window Covering cluster:

## Screenshots
[image]
[image]

### Basic
[image]

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
Window Covering
[image]
[image]
[image]

Smanar, 2021-03-20:
>I can execute both lift and tilt value and percentage, though tilt seems to be off, but this can be a calibration matter.

But on the device I m seing only up and down as wiring, how you can have tilt and lift ?

>Also, I can't see anything about the state of switch input state.

It probably possible, but you can just use the covering return state ? if the covering is open, switch is on "up"

janikmu, 2021-03-20:
Tilt is possible because after doing calibration (either by doing an up and down and blind tilt cycle while pressing the `Prog` button in device, or by manually entering the times in the app - see below) the device knows open/closed tilt limit and thus Tilt Percentage (see Window Covering cluster Attributes). This is why I think it can be easily yielded in the REST API.

[image]

Regarding the switch inputs: I think it depends on the employed switch (sprung - those that need to be pressed continuously - vs fixed - those that remain in position and usually cannot be pressed simultaneously). I can imagine scenarios where it would be good to know what the switch input to the actor is. But this is definitively not a priority. More of a "could".

SwoopX, 2021-03-20:
Well, apparently the deconz code seems to be a little inconvenient in this regard. As for the tilt values, it currently goes like this:

When you pair the device, the window covering type is checked for its current value. As default, it creates only lift for the device. When you change that in deconz GUI, the code should detect that and also expose tilt. However, it appears that there's a chicken and egg situation. When you now have changed the value to also expose tilt, the code is **not called**, as there's a check if a valid HA endpoints exist (which should already be the case) and therefore, a whole passage is skipped. This is especially notable in your case, but should also be a reason for other devices not exposing changed resources (e.g. after a firmware update). I'm not 100% sure if I haven't overlooked anything which compensates this regardless.

What's also notable is that the attribute reports for window covering devices are not processed in the same way as everywhere else. Timestamps do not get updated and the values from the reports do not get send via websocket messages. The window covering cluster is also not polled. so this is kinda problematic for the realistic values of the current position.

As for the switch endpoints you mentioned, there does not seem to be a way to expose them, as they're not there from a zigbee perspective.

SwoopX, 2021-03-20:
Well, apparently the deconz code seems to be a little inconvenient in this regard. As for the tilt values, it currently goes like this:

When you pair the device, the window covering type is checked for its current value. As default, it creates only lift for the device. When you change that in deconz GUI, the code should detect that and also expose tilt. However, it appears that there's a chicken and egg situation. When you now have changed the value to also expose tilt, the code is not called, as there's a check if a valid HA endpoints exist (which should already be the case) and therefore, a whole passage is skipped. This is especially notable in your case, but should also be a reason for other devices not exposing changed resources (e.g. after a firmware update). I'm not 100% sure if I haven't overlooked anything which compensates this regardless.

What's also notable is that the attribute reports for window covering devices are not processed in the same way as everywhere else. Timestamps do not get updated and the values from the reports do not get send via websocket messages. The window covering cluster is also not polled. so this is kinda problematic for the realistic values of the current position.

As for the switch endpoints you mentioned, there does not seem to be a way to expose them, as they're not there from a zigbee perspective.

janikmu, 2021-03-20:
> When you pair the device, the window covering type is checked for its current value. As default, it creates only lift for the device. When you change that in deconz GUI, the code should detect that and also expose tilt.

I don't quite understand this. I had lift attributes exposed without changing anything. 

> The window covering cluster is also not polled. so this is kinda problematic for the realistic values of the current position.

I have witnessed this in Homebridge, where other states other than "opening" and "closing" are not shown (as opposed to the usual percentages).

Smanar, 2021-03-21:
> Tilt is possible because after doing calibration (either by doing an up and down and blind tilt cycle while pressing the Prog button in device, or by manually entering the times in the app - see [...]

I can make the field available in the API, but I realy don't know what will happen.

Do you have a linux machine to make test ?

janikmu, 2021-03-21:
Yes I do, and would happily test

Smanar, 2021-03-21:
You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you

```
sudo apt install deconz-dev
git clone --branch test_tilt https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If I m right, it will enable tilt and lift, so you can make test with it.
You probably need to re  include the devicebut not sure, a restart can be enought.

janikmu, 2021-03-21:
Will do, but shouldnt that
https://github.com/Smanar/deconz-rest-plugin/blob/17b77c4cf75183f67421e49e14a1aa18a86b4972/light_node.cpp#L533

be switched for a more specific check of the Window Covering Type value (Tilt and Lift)? I could imagine other unknown devices could broadcast "Generic UP Device" as well. I also see that VENDOR_INSTA is already implemented but unused.

Smanar, 2021-03-21:
Sure ^^, but it s to test if it work.
But it s rare having 2 time the same model ID, it s the way used by manufacture to reconize their stuff, ( a part Tuya where all devices have the same ...)

SwoopX, 2021-03-22:
@janikmu 
> I don't quite understand this. I had lift attributes exposed without changing anything.

Yes, but I'm talking about tilt. That's what you're missing, right?

@Smanar Your proposed change eliminates the symptoms, but not the root case if I'm right ;)

janikmu, 2021-03-22:
@SwoopX ugh, sorry, I meant to write tilt. Both Lift and tilt values (absolutes and percentages) were in deconz (GUI) without changing anything. This is what you are talking about, right?

janikmu, 2021-03-22:
> You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
> ```https://github.com/Smanar/deconz-rest-plugin.git```
> 
> If I m right, it will enable tilt and lift, so you can make test with it.
> You probably need to re include the devicebut not sure, a restart can be enought.

I've done this. 
Restarted deconz (is the development version using deconz-gui or deconz headless? I tried both), kicked the device out the network and re-joined the network, but still no lift value in the API.

SwoopX, 2021-03-22:
> 
> 
> @SwoopX ugh, sorry, I meant to write tilt. Both Lift and tilt values (absolutes and percentages) were in deconz (GUI) without changing anything. This is what you are talking about, right?

I meant exposure via API, it's clear that the attributes/values are available in deconz GUI if they're supported ;) The feedback you've given so far confirms my theory.

With the modification Smanar proposed, you should only get tilt exposed via API if you delete the device in Phoscon (which basically resets the device) and pair it again.

janikmu, 2021-03-22:
> With the modification Smanar proposed, you should only get tilt exposed via API if you delete the device in Phoscon (which basically resets the device) and pair it again.

See my comment above. No success :(

OT: Is there any motion to refactor the API resources in future major releases such that e.g. window covering devices (that get more popular, especially with the Ikea solutions now) are not listed as `/lights` but rather own resources? I could imagine benefits such as faster, dedicated development of related functionalities?

Smanar, 2021-03-22:
> @Smanar Your proposed change eliminates the symptoms, but not the root case if I'm right ;)

Sure, was a hack to test, but seem not working, 3 solutions
- The model id is unknow at this step
- The attribute WindowCoveringType is not used
- A typo.

Unfortunately I think it s the 2

janikmu, 2021-04-08:
Do you see any path to get this solved @Smanar?

Smanar, 2021-04-08:
To test you can force the device detection
Instead of using 
`if (modelId() == QLatin1String("Generic UP Device"))`

use

`if (true)`

If after that you still not have the field in the json to make test, I think the problem is more on your compilation procedure.

janikmu, 2021-04-11:
Okay, so I looked into it again, and with a hardcoded `hasTilt = true` in line 504 the tilt attribute is exposed to the API which means it is (most likely) number 2:

> The attribute WindowCoveringType is not used correctly (as you said before)

But this is weird to me. Shouldn't any device that has the window covering cluster and is not the `lumi.light.aqcn02` be checked for the WindowCoveringType? The Nexentro device **seemingly** correctly reports that cluster attribute as 8 = Tilt Blind Lift & Tilt (see in the Window Covering Cluster Screenshot from deconz GUI above) and ergo should without any change to the code have `hasTilt = true`...

How can we proceed from here?

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/0cd5d0a328a03af7508c5cfd4ec92e25a5ef9c84/light_node.cpp#L495-L504

janikmu, 2021-05-13:
Not stale, still unresolved

paolanto, 2021-05-18:
Hello, I would only like to add that I also own this device and would be interested in proper support for it in the REST API. I am not practical at all with C++, but please let me know if there is anything I can do to help.

It is really an interesting device and can see it becoming very popular in scenarios where no physical buttons are present (for example when retrofitting electric blinds), since the Bluetooth interface provides a great emergency control solution.

Smanar, 2021-05-18:
Oups, have totally forget this issue.

>Okay, so I looked into it again, and with a hardcoded hasTilt = true in line 504 the tilt attribute is exposed to the API which means it is (most likely) number 2:

So it mean the model id is probably not exposed yet in code else my code would have worked 

```
if (modelId() == QLatin1String("Generic UP Device"))
{
    hasLift = true;
    hasTilt = true;
}
```

>The next thing is, that changes are not reflected. @Smanar said (did you delete that part of your post, I cant find it?) that polling does not (really) work for the Window Covering Cluster - is there a way to make this happen?

Have totaly forget, but this device need to support reporting so it don't need polling, no ?

>In this call the loop looks at each received package individually, but when talking about lift and tilt, it is important to respect the order 1. lift to position then 2. tilt to desired angle at the reached position, because a lift also at the same time implicitly changes the tilt position for shutters. (upwards lift opens the blinds, downwards lift usually closes the blinds).

This is a new problem, but I think it s not deconz problem, on an other device there is no problem, can use the titlt value, even the covering is at 50%
For me better to have a command for lift and another for tilt, and code something on your third app.

janikmu, 2021-06-12:
Happy to know that this is not forgotten 😉 
@SwoopX sure, using the two debug lines provided by @Smanar I guess?
These do not seems to output anything sensible, at least when adding them under https://github.com/dresden-elektronik/deconz-rest-plugin/blob/9a98cbee28519168c2b0339f00dd7f10c47c93c9/light_node.cpp#L527

Logging of adding the device below:
```
15:24:22:146 Debug personnal 1
15:24:22:146 Debug personnal 2 0x0000
15:24:22:147 sql exec SELECT * FROM nodes WHERE mac='84:2e:14:ff:fe:a1:cd:5f-06' COLLATE NOCASE AND state != 'deleted'
15:24:22:148 sql exec SELECT * FROM nodes WHERE mac='0x842e14fffea1cd5f' COLLATE NOCASE AND state != 'deleted'
15:24:22:149 sql exec SELECT * FROM nodes
15:24:22:150 LightNode 4: Window covering device 4 added
15:24:22:151 enqueue event event/added for /lights/4
15:24:22:152 enqueue event config/localtime for /config/
15:24:22:152 add task 433 type 30 to 0x842E14FFFEA1CD5F cluster 0x0004 req.id 25
15:24:22:153 drop request to zombie (rx = 0)
15:24:22:156 Websocket 192.168.178.40:54324 send message: {"e":"added","id":"4","light":{"etag":"d0184dc5319026f5da2a407beaeca2dd","hascolor":false,"lastannounced":null,"lastseen":null,"manufacturername":"Insta","modelid":null,"name":"Window covering device 4","state":{"bri":0,"lift":0,"on":false,"open":false,"reachable":true,"sat":0,"tilt":0},"swversion":null,"type":"Window covering device","uniqueid":"84:2e:14:ff:fe:a1:cd:5f-06"},"r":"lights","t":"event","uniqueid":"84:2e:14:ff:fe:a1:cd:5f-06"} (ret = 443)
15:24:22:483 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
15:24:22:574 poll node 84:2e:14:ff:fe:a1:cd:5f-06
15:24:22:575 Poll light node Window covering device 4
15:24:22:627 read attributes of 0x842E14FFFEA1CD5F cluster: 0x0000: [ 15:24:22:628 0x0005 15:24:22:629 ]
15:24:22:629 add task 435 type 19 to 0x842E14FFFEA1CD5F cluster 0x0000 req.id 29
15:24:22:630 Poll APS request 29 to 0x842E14FFFEA1CD5F cluster: 0x0000
15:24:22:721 Poll APS confirm 29 status: 0xE9
15:24:22:722     drop item attr/swversion
15:24:22:722 0x842E14FFFEA1CD5F error APSDE-DATA.confirm: 0xE9 on task
15:24:22:723 Erase task req-id: 29, type: 19 zcl seqno: 117 send time 0, profileId: 0x0104, clusterId: 0x0000
15:24:23:147 enqueue event config/localtime for /config/
15:24:23:305 poll node 84:2e:14:ff:fe:a1:cd:5f-06
15:24:23:305 Poll light node Window covering device 4
15:24:23:354 read attributes of 0x842E14FFFEA1CD5F cluster: 0x0000: [ 15:24:23:355 0x0005 15:24:23:355 ]
15:24:23:356 add task 439 type 19 to 0x842E14FFFEA1CD5F cluster 0x0000 req.id 33
15:24:23:357 Poll APS request 33 to 0x842E14FFFEA1CD5F cluster: 0x0000
15:24:23:521 Poll APS confirm 33 status: 0xE9
15:24:23:522     drop item attr/swversion
15:24:23:523 0x842E14FFFEA1CD5F error APSDE-DATA.confirm: 0xE9 on task
15:24:23:524 Erase task req-id: 33, type: 19 zcl seqno: 118 send time 0, profileId: 0x0104, clusterId: 0x0000
15:24:24:146 enqueue event config/localtime for /config/
15:24:24:148 poll node 84:2e:14:ff:fe:a1:cd:5f-06
15:24:24:148 Poll light node Window covering device 4
15:24:24:149 Idle timer triggered
15:24:24:199 read attributes of 0x842E14FFFEA1CD5F cluster: 0x0000: [ 15:24:24:200 0x0005 15:24:24:200 ]
15:24:24:201 add task 444 type 19 to 0x842E14FFFEA1CD5F cluster 0x0000 req.id 38
15:24:24:201 Poll APS request 38 to 0x842E14FFFEA1CD5F cluster: 0x0000
15:24:24:322 Poll APS confirm 38 status: 0xE9
15:24:24:324     drop item attr/swversion
15:24:24:326 0x842E14FFFEA1CD5F error APSDE-DATA.confirm: 0xE9 on task
15:24:24:327 Erase task req-id: 38, type: 19 zcl seqno: 119 send time 0, profileId: 0x0104, clusterId: 0x0000
15:24:24:882 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
```

SwoopX, 2021-06-12:
@janikmu I more meant what is described here for attribute 0x0000 of the window covering server cluster
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to...#manually-configure-attribute-reporting

The issue with the current code is that it's in an "act before read" dilemma. The code is correct in that sense, but the read never takes place so the attribute value always remains in its initialization state being 0. Additionally, the passage is only called during initial pairing hence an updated value will never get recognized.

Not too sure yet what the best approach would be to act on a change and what to assume as default.

janikmu, 2021-06-14:
Ah I see @SwoopX. Below is a screenshot. It says `NOT_FOUND` which is "a reportable attribute which has not yet been configured" according to the linked documentation - I guess that is a good sign?

[image]

SwoopX, 2021-06-17:
I guess... So this code is based on the assumption that the attribute is reportable. It dynamically adds and deletes the required resource items based on the set covering type and also configures all attribute reports for the window covering cluster. Re-pairing the device shouldn't be necessary.

`git clone --branch=insta https://github.com/swoopx/deconz-rest-plugin.git`

janikmu, 2021-06-22:
I'm getting errors building your latest changes @SwoopX 

```
bindings.cpp: In member function ‘bool DeRestPluginPrivate::sendConfigureReportingRequest(BindingTask&)’:
bindings.cpp:2083:13: error: ‘rq2’ was not declared in this scope
             rq2.dataType = deCONZ::Zcl8BitEnum;
             ^~~
bindings.cpp:2083:13: note: suggested alternative: ‘rq’
             rq2.dataType = deCONZ::Zcl8BitEnum;
             ^~~
             rq
bindings.cpp:2088:13: error: ‘rq3’ was not declared in this scope
             rq3.dataType = deCONZ::Zcl8BitUint;
             ^~~
bindings.cpp:2088:13: note: suggested alternative: ‘rq’
             rq3.dataType = deCONZ::Zcl8BitUint;
             ^~~
             rq
bindings.cpp:2094:68: error: no matching function for call to ‘DeRestPluginPrivate::sendConfigureReportingRequest(BindingTask&, <brace-enclosed initializer list>)’
             return sendConfigureReportingRequest(bt, {rq, rq2, rq3});
                                                                    ^
bindings.cpp:679:6: note: candidate: ‘bool DeRestPluginPrivate::sendConfigureReportingRequest(BindingTask&, const std::vector<ConfigureReportingRequest>&)’
 bool DeRestPluginPrivate::sendConfigureReportingRequest(BindingTask &bt, const std::vector<ConfigureReportingRequest> &requests)
      ^~~~~~~~~~~~~~~~~~~
bindings.cpp:679:6: note:   no known conversion for argument 2 from ‘<brace-enclosed initializer list>’ to ‘const std::vector<ConfigureReportingRequest>&’
bindings.cpp:836:6: note: candidate: ‘bool DeRestPluginPrivate::sendConfigureReportingRequest(BindingTask&)’
 bool DeRestPluginPrivate::sendConfigureReportingRequest(BindingTask &bt)
      ^~~~~~~~~~~~~~~~~~~
bindings.cpp:836:6: note:   candidate expects 1 argument, 2 provided
```

SwoopX, 2021-06-22:
@janikmu fixed

janikmu, 2021-06-22:
...leading to even more / new errors during make:

```
de_web_plugin.cpp: In constructor ‘ApiRequest::ApiRequest(const QHttpRequestHeader&, const QStringList&, QTcpSocket*, const QString&)’:
de_web_plugin.cpp:565:45: error: invalid initialization of reference of type ‘const QLatin1String&’ from expression of type ‘const QString’
         version = getAcceptHeaderApiVersion(accept);
                                             ^~~~~~
de_web_plugin.cpp:527:19: note: in passing argument 1 of ‘ApiVersion getAcceptHeaderApiVersion(const QLatin1String&)’
 static ApiVersion getAcceptHeaderApiVersion(const QLatin1String &hdrValue)
                   ^~~~~~~~~~~~~~~~~~~~~~~~~
de_web_plugin.cpp: In member function ‘void DeRestPluginPrivate::addLightNode(const deCONZ::Node*)’:
de_web_plugin.cpp:2103:34: warning: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘unsigned int’} and ‘int’ [-Wsign-compare]
             for (size_t c = 0; c < sd->inClusters().size(); c++)
                                ~~^~~~~~~~~~~~~~~~~~~~~~~~~
de_web_plugin.cpp:2165:30: warning: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘unsigned int’} and ‘int’ [-Wsign-compare]
         for (size_t c = 0; c < i->inClusters().size(); c++)
                            ~~^~~~~~~~~~~~~~~~~~~~~~~~
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++1y -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DSKIP_EMPTY_PARTS=QString::SkipEmptyParts -DUSE_WEBSOCKETS -DHAS_SQLITE3 -DHAS_OPENSSL -DGW_SW_VERSION=\"2.12.01\" -DGW_SW_DATE=1624400674 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"37225e95a65298fbb5b26445f9a1b405d9ec8f0e\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -Irelease -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o release/de_otau.o de_otau.cpp
de_web_plugin.cpp: In member function ‘virtual bool DeRestPlugin::isHttpTarget(const QHttpRequestHeader&)’:
de_web_plugin.cpp:17561:13: error: ‘const class QHttpRequestHeader’ has no member named ‘pathAt’; did you mean ‘path’?
     if (hdr.pathAt(0) == QLatin1String("api"))
             ^~~~~~
             path
de_web_plugin.cpp:17565:18: error: ‘const class QHttpRequestHeader’ has no member named ‘pathAt’; did you mean ‘path’?
     else if (hdr.pathAt(0) == QLatin1String("description.xml"))
                  ^~~~~~
                  path
In file included from /usr/include/arm-linux-gnueabihf/qt5/QtCore/qcoreapplication.h:43,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/QCoreApplication:1,
                 from de_web_plugin.cpp:11:
de_web_plugin.cpp: In member function ‘virtual int DeRestPlugin::handleHttpRequest(const QHttpRequestHeader&, QTcpSocket*)’:
de_web_plugin.cpp:17592:81: error: no matching function for call to ‘asString(const QUrl&)’
         DBG_Printf(DBG_HTTP, "HTTP API %s %s - %s\n", qPrintable(hdr.method()), qPrintable(hdr.url()), qPrintable(sock->peerAddress().toString()));
                                                                                 ^~~~~~~~~~
In file included from /usr/include/arm-linux-gnueabihf/qt5/QtCore/qcoreapplication.h:44,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/QCoreApplication:1,
                 from de_web_plugin.cpp:11:
/usr/include/arm-linux-gnueabihf/qt5/QtCore/qstring.h:1862:23: note: candidate: ‘const QString& QtPrivate::asString(const QString&)’
 inline const QString &asString(const QString &s)    { return s; }
                       ^~~~~~~~
/usr/include/arm-linux-gnueabihf/qt5/QtCore/qstring.h:1862:23: note:   no known conversion for argument 1 from ‘const QUrl’ to ‘const QString&’
/usr/include/arm-linux-gnueabihf/qt5/QtCore/qstring.h:1863:18: note: candidate: ‘QString&& QtPrivate::asString(QString&&)’
 inline QString &&asString(QString &&s)              { return std::move(s); }
                  ^~~~~~~~
/usr/include/arm-linux-gnueabihf/qt5/QtCore/qstring.h:1863:18: note:   no known conversion for argument 1 from ‘const QUrl’ to ‘QString&&’
de_web_plugin.cpp:17595:13: error: ‘const class QHttpRequestHeader’ has no member named ‘httpMethod’; did you mean ‘method’?
     if (hdr.httpMethod() == HttpPost && hdr.hasKey(QLatin1String("Content-Type")) &&
             ^~~~~~~~~~
             method
de_web_plugin.cpp:17595:29: error: ‘HttpPost’ was not declared in this scope
     if (hdr.httpMethod() == HttpPost && hdr.hasKey(QLatin1String("Content-Type")) &&
                             ^~~~~~~~
de_web_plugin.cpp:17596:26: error: could not convert ‘(& hdr)->QHttpRequestHeader::value(QString(QLatin1String(((const char*)"Content-Type"))))’ from ‘const QString’ to ‘QLatin1String’
        contains(hdr.value(QLatin1String("Content-Type")), QLatin1String("multipart/form-data")))
[... cut ...]
de_web_plugin.cpp:17752:53: error: ‘const class QHttpRequestHeader’ has no member named ‘pathAt’; did you mean ‘path’?
                 const QLatin1String apiModule = hdr.pathAt(2);
                                                     ^~~~~~
                                                     path
de_web_plugin.cpp:17836:46: error: ‘const class QHttpRequestHeader’ has no member named ‘httpMethod’; did you mean ‘method’?
                 if (resourceExist && req.hdr.httpMethod() == HttpGet)
                                              ^~~~~~~~~~
                                              method
```

SwoopX, 2021-06-22:
But this time they're not mine 😉 

Can you try to remove the package `deconz-dev` and then reinstall it? Please make sure you use the beta branch for installing it. You can switch back to stable afterwards.

janikmu, 2021-06-23:
Weird, I re-installed `deconz-dev` (which was indeed outdated) and rebuild your changes, but one error persists:

```
firmware_update.cpp: In member function ‘void DeRestPluginPrivate::checkFirmwareDevices()’:
firmware_update.cpp:548:73: error: ‘instance’ is not a member of ‘deCONZ::DeviceEnumerator’
     deCONZ::DeviceEnumerator *devEnumerator = deCONZ::DeviceEnumerator::instance();
                                                                         ^~~~~~~~
make[1]: *** [Makefile.Release:952: release/firmware_update.o] Error 1
```

But this is no file you ever touched in your changes...

SwoopX, 2021-06-23:
Yeah, I guess everybody has those errors... Just ignore them. How's the device doing?

janikmu, 2021-06-23:
Can't ignore them, I can't finish the `make` process to build the plugin...

SwoopX, 2021-06-23:
Sure, but please make sure you upload a file 😉 

Does it just not get past the device selection or does it actually crash with a core dump?

janikmu, 2021-06-25:
Actually, the startup log from `/usr/bin/deCONZ --platform minimal --dbg-info=2` is quite short:

```
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
Segmentation fault
```

The startup log from `systemctl start deconz` followed by `journalctl -u deconz` is longer, but only repeats itself, but the important bit is this:

```
Jun 25 12:46:09 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Jun 25 12:46:09 raspberrypi deCONZ[22827]: QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-pi'
Jun 25 12:46:09 raspberrypi deCONZ[22827]: libpng warning: iCCP: known incorrect sRGB profile
Jun 25 12:46:10 raspberrypi systemd[1]: deconz.service: Main process exited, code=killed, status=11/SEGV
Jun 25 12:46:10 raspberrypi systemd[1]: deconz.service: Failed with result 'signal'.
```

After this it only schedules restarts that then fail with the same exact message.

SwoopX, 2021-06-27:
Hm, that's not good. Can you please start deconz as follows:

`gdb --args /usr/bin/deCONZ --platform minimal --dbg-info=2`

1-2 seconds afterwards, you shourld get the gdb promt where you simply type `r` followed by pressing return. When the application crashes, you should get the promt again, then please type `bt` and paste the output here, please.

janikmu, 2021-06-27:
There you go:

```
#0  0x76ed0744 in deCONZ::SimpleDescriptor::inClusters() const () from /usr/lib/libdeCONZ.so.1
#1  0x72566b48 in DeRestPluginPrivate::addLightNode(deCONZ::Node const*) ()
   from /usr/share/deCONZ/plugins/libde_rest_plugin.so
#2  0x7258e928 in DeRestPluginPrivate::nodeEvent(deCONZ::NodeEvent const&) ()
   from /usr/share/deCONZ/plugins/libde_rest_plugin.so
#3  0x726ac1e0 in DeRestPluginPrivate::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) ()
   from /usr/share/deCONZ/plugins/libde_rest_plugin.so
#4  0x75ff1244 in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5
#5  0x76ed96e0 in deCONZ::ApsController::nodeEvent(deCONZ::NodeEvent const&) () from /usr/lib/libdeCONZ.so.1
#6  0x0003a6b8 in ?? ()
#7  0x000491bc in ?? ()
#8  0x000353c0 in ?? ()
#9  0x759ee718 in __libc_start_main (main=0x7e9ce614, argc=1974603776, argv=0x759ee718 <__libc_start_main+268>,
    init=<optimized out>, fini=0xc6a10, rtld_fini=0x76f314c4 <_dl_fini>, stack_end=0x7e9ce614) at libc-start.c:308
#10 0x00035c74 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

janikmu, 2021-06-27:
@SwoopX maybe it was all just a hick-up.

I re-installed everything (including dependencies), freed up some space (rookie error, SD card of my Pi was at 100%) and I don't get any segmentation fault anymore. However, the `tilt` attribute is still not there (the only thing I did not do is re-pair the device).

Log is attached below
[deconz-insta-dbg.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6721988/deconz-insta-dbg.txt)

SwoopX, 2021-06-27:
Hm, the output looks definitively odd. There's indications your network configuration is corrupt/not working.

```
[...]
16:58:57:150 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
16:59:07:149 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
16:59:17:148 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
16:59:27:148 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration
```

There's no binding attempts, not reporting, no nothing in there I'm afraid. Normally, based on the changes, a read of attribute 0x0000 or an attribute report should trigger the changes on the resource item.

janikmu, 2021-06-30:
I will reset and repair everything. Is there a best practice for doing so with the Raspbee? It's the only device in my testing instance so I don't need to retain anything.

SwoopX, 2021-07-09:
Ah, sorry. Missed it again.

Not really. Under Linux, there's a folder `dresden-elektronik` under `.share` in the home directory of the user running deconz, which almost holds everything (except for the network settings kept on the stick). In deconz GUI, you could probably press the pane to leave the network, then open the network settings and set the PAN ID yourself to all 0 instead of D.

paolanto, 2021-07-17:
Hi SwoopX, thank you for your effort so far.

Your latest changes are contained in version 12.12.3-beta, correct? I have upgraded to that version, paired the two blind actuators again and the tilt attribute is still not exposed over the API. If you have time to get back to this issue, then please let me know if there is any log or information I can provide to help you. I am very grateful for your support.

SwoopX, 2021-07-17:
No worries. No, the changes are not yet release and hence only available in my repo since I do not know if it works or breaks anything. That's why you'd need to compile the custom plugin once more.

I'll quickly update the branch to the latest changes...

paolanto, 2021-07-17:
Ok, I see. Unfortunately the steps to compile the plugin locally seem to be for Linux only and I don't have any Linux environment available for this. I normally run deconz using marthoc's docker container a NAS as well as on Windows for trying out stuff. I am afraid I am only able to try it out and provide feedback if anybody builds and shares binaries compatible with either of these two environments (perhaps the binaries within the docker container can be replaced without rebuilding the container image). Of course I remain available for anything it could be useful to check out or trying out with this device.

paolanto, 2021-08-13:
Not stale, issue still unresolved. @janikmu: any chance you could find the time to test SwoopX changes? I would be very grateful. If not, I can try it myself and confirm, however I would need to setup a Linux system just for doing this.

janikmu, 2021-08-23:
Sorry both of you, I had zero time to answer.
I have tried repairing, re-installing, everything, still no tilt attribute in the API.
However, my weird network configuration seemingly is not fixed, at least the lines remain that state:

`apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration`

Full log: [test_debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7032359/test_debug.txt)

For what it is worth, I have switched to zigbee2mqtt, which works flawlessly with that device. I am sorry, but I will not continue to test around with deconz.

janikmu, 2021-10-21:
@SwoopX that actually sounds like a good route to go, instead of Device Requests piling up here!

fkuep, 2021-10-21:
I partially understood what You two meant and will try find You on discord.
Notice that a guideline sent me to open a [device request in the forum](https://forum.phoscon.de/t/insta-nexentro-blinds-actuator-mini/687) 
and instead of "~poluting You with device requests" I was accompanied by
Your community coordinator @Mimiix  who reopened this ticket.

Anyways, as mentioned there I want to seek help and make it easy to help me.
So once I found You discord , I will ask again:
* what do I do ?
* do we close this ticket again ?
* do we remove the forum entry?

closed 2021-11-22 by github-actions[bot]

nook24, 2022-05-01:
Did you guys managed to connect this thing? It is connecting, but I can only stop and move down my windows cover. The up key isn't doing anything.
[image]

- Phoscon Version: 2.15.03 / 21.4.2022
- deCONZ: 2.15.3
- Jalousieaktor Firmware: 00.00.49.01

Smanar, 2022-05-01:
Hello, what is the model id of the device ?

cremor, 2022-05-01:
@nook24 Looks like the device is not reporting the cover position correctly. Try using the hardware buttons to first fully close the cover, then fully open the cover again. This should "initialize" the device, after which it should report the position correctly again.

This "initialization" is needed every time after you change any device setting in the Nexentro config app.

nook24, 2022-05-01:
@Smanar I'm not sure, where do I find the _model id_? On the device itself it says: `57008000 V00` like in this picture: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4619#issue-836882742
[image]

@cremor I will give this a shot, I haven't tasted this so far. At the moment this thing is laying on my bench just connected to a light bulb for testing. Is the device measuring the current draw of the connected motor to detect when the motor is in position?

cremor, 2022-05-01:
I don't know, but I don't think so. Since you have to configure the time it takes to fully close the cover, I think the cover position is just inferred from the movement time. Same for the tilt position.

Smanar, 2022-05-02:
> Manufacturer: Insta GmbH
> Model identifier: Generic UP Device

It's fine was  just to be sure, you have exactly the same device, and except the MAC adress range, all looks fine.

Try with the calibraion procedure, but it's strange you have a different working mode than other.

nook24, 2022-05-02:
I'm not sure if this is the right product for me. My currently wall mounted switch (not push button) is always passing current to the motor. The motor itself knows when to stop. So when this device is not doing any current measuring, i don't know how it would be doing any calibration procedure at all. How ever, I will test this with an incandescent light bulb today as my "motor", before i start ripping apart my wiring :)

fkuep, 2022-05-02:
> I'm not sure if this is the right product for me. My currently wall mounted switch (not push button) is always passing current to the motor. The motor itself knows when to stop. So when this device [...]

I guess most of us assume the hardware is correct. I will now explain why and what Your next steps should be. After that is done it either works or You can detail to us better where it goes wrong:  

* The motor electronics are reponsible that it does not run hot after the end is reached 
  * german: endschalter
  * Therefore it is ok to run those with switches 
* the actuator You bought can not know , out of the factory, how "long" the bilnds go, since the motor does not know
  * But it has functionality to "set the blinds to 60% " if calibrated.
* The actuators manual control inputs have to be push buttons (Taster)
 * See switches (schalter) have state ("I am in all closed position") which would most often conflict with what You set over the computer..

What You Do is:  
* since it only lets You go down, You move it down until it shuts off.
* Then You use the bluetooth calibration tool to set the time it takes 100% to 0%.
  * You can guess the first time.
* after that the up controll should appear.  

Mine looks like this:  
[image]  

* To simulate buttons You can use Your switches , but always put them back "into neutral"
  * If all works as expected You would by buttons. 

 Does this help ?

cremor, 2022-05-02:
> * The actuators manual control inputs have to be push buttons (Taster)
> 
> * See switches (schalter) have state ("I am in all closed position") which would most often conflict with what You set over the computer..

It also works with switches instead of push buttons. I'm personally using it with switches. Right now the controls are a bit weird with switches, but according to the vendor official switch support will come with a software update "soon" (I was told mid April, but since this has already passed, I don't know how soon it will be).

Right now the behaviour with switches is:
- Switch the up/down switch to "on" for a short time (less than about 1 sec): Only move the cover for as long as the switch was on "on" state.
- Switch the up/down switch to "on" for a longer time: Completely open or close the cover.
- Switch the up/down switch to "on" for a short time, while the cover is already moving: Stop the cover

nook24, 2022-05-02:
Many thanks @fkuep and @cremor it is working now.
As you have mentioned it was required, to setup a time (I took 20 seconds for faster testing) and than "push a button" connected to the actor first (`E1` and `E2`). 
After playing around with the hardware buttons I was able to open, close and stop the window cover via Home Assistant.
 
I'm still on my bench, a load is not required for the initial setup process.

From my tests I definitely can confirm that you should go with push buttons (german Taster) not switches (german Schalter)