dresden-elektronik/deconz-rest-plugin #4620: Lutron Aurora pairs, detected as a switch, but controls doesn't do anything.

incognit-o, 2021-03-21:
## Describe the question or issue you are having

My Lutron Aurora (Z3-1BRL / Friends of Hue Dimmer) successfully pairs and is recognized as a switch in deCONZ. In addition to this, I have 2 RGB LED strip & 2 Sengled RGB LED in a group that works fine with strong connections with my ConBee. They do respond to my controls via web page reliably. I tried adding my Lutron dimmer in the group and mapping controls to it in the switch editor. However, rotating the dimmer dial and the button do nothing.

## Screenshots

[image]

[image]

[image]

[image]

[image]

[image]

## Environment

 - Host system: Raspberry Pi 4 Model B 4GB
 - Running method: Marthoc Docker container on Raspbian (deployed with docker-compose)
 - Firmware version: 26580700
 - deCONZ version: 2.10.03
 - Device: ConBee II
 - Do you use an USB extension cable: No
 - Is there any other USB or serial devices connected to the host system: None

## deCONZ Logs

I'm not sure which logs to include here. I tried checking out the output with APS, ZDP, ZCL, and INFO and did not see anything potentially related to this issue although I can be wrong. Please let me know if there's anything else I should include.

## Additional context

I know the dimmer switch works and is not defective because the dimmer dial and the button work fine with ZHA. See below for events via ZHA:
[image]
When I have everything paired with deCONZ and listening on deconz_event via Developer's Tools on Home Assistant always turns up empty.

Please let me know if I need to include anything more here. I'd like to get this dimmer switch working with deCONZ!

incognit-o, 2021-03-21:
I just updated my ConBee II from firmware 26580700 to 26680700 but I'm still having the same problem.

[image]

[image]

incognit-o, 2021-03-21:
I tried running deCONZ bare metal on my Windows laptop with ConBee II plugged in that instead of in Docker container on Raspberry Pi. I'm still getting the same problem. Pairing works fine, deCONZ recognizes Lutron Aurora correctly, but controls still don't do anything.

Mimiix, 2021-03-22:
Do you get any events from it in HA?>

incognit-o, 2021-03-22:
@Mimiix I tried listening for 'deconz_event' on HA. Nothing shows up when I try pressing the button or rotating the dial.

Mimiix, 2021-03-22:
Wait, is this a dimmer without power?

incognit-o, 2021-03-22:
Yes, it's a battery operated dimmer if that's what you mean. It goes over a pre-existing light switch like so:
[image]

Mimiix, 2021-03-22:
No but, is it a zgp switch? 

Either way, can you get logs on aps 2, info 2 and error 2?

ebaauw, 2021-03-23:
See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2305.

incognit-o, 2021-03-23:
@ebaauw Thanks, I just read that issue. You mention this in the post:
> The switch is supported by the REST API plugin.

Are you suggesting that it doesn't work in Phoscon, only REST API?

Now, I'm not familiar with deCONZ but I'm assuming HA uses REST API for listening on `deconz_event`? If the switch is indeed working via REST API, button events should show up there, right? If so, this is not happening for some reason...

Should I try connecting to deCONZ via WebSockets instead? Would it work there? 

I'm not really exactly sure what to do here... I guess I'd have to watch logs with `--dbg-info=2 and --dbg-error=1` while pairing with Aurora?

ebaauw, 2021-03-23:
I don't know what changed, but indeed, my Lutron Aurora no longer seems to control anything after pairing to v2.10.3-beta.  Furthermore, the resource now looks like:
```
{
  "config": {
    "battery": 100,
    "group": "162",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "ad19686e65ba0c00aa9cffef8b4bc29a",
  "lastseen": "2021-03-23T21:02Z",
  "manufacturername": "Lutron",
  "mode": 1,
  "modelid": "Z3-1BRL",
  "name": "Z3-1BRL 19",
  "state": {
    "buttonevent": null,
    "eventduration": null,
    "lastupdated": "none"
  },
  "swversion": "3.8",
  "type": "ZHASwitch",
  "uniqueid": "ff:ff:00:0f:e7:fd:e9:08-01-fc00"
}
```

ebaauw, 2021-03-23:
What firmware is your dimmer on?  3.8?.  Any-one still on 3.4 having issues as well?

SwoopX, 2021-03-23:
3.4 seems to work. I recall that from Discord.

incognit-o, 2021-03-24:
My dimmer is on 3.8.
[image]

incognit-o, 2021-03-24:
I think this is what @SwoopX was referring to.
[image]

NBradke, 2021-03-25:
I have now tested it in the last beta version (2.10.04). Here, too, I can integrate the switch (3.8), but not configure it.
[image]

The Lutron Aurora (3.8) is recognized. But it cannot be configured.
[image]

NBradke, 2021-03-25:
In version 3.4, however, it works without any problems.
[image]

ebaauw, 2021-03-25:
So it looks like something changed in the Aurora firmware, and the issues are not dependent on the deCONZ version.  I need again to sniff the pairing of the Aurora with the Hue bridge, to see if they do something different now for the new firmware.  No high hopes, though; never managed to get the Aurora to act with deCONZ as it does with the Hue bridge.

I’m afraid we don’t have a link to the 3.4 firmware, and if we do, we do, we still don’t know whether the Aurora allows OTA downgrading the firmware.  Still, could you post a screenshot of the. _OTAU_ cluster for the 3.4 firmware, making sure to have read each attribute?  Maybe we can deduce the firmware location.

ebaauw, 2021-03-31:
Here's the log of pairing the Aurora running 3.8 with the Hue bridge:
[Pair_Lutron_Aurora_3_8_ to_Hue_Bridge.pcapng.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6237981/Pair_Lutron_Aurora_3_8_.to_Hue_Bridge.pcapng.zip)

ebaauw, 2021-04-05:
I found the 3.4 firmware for the Aurora, see https://github.com/dresden-elektronik/deconz-ota-plugin.  I looks like the controller is happy to downgrade the firmware, if you select the file manually.  Currently installing; 2% in 2:20.

NBradke, 2021-04-07:
I found time yesterday to deal with it. The downgrade to version 3.4 took about 1 hour.
Then I reconnected the two switches and was finally able to configure them. Thank you very, very, very much!!!
Tip: Note version 3.8 as incompatible, otherwise others will be in trouble.

Mimiix, 2021-04-07:
> Tip: Note version 3.8 as incompatible, otherwise others will be in trouble.

Feel free to edit the wiki page yourself ;)

SwoopX, 2021-06-11:
Closing this one as the solution is obviously to downgrade the firmware on the device 👍

closed 2021-06-11 by SwoopX

barryl93, 2022-01-25:
Sorry to reopen this, but this is the only place I've found that comes close to resolving my issue. I, too, have an Aurora with the 3.8 firmware. I understand I need to downgrade the firmware, but I can't seem to find instructions on how to do this. Can anyone point me in the right direction? Thanks!

jorugit, 2022-11-07:
Thats a very bad solution! Downgrade to an older Version...

pnewell, 2022-12-31:
I am attempting to downgrade the aurora to 3.4 but when I select a file in OTA Update and hit the Update button it just stays on Queued indefinitely and never actually updates. I have turned off the only other device on the network (a lightbulb). Any help would be appreciated.

Source Routing is disabled

deCONZ v2.19.3
Raspbian GNU/Linux Buster (10)
Raspbee 2 (26690700)

ebaauw, 2022-12-31:
Make sure the Aurora is awake by pressing / turning it, while pressing _Update_.