dresden-elektronik/deconz-rest-plugin #4624: Device name

ingo01978, 2021-03-21:
Hello, can you include this sensor. It is a self construction sensor for monitoring plants.
This sensor can measure soil moisture, temperature and brightness.
Here is the link to the original site:

## Device

Product name: DIYRuZ_FlowerV2
Manufacturer: modkam.ru
Model identifier: DIYRuZ_Flower
Device type : Sensor , Other

## Screenshots
[image]
[image]
[image]
[image]

### Basic

https://modkam.ru/?p=1700
https://github.com/diyruz/flower

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2021-03-21:
Hu ? It s normal all cluster are OUT cluster and not IN cluster ?

ingo01978, 2021-03-25:
Hello yes this is normal. The sensor does not send its values every 30 minutes and more. No settings can be made.

ingo01978, 2021-03-25:
On the 4 picture you can see the value 0408 with unknown, that should be the soil moisture.

Smanar, 2021-03-25:
But the problem is cluster need to be the reverse ? Look on other sensor.
It s like this device is able to receive temperature but not sending them.

closed 2021-04-23 by github-actions[bot]