dresden-elektronik/deconz-rest-plugin #4625: Ledvance SMART+ Motion Sensor disappearing

MaMaKow, 2021-03-21:
## Describe the question or issue you are having
I have a "Ledvance SMART+ Motion Sensor" connected to my ConBee II.
The ConBee II lives on a Raspberry Pi 4 with openhabian.

The "Ledvance SMART+ Motion Sensor" tends to "disappear".
Status Not reachable

## Screenshots
[image]
[image]
[image]

after new pairing:
[image]

## Environment
 - Host system: Raspberry Pi 4
 - Running method: Raspbian
 - Firmware version: 26660700
 - deCONZ version: 2.07.01 / 8.12.2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
[debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6178408/debug.txt)

## Additional context
There are 4 LEDVANCE Smart+ Plugs and one LEDVANCE Smart+ LED in the same network.
The distance between the Motion Sensor and the next Plug (->mesh) is below 1 meter.

Mimiix, 2021-03-22:
Can you provide logs on dbg-aps=2 and dbg-error=2 ?

MaMaKow, 2021-03-22:
I started `nohup /usr/bin/deCONZ -platform minimal --dbg-info=2 --dbg-aps=2 --dbg-error=2 > debug.txt & `
Right now the sensor is there. 
Should I let the log get bigger until the thing is gone?
Or Should I just stop after it has reached about a few Megabytes?
Is it OK, to have one logfile with all three options? Or is it better to have seperate files for the three options?
If so, can there be three processes of `/usr/bin/deCONZ` at once?

Mimiix, 2021-03-23:
Just have them in one log :) That's fine.

You can provide the logs already. If there's a fundamental issue, it would show regardless (interference, bad connections etc).

MaMaKow, 2021-03-26:
[splitme00.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6213664/splitme00.log)

MaMaKow, 2021-03-26:
[splitme02.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6213681/splitme02.log)
[splitme03.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6213704/splitme03.log)
[splitme04.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6213716/splitme04.log)
[splitme05.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6213726/splitme05.log)
[splitme06.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6213732/splitme06.log)
[splitme07.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6213746/splitme07.log)
[splitme08.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6213759/splitme08.log)
[splitme09.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6213780/splitme09.log)
[splitme10.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6213794/splitme10.log)

closed 2021-04-24 by github-actions[bot]