dresden-elektronik/deconz-rest-plugin #4626: Zemismart lxn59-2s7lx1 2 gang inline switch (wired) Zigbee 3.0

dodo124, 2021-03-22:
## Device

- Product name: Zemismart lxn59-2s7lx1 2 gang inline switches for Zigbee 3.0.
https://nl.aliexpress.com/item/4001115310745.html?spm=a2g0o.detail.1000060.1.32d3647fZRUTVq&gps-id=pcDetailBottomMoreThisSeller&scm=1007.13339.169870.0&scm_id=1007.13339.169870.0&scm-url=1007.13339.169870.0&pvid=f01159d0-0103-47cf-8fa7-d17c3f43139d&_t=gps-id:pcDetailBottomMoreThisSeller,scm-url:1007.13339.169870.0,pvid:f01159d0-0103-47cf-8fa7-d17c3f43139d,tpp_buckets:668%230%23131923%2380_668%230%23131923%2380_668%23888%233325%232_668%23888%233325%232_668%232846%238114%231999_668%232717%237562%23471_668%231000022185%231000066059%230_668%233468%2315609%23234_668%232846%238114%231999_668%232717%237562%23471_668%233164%239976%23763_668%233468%2315609%23234
- Manufacturer: Zemismart
- Model identifier: unknown
- Device type : Please remove all unrelated device types. 
  - Light

## Screenshots

### On/Off

So the switch is detected and working as long as I only want to controll the directly wired device (the main Bulb).
I can turn it on/off at the switch and turn on/off instantly on any remote device (Hue Essential, Phoscon, IO Broker, deconz).
Everything is fine until I want to use the same switch to controll another device connected to the hub.

e.g beside the main bulb there are other bulbs inside the room not directly connected to the same wire going to the main bulb and the switch. I want to turn these bulbs on when I turn on the main bulb in the room. Therfor I installed other on/off relays in the room to the additional bulbs. These relays are working fine andI also included the neccessary code in IO Broker and the code is working fine.

BUT it sometimes takes ages until the command is executed or it is not executed at all. You can see that directly inside the room becaus the main bulb turns (on/off) and the secoundary are doing nothing. Furthermore I can verify this inside deconz/phoscon where the state does not change. 

I checke the wiring and the code with a Lidl Silvercrest wall plug which is implimented as a "light" as well and the wall plug works perfectly (instantly).

Any ideas? 

Regards Dodo

Smanar, 2021-03-22:
Hello, the basic attribute panel i empty ^^, can you retry with pressing the "read" button.
And BTW can you show the cluster list ? To see if this device use tuya cluster or classic one ?

If the device use tuya cluster, It don't work like other device, for this one I wait for return to update the sate, other update the api first and send the zigbee request after.

So the first mode is more sensible to connexion issues, wich one deconz version are you using ? there is lot of change in tuya device.

dodo124, 2021-03-23:
Hi Smanar,

I updated the information above as best as I could. 
I do have some information from the manufacturer which I added as well.
And I do have the general.xml file for the cluster info but where do I put this?

Regards Dodo

Smanar, 2021-03-23:
Ok, so your device seem 100% zigbee compliant, nothing to see with exostic device.
So it works with standard mode. And because of that it s strange you have problem.

When you use the api to set the device "on", the api update the field before sending the command, so even if the device is out and don't receive the command, the state/on value become "true".
So if you are using this field for your automatism to act on other device, it s strange you have problem. You can have other device that turn "on" but not the zemismart, but I don't see how you can have the reverse ....

dodo124, 2021-03-23:
Yes you are right. Thats the situation. And as long as I turn on/off the switch inside Phoscon/hue etc. the reaction is as intended. But as soon as I press the button on the switch no information is transfered to the hub. It is like the state is not updated correctly.
I saw a capability in the manufactueres code labeld "refresh" maybe this is causing the problem. It seems to me that sometimes the switch instantly activates the command on the hub like it is in some kind of normal "update procedure" but most of the time it´s not.

Anything I can do to solve this mistery :-)

Greeting Dodo

Smanar, 2021-03-23:
>as I press the button on the switch no information is transfered to the hub

Ha, it s when you use the device manualy ? Do you know what is reporting ? and have you tried some other setting in deconz ?
The bug is systematic or random ?

dodo124, 2021-03-24:
Hi Smanar,

yes this occurs only when pressing the button manualy ( the physical button on the switch). 

Unfortunately I have no clue what it is sending, how do I find out? I´m missing a read "raw zigbee information" panel in deconz.

What other settings in deconz can be set? I did not find anything in the manual. I tried to set it as a sensor but could not find anything. 

The bug is pretty much systematic. It doesn´t work in 95% and then it suddenly works once only to fail 95% afterwards.

Smanar, 2021-03-24:
Here you can find a sample https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2196#issuecomment-695843587
You device use probably reporting, so you can take a look on the value for the attribute 0x0000 (on/off) in the cluster 0x0006 (on/off). If you use the button "read config" you can read previous values.

But I think this is set by defaut, it don't have something special ? have you tried to set it alone or in group ? Routing issues ? Network spam ?

On log deconz/helo/log, you can enable "error" and "error_l2", hopping for an error displayed, if you want  "raw zigbee information" try with "APS".

dodo124, 2021-03-25:
Hi Smanar,

you are the best. THe first hint was a success. After making a manual binding everything is working as expected. :-)

The question remains why it is not working with autobind? I will try the other options as well.

But the thread is closed.

Thank you very much.

closed 2021-03-25 by dodo124

Smanar, 2021-03-25:
>The question remains why it is not working with autobind? I will try the other options as well.

Good question, your device realy seem classic, so deconz should have used the classic configuration, and if it have failed, deconz is able to see it, I really have no ideas .

Smanar, 2021-03-27:
Ok I think swoop have find the problem https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4654