dresden-elektronik/deconz-rest-plugin #4628: Unable to pair 3 of 4 ZW25TQ

gmitch64, 2021-03-23:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi)
 - Running method: (Raspbian)
 - Firmware version: (26390500)
 - deCONZ version: (2.10.03 / 3/14/2021)
 - Device: (ConBee I)

## deCONZ Logs

## Additional context

Mimiix, 2021-03-23:
As per template, can you add the requested logs? Aps2, error 2 info 2?

Smanar, 2021-03-23:
They are battery device ? Or are represented as End device in deconz ?

gmitch64, 2021-03-23:
These are mains powered devices.
I'll try and pull logs during the joining process later today. Anything specific that you want?

Mimiix, 2021-03-23:
APS 2 , Error 2 and info 2 :)

gmitch64, 2021-03-24:
Just ran deCONZ to do the debugging, and I'm getting an error that I need a device firmware update, which is odd, as I think I'm on the latest firmware.

[image]

Smanar, 2021-03-24:
I think it can be usefull for you to make the try on a PC, to check if the problem is not from the conbee.

gmitch64, 2021-03-24:
I have a RaspBee, so that's not an option unfortunately. 
I see that I have the environment wrong in the initial report. Sorry.

gmitch64, 2021-03-24:
And a power off/on seems to have brought it somewhat back into life. Running deCONZ from ssh was giving me Qt errors, so I ran it from a VNC session, and now the only devices I'm seeing are these.

[image]

gmitch64, 2021-03-24:
Does this mean my database is dead? It's looking pretty dead from the webUI.
Does deCONZ make backups of the database anywhere?

gmitch64, 2021-03-24:
OK. I'm still not sure what's going on, but when running in debug mode, the 2 motors connected. The 2nd one doesn't look right, as it's certainly not "fairy lights".

[image]

gmitch64, 2021-03-25:
And finally the log files.

[M3.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6201210/M3.zip)

gmitch64, 2021-03-25:
I took deCONZ out of debug mode, rebooted, and they now showed up as smart light 28 and smart plug 29. I'm not sure what is going on.

I also tried to add the 2 additional motors that weren't adding in before, and THEY have added too.

[image]

For On/Off light 28

[image]

[image]

Form Smart plug 29

[image]

[image]

Smanar, 2021-03-25:
Just for information you have a third app that try to connect to the API unsuccessfully.
>21:53:30:477 API error 1, /sensors/new, unauthorized user

Remember you need 2 inclusions for tuya device, and without deleting the first entry, to transform the plug entry in covering one.

gmitch64, 2021-03-25:
Thanks for the info on the 3rd party app - I think I know what that is.

I didn't know about the double inclusion for the tuya device. Just to make sure that I did it right for these motors, since when I did it nothing changed...

Don't delete anything from Phoscon or the VNC console
Do a 3 press to get into inclusion mode
Search for new lights in Phoscon
Wait for things to happen

Did I do it right, or did I get it wrong? I've tried on all 4 motors, and nothing happens.

gmitch64, 2021-03-28:
Just to update this, in case anyone finds it via search later.
The 2nd inclusion for any of the 3 missing tubular motors didn't work.

What DID work, was reading the basic cluster for each of the motors once they were included. This changed the device in the JSON to a window covering device, which is what they are.

closed 2021-03-28 by gmitch64

Smanar, 2021-03-29:
Oups sorry, have forget you.
Yep, the second trick work too, but in fact it s the same thing than making 2 inclusion, idk why it have not worked.
There is a weak ATM in deconz, it miss the manufacture name during inclusion. So the first inclusion create a plug device and memorise data, the second one can use the previous stored data to reconize the device and include it as covering.

It s for that, you need to not delete something.