dresden-elektronik/deconz-rest-plugin #4630: SONOFF 01MINIZB goes in pairing mode alone

LeoeLeoeL, 2021-03-23:
## Describe the bug
 After few hours the device starts flashing its led and goes in pairing mode

## Steps to reproduce the behavior
 The issue happens always. To be reproduced I have to:
1) Go to Phoscon App and ADD NEW LIGHT
2) When paired, I have only to wait few hours until the Sonoff becomes gray in Phoscon and flashes its green led (Pairing mode). 

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 2668070
 - deCONZ version: v2.10.3-beta (same issue with previous versions)
 - Device: ConBee II
 - Do you use an USB extension cable: Yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? APC ups

## deCONZ Logs
 
[Log_2.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6189308/Log_2.txt)

## Additional context
In the network, the Sonoff described here has the following name: On/Off light 53

Mimiix, 2021-03-23:
Duplicate of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4564

closed 2021-03-23 by Mimiix

LeoeLeoeL, 2021-03-23:
The #4654 has been closed before opening this one because the #4654 was a USER QUESTION.
But it's a bug.
Now, both #4654 and #4630 are closed.
Please, open this one again.