dresden-elektronik/deconz-rest-plugin #4633: 951 internal error , type 901

Ran-M96, 2021-03-24:
## Describe the question or issue you are having
Hi everyone,
I would like to know if anyone can help me with an error that has been raising lately on my system:

I am currently using the deconz api to control and monitor 8 zigbee devices, this includes sending on/off request and also retrieving data consumption from the sensors associated to the lights.

This system has been working great, but lately, i have an internal error 951 when i start sending requests to the devices.

I did find that this might be due to the fact that i send too many requests in a short time, so i added a timeout of 10 seconds between every consumption get request, and 1 minute between the on/off put request, and it didn't change anything.

I also used an usb extension cable for the conbee to avoid trouble.
if anyone has an idea, it would be great.

## Screenshots

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Raspbian
 - deCONZ version: 2.09.03
 - Build with Qt 5.11.3
Node limit 5 (max. 200)
Commit b55b2ea7f535bb5d59878c9e6eab75ae10f1a8ef
 - Device: ConBee II 
 - Do you use an USB extension cable: yes

## deCONZ Logs

## Additional context
I would also like to have an official documentation of these "951" error on deconz which i didn't find yet, thank you!

Kane610, 2021-03-24:
901 error means bridge is busy, there is a queue of 20 commands that if exceeded will throw this error

Why don't you use the websocket for state changes?

Ran-M96, 2021-03-24:
> 901 error means bridge is busy, there is a queue of 20 commands that if exceeded will throw this error
> 
> Why don't you use the websocket for state changes?

Thank you for your response, well when i started this, the websocket tool was still under devlopment so i chose the stable version
I am considering it , but for now i would need a stable solution with this, i did read the 20 commands comments but found it nowhere on the doc, also is it 20 commands for each device, or on the whole REST api queue ? on both cases i put a timeout so that doesn't happen 

Do you have a link or something to see how this sueue is managed ? thank you !

Kane610, 2021-03-24:
It's for the rest api AFAIK. I have no references to how it's handled in deconz code

Smanar, 2021-03-24:
You can't use group feature ?
What is the command you need to send ?

Ran-M96, 2021-03-24:
> Hello,
> You can't use group feature ?
> What are commands you need to send ?
> 
> I don't think the problem is when you read data.

well, i send different commands to each light (there is 8 of them), so a group won't make it.
the commands i send is put requests via the rest api to put the state on "on"/"off", and get data to retrieve consumption from the sensors (only 5)
yeah i also don't think it is due to the number of requests as they are quite few, but the get consumption request is the one i use the most often so if the error is related to the number of requests, it would make sense that it comes from the get requests

Smanar, 2021-03-24:
Yep but when you read data, you just read values already stored in the api, you don"t use the zigbee network, so it s faster than making zigbee request to change a state.

Ran-M96, 2021-03-24:
> Yep but when you read data, you just read values already stored in the api, you don"t use the zigbee network, so it s faster than making zigbee request to change a state.

But to retrieve data from the device, the api get it from the device via the zigbee protocole right ? Or does it do it on a regular basis with or without request ?

Smanar, 2021-03-24:
No, it s not possible query a sensor, and for light we do same, we wait for the device send itself the information (with reporting for example) and store it.

Ran-M96, 2021-03-24:
> No, it s not possible query a sensor, and for light we do same, we wait for the device send itself the information (with reporting for example) and store it.
 OK so what do you think this internal error come from ?

Smanar, 2021-03-24:
I think it s more from the command you are sending to the light, not when you read data.
But taking a look in log is a good way to see the blocking part.

SwoopX, 2021-03-24:
If you'd provide some logs, we should be able to find out.

Ran-M96, 2021-03-24:
Okay, i will publish them tomorrow, i also switched to the websocket feature for the sensor devices, this might not change anything but still can reduce data latency,  thank you for all your answerq, i'll  add the logs tomorrow !

SwoopX, 2021-03-24:
kk, this is currently a bit like trying to diagnose an engine problem without popping the hood ;)

Ran-M96, 2021-03-25:
Hello again !
I think i found the origin of the problem, may not be the bridge busy  option.
I launched the deconz log once i got the internal problem, and redirected it to a file, using this command : 
[](url)
`deCONZ --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1 > Deconz-log.txt`

Once i did that , the deconz gui popped up, but there was no conbee detected,  which wasn't the case on the first ten minutes when i launched my program.
I rebooted the raspberry, and launched the log files from the beggining, so far there is no internal error, i'm waiting for it to pop

Also, the deconz gui asked me for an firmware update Twice ! which i did, the second time around, i clicked on it but nothing happen

PS : the command **GCFFlasher_internal -l** shows the conbee at first, but once the internal error pops, there is nothing but the  firmware is 0x000000000

I'm waiting for the logs to catch the error, so far here is what i got 
[Deconz-log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6205639/Deconz-log.txt)

Smanar, 2021-03-25:
Make a try on a PC to test if you have not problem on the device ? The disconnection can be from the firmware (can need to re upload it), if it s that will be fast to see the USB disconnection on it.

Ran-M96, 2021-03-25:
yep, here's what i got : 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian

Smanar, 2021-03-25:
And have same issue on a PC machine ?

Ran-M96, 2021-03-30:
Hi, here are the logs when the internal error occurs

I didn't get the chance to try it on pc yet 
[Deconz-log2.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6229801/Deconz-log2.txt)

Ran-M96, 2021-03-30:
I tried it on pc, didn't use my software in it as it would take me some time, but when the conbee is plugged, the deconz app detect it and the firmware is good, also the lights are directly found, seems like there is no problem on win10

Smanar, 2021-03-30:
```
12:34:40:793 COM: /dev/ttyUSB0 : USB-RS232 Cable (0x0403/0x6001)
12:34:40:794 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
12:34:40:795 COM: /dev/ttyS0 :  (0x0000/0x0000)
12:34:40:795 dev /dev/ttyS0 (/dev/ttyS0)
12:34:40:796 auto connect com /dev/ttyS0
```

What is the result of "lsusb" and "GCFFlasher_internal -l" ?

Ran-M96, 2021-04-15:
> 
> 
> ```
> 12:34:40:793 COM: /dev/ttyUSB0 : USB-RS232 Cable (0x0403/0x6001)
> 12:34:40:794 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
> 12:34:40:795 COM: /dev/ttyS0 :  (0x0000/0x0000)
> 12:34:40:795 dev /dev/ttyS0 (/dev/ttyS0)
> 12:34:40:796 auto connect com /dev/ttyS0
> ```
> 
> It s worse than last time.
> What is the result of "lsusb" and "GCFFlasher_internal -l" ?

Hi, 
sorry i took some time to respond, it appeared that there was another zigbee network near mine , so i removed it and that fixed the situation at first, i was sure that was the problem
As you might have guessed, the problem appeared again, the result of 'lsusb' and 'GCFFlasher_internal -l' seem normal, 
one weird thing is that on the deconz app, the firmware is on and connected, but on the phoscon app, the conbee is detected but the firmware is mentionned as "not connected"
Thannk you again !

[image]

Ran-M96, 2021-04-15:
Also, i updated the last version of the firmware

Smanar, 2021-04-15:
>on the deconz app, the firmware is on and connected, but on the phoscon app, the conbee is detected but the firmware is mentionned as "not connected"

You are sure not having 2 deconz instances ?

Ran-M96, 2021-04-15:
[image]

Seems to be only one :/

Smanar, 2021-04-15:
I m not a linux proffessionnal ^^ but I know better result with "ps ax | grep deCONZ"

Ran-M96, 2021-04-15:
ok, here's the result, there is the deconz upsate.sh but not quite sure what is it for
[image]

In the logs nothing new, no particular warning or error showing ...

Smanar, 2021-04-15:
no, it s fine, have same.

Ran-M96, 2021-04-20:
I think i found the problem, on the deconz application, i lose the conbee stick sometimes which probably cause the internal error, the thing is that i use an extension cable (which i can change to test its quality), but beside that
Has anyone experienced this before ? and is this due to the app or the material ?
[image]

Ran-M96, 2021-04-20:
[log deconz.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6343629/log.deconz.txt)

Je joins les derniers logs, collectés pendant l'internal error, je précise qu'à ce moment là j'avais débranché le conbee et rebranché , le network etait donc connecté, cependant l'internal error est toujours là

Mimiix, 2021-04-20:
@Ran-M96  CAn you please speak english? Nothing strange in the log.

Ran-M96, 2021-04-20:
> 
> 
> @Ran-M96 CAn you please speak english? Nothing strange in the log.

Sorry ! i wrote in a rush,  i didn't pay attention :)

speaking of french, i was reading in a deconz french forum that the conbee disconnection was due to some material problem happening on some of the conbee II devices, if i can find a new user having this problem too, i will join the deconz support to see how this can be fixed ( update or change of the conbee)

Smanar, 2021-04-20:
@Ran-M96 the only thing you can do to be sure is
- Uploading the firwmare again
- Make test on a REAL PC (not a virtual machine), with extension cable, with avoiding USB 2.0.

PC (windows or ubuntu for exemple) have more power, better shield, better hardware, better OS, realy less issues

If you still have the disconnection issue (visible on log or on deconz interface), yes I think you can contact the support with explaining your tests.

closed 2021-05-19 by github-actions[bot]