dresden-elektronik/deconz-rest-plugin #4634: _TZ3000_xzqbrqk1 inverted state

bastshoes, 2021-03-24:
## Describe the bug

After upgrade to 2.10.04 blinds switch reports inverted state.

```
{
    "etag": "c692cfa81a495f8f33fdc1f2f4fbdab4",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-03-24T12:32Z",
    "manufacturername": "_TZ3000_xzqbrqk1",
    "modelid": "TS130F",
    "name": "Livingroomr Blinds",
    "state": {
        "bri": 254,
        "lift": 100,
        "on": true,
        "open": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Window covering device",
    "uniqueid": "5c:02:72:ff:fe:ca:a1:9b-01"
}
```

## Steps to reproduce the behavior

Upgrade to 2.10.04 from 2.10.02
## Expected behavior

Report proper state
## Screenshots

## Environment

 - Host system: PC
 - Running method: Marthoc Docker container
 - Firmware version: 26680700
 - deCONZ version: 2.10.04
 - Device:  ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? NO

## deCONZ Logs

## Additional context

Mimiix, 2021-03-24:
@Smanar Did you integrate it?

bastshoes, 2021-03-24:
Last known working version in 2.09.03

Smanar, 2021-03-24:
IDK what happen, I m comparing the version 2.9.3 and 2.10.4, https://github.com/dresden-elektronik/deconz-rest-plugin/compare/v2.9.3...v2.10.4
And the only difference I m seing is the productid change from "11830304(2) Switch" to "Zigbee curtain switch", but the code handle them the same way.

```
Remove            R_GetProductId(taskRef.lightNode) == QLatin1String("11830304(2) Switch") ||
Add                  R_GetProductId(taskRef.lightNode) == QLatin1String("Zigbee curtain switch") ||
```

bastshoes, 2021-03-24:
Yes. I'm sure.

I've tried all betas (2.10.03, 2.10.02, 2.10,01) and all reports inverted state. When I've returned to 2.09.03 all works fine.

I've noticed one strange thing. State shows correctly after upgrade from 2.09.03 to any version. But it will be inverted on first device state change.

bastshoes, 2021-03-24:
@Smanar I guess I've found the problem.

If I sent GET API call on curtain switch I see the following in the logs
```
19:39:35:701 Websocket 172.18.0.1:58902 send message: {"attr":{"id":"7","lastannounced":null,"lastseen":"2021-03-24T16:39Z","manufacturername":"_TZ3000_xzqbrqk1","modelid":"TS130F","name":"Livingrooml Blinds","productid":"11830304(2) Switch","swversion":null,"type":"Window covering device","uniqueid":"5c:02:72:ff:fe:ca:a1:e4-01"},"e":"changed","id":"7","r":"lights","t":"event","uniqueid":"5c:02:72:ff:fe:ca:a1:e4-01"} (ret = 366)
```

ProductId is still prevoius one - "productid":"11830304(2) Switch"

That is way it's changed behavior.

How can I change productid?

bastshoes, 2021-03-24:
Deleting device and adding it again gives right productid and fixes the issue.

Smanar, 2021-03-24:
ha yes nice, I think the old one was memorised on the database, good to know, thx a lot, I think you will not be the only one with this kind of issue with all the chnage I do all the time.

bastshoes, 2021-03-25:
For sure. Do we have any tool, that can check item productid and change it on start?

Smanar, 2021-03-25:
It more my fault ^^, produc id is not made to be changed at every version, but this code will move again soo, deconz will use another approch to manage devices, easier for us and for you.

bastshoes, 2021-03-26:
Will repair required?

Smanar, 2021-03-26:
For the new modification ?
IDK yet, I hope no, but the code will use JSON file, so big change, easier to custom with users shy with c++ or without linux machine.

closed 2021-04-24 by github-actions[bot]