dresden-elektronik/deconz-rest-plugin #4639: Aqara leak water sensor SJCGQ11LM, no information on Deconz

vini59fr, 2021-03-24:
## Describe the bug
Hi

I have an issue with a SJCGQ11LM, Aqara water leak sensor
It works fine with my Home Assistant / Conbee2 / Phoscon-Deconz

But this sensor don't give any information on Deconz

## Steps to reproduce the behavior

## Expected behavior

## Screenshots
[image]

## Environment
Version 2.10.04 / 14/03/2021
Firmware 26660700

 - Host system: Raspberry Pi4
 - Running method: Home Assistant deCONZ Add-on 
 - Firmware version: 26607000
 - deCONZ version: 2.10.04
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? USB Storage disk (SSD inside)

## deCONZ Logs

## Additional context

SwoopX, 2021-03-24:
> It works fine with my Home Assistant / Conbee2 / Phoscon-Deconz

So what is the bug then?

vini59fr, 2021-03-24:
> > It works fine with my Home Assistant / Conbee2 / Phoscon-Deconz
> 
> So what is the bug then?

There's non information on this sensor (look at the screenshot) 
It even don't appear on the OTAU Update list

SwoopX, 2021-03-24:
Yes, this is all expected and known. The various issues around the device mention that as well. It also doesn't have/support any OTAU.

vini59fr, 2021-03-24:
> Yes, this is all expected and known. The various issues around the device mention that as well. It also doesn't have/support any OTAU.

Oh, I didn't know that. 
Thank's for the informations

i close so

closed 2021-03-24 by vini59fr