dresden-elektronik/deconz-rest-plugin #4641: 2.10.04 breaks turning off Moes thermostats: /sensors/70/config/mode invalid value, off, for parameter mode

HenningGithub, 2021-03-25:
## Describe the bug
After updating the deconz plugin in home assistant to 6.8.0 yesterday, my thermostats won't turn off any more (climate.turn_off service in HA).
Turning them on (climate.turn_on in HA -> mode: "heat") still works.
It worked just fine until the update. I had addon_core_deconz_6.7.2 before.

HA log shows:
pydeconz.errors.RequestError: /sensors/70/config/mode invalid value, off, for parameter mode
2021-03-25 10:17:40 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [1792183288] /sensors/70/config/mode invalid value, off, for parameter mode
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 141, in handle_call_service
await hass.services.async_call(
File "/usr/src/homeassistant/homeassistant/core.py", line 1488, in async_call
task.result()
File "/usr/src/homeassistant/homeassistant/core.py", line 1523, in _execute_service
await handler.job.target(service_call)
File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
await self.hass.helpers.service.entity_service_call(
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 642, in entity_service_call
future.result() # pop exception if have
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 681, in async_request_call
await coro
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 679, in _handle_entity_call
await result
File "/usr/src/homeassistant/homeassistant/components/climate/init.py", line 487, in async_turn_off
await self.async_set_hvac_mode(HVAC_MODE_OFF)
File "/usr/src/homeassistant/homeassistant/components/deconz/climate.py", line 193, in async_set_hvac_mode
await self._device.async_set_config(data)
File "/usr/local/lib/python3.8/site-packages/pydeconz/deconzdevice.py", line 33, in async_set_config
await self.async_set(field, data)
File "/usr/local/lib/python3.8/site-packages/pydeconz/api.py", line 118, in async_set
await self._request("put", field, json=data)
File "/usr/local/lib/python3.8/site-packages/pydeconz/gateway.py", line 109, in request
_raise_on_error(response)
File "/usr/local/lib/python3.8/site-packages/pydeconz/gateway.py", line 225, in _raise_on_error
raise_error(data["error"])
File "/usr/local/lib/python3.8/site-packages/pydeconz/errors.py", line 58, in raise_error
raise cls("{} {}".format(error["address"], error["description"]))
pydeconz.errors.RequestError: /sensors/70/config/mode invalid value, off, for parameter mode

Device as reported from GET /sensors/70/:
{
"config": {
"heatsetpoint": 1900,
"locked": false,
"mode": "heat",
"offset": 0,
"on": true,
"preset": "program",
"reachable": true,
"schedule": {},
"schedule_on": false,
"setvalve": false
},
"ep": 1,
"etag": "a33d008e6cac58f652a2c6f95e2806de",
"lastseen": "2021-03-25T09:26Z",
"manufacturername": "_TZE200_aoclfnxz",
"modelid": "TS0601",
"name": "Wohnzimmer Heizung",
"state": {
"lastupdated": "2021-03-25T09:24:09.206",
"on": null,
"temperature": 1900
},
"type": "ZHAThermostat",
"uniqueid": "84:71:27:fe:ff:1b:77:02-01-0201"
}
state in HA:
hvac_modes:
  - auto
  - heat
  - 'off'
min_temp: 7
max_temp: 35
preset_modes:
  - auto
  - boost
  - comfort
  - complex
  - eco
  - holiday
  - manual
current_temperature: 17.5
temperature: 19
preset_mode: null
locked: false
friendly_name: Wohnzimmer Heizung
supported_features: 17

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment
 - Host system: Raspberry Pi 3B+
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26390500
 - deCONZ version: 2.10.04 / 3/14/2021
 - Device: RaspBee I / RaspBee II (dunno which)

## deCONZ Logs

## Additional context

Mimiix, 2021-03-25:
Home-assistant tries to set a mode that isn't available in the rest API. 

@Smanar did you change anything to this thermostat? Can you help debug a bit?

Smanar, 2021-03-25:
Have found the problem, haven't see what happen yet, but I think it s definitely my fault.
BTW, you haven't an unix machine to make test ?

The FIX will be light, perhaps we can have it in the next beta.

HenningGithub, 2021-03-25:
Unix yes, Linux no.
You probably have released 5 more versions before I get around to even set up cross compilers to armv7l-linux...

Smanar, 2021-03-25:
Lol, yes sorry, but from my memory HA use to beta version ?

HenningGithub, 2021-03-25:
I have no idea if HA can be tricked into using a beta addon. I have always claimed I want to use HA like an appliance, and it isn't too much of a lie ;)
My immediate problem - heaters not turning off is one ;) - was solved by restoring v6.7.2 of the addon right after writing the bug report.

closed 2021-03-27 by manup