dresden-elektronik/deconz-rest-plugin #4642: Ewelink ZigBee Relais

b2un0, 2021-03-25:
## Device

- Product name: Ewelink ZigBee Relais
- Manufacturer: eWeLink
- Model identifier: currently as ZB-SW01, but looks wrong
- Device type : on/off Plug (Relay)
- Product: https://a.aliexpress.com/_u60ZaN

The Device is currently discovered as Light, instead of Plug

## Screenshots

[image]

[image]

### Basic
[image]

### On/Off
[image]

SwoopX, 2021-03-28:
The device should be fully working, or not?

b2un0, 2021-04-06:
yes, the device is fully working.

closed 2021-05-05 by github-actions[bot]

goermezer, 2021-05-27:
The OnTime and OffWaitTime attributes are read only. Is it possible to define them as writable, so that I can change them? In the UI of deCONZ provides the possibility to change them. But the attributes do not change.

I would like to use this device as a door opener, but the inching time with ~0.5s is too short. I would like to set about 3-5 seconds wait time until switching the relay off.

PR #6899 (2023-04-19): Fix ZCLDB On/Off cluster attribute compliance
Update On/Off Cluster (0x0006) to be complinant with Zigbee Cluster library specifications : 
- Default value for `GlobalSceneControl` should be 0x01
- `OnTime` and `OffWaitTime` should be writeable

Close #4642 and fix #6888