dresden-elektronik/deconz-rest-plugin #4644: 2.10.04 showing 2.09.03 / 14.3.2021

popy2k14, 2021-03-25:
## Describe the question or issue you are having
As stated in the title. After Updating to 02.10.0.4 on my RPi4 Phoscon and HUE Essential (REST API)
are showing 2.09.03.

`
 sudo apt-get install deconz
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  lxplug-volume wiringpi
Use 'sudo apt autoremove' to remove them.
The following packages will be upgraded:
  deconz
1 upgraded, 0 newly installed, 0 to remove and 23 not upgraded.
Need to get 6,184 kB of archives.
After this operation, 57.3 kB of additional disk space will be used.
Get:1 http://phoscon.de/apt/deconz buster/main armhf deconz armhf 2.10.04-raspbian-buster-stable [6,184 kB]
Fetched 6,184 kB in 1s (7,485 kB/s)
Reading changelogs... Done
(Reading database ... 240696 files and directories currently installed.)
Preparing to unpack .../deconz_2.10.04-raspbian-buster-stable_armhf.deb ...
Important: /boot/cmdline must not contain console=serial0 (RaspBee only)

use raspi-config to configure serial port access as described in user manual (RaspBee only)
Unpacking deconz (2.10.04-raspbian-buster-stable) over (2.09.03-raspbian-buster-stable) ...
Setting up deconz (2.10.04-raspbian-buster-stable) ...
Important: /boot/cmdline must not contain console=serial0 (RaspBee only)

use raspi-config to configure serial port as described in documentation (RaspBee only):
https://github.com/dresden-elektronik/deconz-rest-plugin#software-requirements
Processing triggers for mime-support (3.62) ...
Processing triggers for desktop-file-utils (0.23-4) ...
pi@rfhem-pi:~ $ sudo systemctl status deconz
`

## Screenshots
[image]

## Environment
 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26680700
 - deCONZ version: 2.09.03 but it should be 2.10.04
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? NO If so: Which?

## deCONZ Logs

## Additional context

popy2k14, 2021-03-25:
Is this just an "forgot to bump the version" or do i have some frankesteining going on here :-)

robidog, 2021-03-25:
Observed the same yesterday right after updating. Today it magically shows the correct version. <shrug>

popy2k14, 2021-03-25:
Ok, thx. Then I will wait for tomorrow and hope for the magic happen 😉

popy2k14, 2021-03-26:
It still shows 2.09.03. 
ANy hints whats going on?

senilio, 2021-03-28:
Clear browser cache, or try another browser.

popy2k14, 2021-03-29:
Done clearing cache, same issue on Chrome.
Also edge showing 2.09.03 but i NEVER opened Phoscon there (so also could be no cache issue).
Same issue on my smartphone in chrome and in HUE Essentials.

Any hints?

ebaauw, 2021-03-29:
Self-compiled REST API plugin?  What version does the GUI show under _Help_ | _About deCONZ_?  What version does the API show under `swversion` when doing a GET of `/api/config`?

popy2k14, 2021-03-29:
Never compiled it my self. Just back used an self compiled version from here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3744 but this one was 2.8.x.

Here is the gui help -> about:

[image]

And the "api/config" request:

`{"apiversion":"1.16.0","bridgeid":"XXXXXX","datastoreversion":"93","devicename":"ConBee II","factorynew":false,"mac":"XXXXXXX","modelid":"deCONZ","name":"XXXXXX","replacesbridgeid":null,"starterkitid":"","swversion":"2.9.3"}`

Here is the output fro the plugins folder:

`pi@rfhem-pi:/usr/share/deCONZ/plugins $ ls -l
total 2692
-rwxr-xr-x 1 root root 2518356 Mar 23 18:09 libde_rest_plugin.so
-rwxr-xr-x 1 root root   38952 Mar 23 18:09 libde_signal_plugin.so
-rwxr-xr-x 1 root root  193236 Mar 23 18:09 libstd_otau_plugin.so
pi@rfhem-pi:/usr/share/deCONZ/plugins $`

Nothing self compiled, all have the same build date and time.

Strange!?
Any hints what i am doing wrong :-)

popy2k14, 2021-03-29:
Now its really weird!!!
Like @robidog stated, the version number changed magically, it was the same now!
Did restarted the service a few times after update to 2.10.4, but it showed 2.9.3.
Also the api/config request from a few minutes ago showed 2.9.3.

Now after stopping  the "deconz" (cli) version, starting the "deconz-gui" version (just for the screenshot) and doing the same in reverse did the "magic" trick for me.

Now it shows 2.10.4 on phoscon and als api/config.

[image]

`{"apiversion":"1.16.0","bridgeid":"XXXXX","datastoreversion":"93","devicename":"ConBee II","factorynew":false,"mac":"XXXXX","modelid":"deCONZ","name":"XXXXX","replacesbridgeid":null,"starterkitid":"","swversion":"2.10.4"}`

@robidog did you also swapped deconz versions (gui and cli) when it started to show the correct version?

@ebaauw where is the version strin pulled from? DB or hardcoded? Could it be that just the "gui" version updating something in DB which can lead to the issue we faced?

thx

ebaauw, 2021-03-29:
Afaik, the version of the REST API plugin is hard-coded into the shared library, through a compile-time define of `GW_SW_VERSION` in `de_web.pro`.  I don't know about the core programme, but haven't seen anything in the database holding the version.  The GUI shows the core programme version; Phoscon and any other API client, of course, show the API plugin version.

closed 2021-04-27 by github-actions[bot]

popy2k14, 2021-04-27:
As stated by @robidog - also for me it fixed itself after a few restarts.
An no, it was not a browser cache issue! Tested with several browsers, cleared cache of my main browser (chtome), with no success.
Also Hue Essentials showed the wrong version.