dresden-elektronik/deconz-rest-plugin #4645: RGBGenie ZB-5004

tarasfilonenko, 2021-03-25:
## Device

- Product name: ZB-5004
- Manufacturer: RGBgenie 
- Model identifier: RGBgenie ZB-5004 
- Device type :  Remote
- Product: https://rgbgenie.com/?product=rgbgenie-4-zone-2-scene-remote-zigbee

## Note
Similar device issue: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2648

## Screenshots
### Basic
[image]
[image]

### Level Control
[image]
[image]

## Additional
[image]

SwoopX, 2021-03-28:
Could you please check if the following button map also applies for your model? If yes, if's probably already sufficient if you find the file `button_maps.json` on your system and add your model.

```
        "icasaRemoteMap": {
            "vendor": "iCasa",
            "doc": "Remotes",
            "modelids": ["ICZB-RM11S", "ZGRC-KEY-012", "ZGRC-KEY-013", "RGBgenie ZB-5001"],
            "map": [
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "1", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "Move down (with on/off)"],
                [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "1", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "Move up (with on/off)"],
                [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x01", "SCENES", "RECALL_SCENE", "1", "S_BUTTON_9", "S_BUTTON_ACTION_SHORT_RELEASED", "Recall scene 1"],
                [1, "0x01", "SCENES", "RECALL_SCENE", "2", "S_BUTTON_10", "S_BUTTON_ACTION_SHORT_RELEASED", "Recall scene 2"],
                [1, "0x02", "ONOFF", "OFF", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x02", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "1", "S_BUTTON_3", "S_BUTTON_ACTION_HOLD", "Move down (with on/off)"],
                [1, "0x02", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "1", "S_BUTTON_3", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x02", "ONOFF", "ON", "0", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x02", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_4", "S_BUTTON_ACTION_HOLD", "Move up (with on/off)"],
                [1, "0x02", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_4", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x03", "ONOFF", "OFF", "0", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x03", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "1", "S_BUTTON_5", "S_BUTTON_ACTION_HOLD", "Move down (with on/off)"],
                [1, "0x03", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "1", "S_BUTTON_5", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x03", "ONOFF", "ON", "0", "S_BUTTON_6", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x03", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_6", "S_BUTTON_ACTION_HOLD", "Move up (with on/off)"],
                [1, "0x03", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_6", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x04", "ONOFF", "OFF", "0", "S_BUTTON_7", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x04", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "1", "S_BUTTON_7", "S_BUTTON_ACTION_HOLD", "Move down (with on/off)"],
                [1, "0x04", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "1", "S_BUTTON_7", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x04", "ONOFF", "ON", "0", "S_BUTTON_8", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x04", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_8", "S_BUTTON_ACTION_HOLD", "Move up (with on/off)"],
                [1, "0x04", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_8", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"]
            ]
        },
```

tarasfilonenko, 2021-03-28:
Thanks @SwoopX , it worked please see raised PR: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4667

SwoopX, 2021-04-04:
Closing this one based on confirmation and the already merged PR.

closed 2021-04-04 by SwoopX