dresden-elektronik/deconz-rest-plugin #4648: Mi Light Detection Sensor

goegsig, 2021-03-26:
## Device

- Product name: Mi Light Detection Sensor
- Manufacturer: XIAOMI
- Model identifier: lumi.sen_ill.mgl01
- Device type : 
  - Sensor

## Screenshots

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]
[image]

SwoopX, 2021-03-26:
Hi,

would you be able to compile and test an updated plugin version?

goegsig, 2021-03-27:
At the moment iam using Home Assistent.  But if there is some guide i can follow i'am willing to try setting at virtual linux op and trying to compile and verify that it works.

Finne75, 2021-03-30:
Three devices of this type did arrive today. What is a bit curious is that I see this device in the deCONZ GUI with measurements updated (after I did manual binding as recommended in #2380), but it is not visible in the REST API. I did a restart of the gateway, but no success. That means that I also cannot use the device in home assistant currently :/..

Finne75, 2021-04-02:
@SwoopX success! Do you need any more details?

[image]

[image]

closed 2021-04-02 by manup

johandijkstra, 2021-05-12:
Hi guys, I have the same sensor and issues stated here :)
How can I get the GZCGQ01LM to work in HA? :)

sj3fk3, 2021-05-12:
> Hi guys, I have the same sensor and issues stated here :)
> How can I get the GZCGQ01LM to work in HA? :)

Wait a couple of days and it will be in the new version of deconz :-)