dresden-elektronik/deconz-rest-plugin #4655: UseeLink Zigbee Smart Power Strip (4 plus+USB) Supported?

majorsl, 2021-03-27:
## Describe the question or issue you are having

Just curious if anyone is using this Zigbee power strip and if it is supported.  If not, I'm tempted to get one and do a device request. If it works like the Wifi version, each plug and the set of 4 USB ports are controllable so 5 relays in total.

The link is to the US version, but there are also EU and others too, so probably all the same controller inside.

## Screenshots

https://www.aliexpress.com/item/4000734576353.html

## Environment

 - Host system: (Raspberry Pi)
 - Running method: (Marthoc Docker container)
 - Firmware version: (26680700)
 - deCONZ version: (2.09.03)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes)
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

NA
## Additional context

NA

Mimiix, 2021-03-30:
I havent seen this device before. But Sockets /relays are often working out of the box. If not, it shouldn't be hard to integrate.

#followingthiswithinterest 😄

closed 2021-04-29 by github-actions[bot]

majorsl, 2021-08-17:
Sorry for the delay for those following, but I finally purchased one and it works like a charm. 4 outlets + 1 USB bank shows as 5 lights.  Individually controllable, of course, and picked up also by Home Assistant perfectly.

loryanstrant, 2022-11-09:
How did you get it picked up by ZigBee? I have no issue pairing it with the Tuya app, but when I put it in pairing mode ZigBee2MQTT doesn't see it.