dresden-elektronik/deconz-rest-plugin #4658: ELKO SmartDim Puck Uni 200W (puck dimmer)

andersop91, 2021-03-27:
## Device

- Product name: SmartDim puck Uni 200W
- Manufacturer: Schneider Electric
- Model identifier: PUCK/DIMMER/1
- Device type : 
  - Light (dimmer)

User guide:
https://proff.elko.no/getfile.php/13158642/produkter/doc/EKO07090/Bruksanv.%20SmartDim%20Puck%202.pdf
Information:
https://elkosmart.elko.no/produkter/smartdim-puck-uni-200w
https://proff.elko.no/smartpuck/category2982.html

## Screenshots
[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]

### Scenes
[image]

### On/Off
[image]
[image]

### Level Control
[image]
[image]

### Ballast configuration
[image]

### Diagnostics
[image]

### OTAU
[image]

Smanar, 2021-03-29:
Hello
>The unit works almost out of the box

What is missing ?

andersop91, 2021-04-04:
Configuration of ballast, and minimum/maximum dim level.

Smanar, 2021-04-05:
I have read the doc, relay nice device, dimmer module without neutral.
But nothing about "ballast configuration" in the doc, are you sure it s something "made to be modified" ?
Have you tried to use the dimmer feature with deconz ? In cluster 0x0008, it don't work ? Or the device use the "ballast configuration" cluster to make variation ?

closed 2021-05-05 by github-actions[bot]