dresden-elektronik/deconz-rest-plugin #4663: Moes ZigBee Smart DIY Motorized Roller AM43-0.45/40-ES-EZ(TY) with Integrated Battery

Aflow80, 2021-03-28:
## Device

- Product name: Moes ZigBee Smart DIY Motorized Roller AM43-0.45/40-ES-EZ(TY) with Integrated Battery
- Manufacturer: _TZ200_rddyvrci
- Model identifier: TS0601
- Device type :
  - Other: Window Covering

## Screenshots

[image]

[image]

### Basic
[image]

Smanar, 2021-03-28:
Hello, have you a linux machine with classic OS to test a code modification ?

Aflow80, 2021-03-28:
> Hello, have you a linux machine with classic OS to test a code modification ?

Is a Raspberry Pi4 with Raspbian valid?

Smanar, 2021-03-29:
Sure.

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you
```
sudo apt install deconz-dev
git clone --branch am43 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

With this code, you need 2 pairing without deleting something, the first one will create a battery sensor, the second one wil use informations from the battery sensor to create the covering device

Aflow80, 2021-03-29:
> Sure.
>
> You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
> So for you
>
> ```
> sudo apt install deconz-dev
> git [...]

Thanks for the help but unfortunately I’m still unable to get the device recognized in phoscon after compiling the module. I tried 2 consecutive times the inclusion after deleting the original node but I don’t see nor the battery sensor nor the window covering. What additional info can I provide to try to find the issue?.

Mimiix, 2021-03-30:
It won't show in Phoscon. It will only show in the RestAPI.

Smanar, 2021-03-30:
Yep, you need to use third app or direclty the API.
If you haven't third app, I can explain how to use the API ?

Aflow80, 2021-03-30:
Hi, After your comments and having the same results, I reviewed the source code and I have found a small typo: the manufacturer ID was incorrectly mistyped as "_TZ2000_rddyvrci" instead of "_TZE200_rddyvrci". After modifying that in de_web_plugin.cpp and resource.cpp files everything start to work properly. The device has been correctly recognized!!!.

Smanar, could you please correct it in your source code?. Thanks a lot for the help.

Smanar, 2021-03-30:
Done thx, corrected and PR online.

Aflow80, 2021-03-31:
I’m getting a low battery alarm in Homekit for the window covering. Is it possible to get the actual battery level or at least remove the created battery sensor in the api after inclusion??.

thanks.

Aflow80, 2021-04-01:
> Good question.
> Can you share the battery device json ? or at least give me the value used.
> You have alarm because of low battery or "null" value ?
> 
> Yes you can remove the battery sensor (using the api), but better having it working.

Here is the json of the battery device as seen in the rest api:
```
"54": {
    "config": {
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "df8f5678466740ae267553149562e156",
    "lastseen": "2021-04-01T07:45Z",
    "manufacturername": "_TZE200_rddyvrci",
    "modelid": "TS0601",
    "name": "Battery 54",
    "state": {
      "battery": null,
      "lastupdated": "none"
    },
    "type": "ZHABattery",
    "uniqueid": "84:71:27:ff:fe:0b:6e:0e-01-ef00"
  }
```
The alarm is probably generated by the "null" value reported. I'm see also that the last updated field is none.

Thanks.

Smanar, 2021-04-01:
Corrected, the code compile now.

Smanar, 2021-04-01:
Ok, so if after a while the value is still "null", there is a problem.
Have you tried the official application to know if there is an information about the battery level, or a low level warning ? If it s a low level warning, I need to make change.

I don't found information, about this missing command. 
Have you a machine able to log deconz during at least some hour ? To try to catch the missing command.

Smanar, 2021-04-01:
Edit wait have found a problem

Aflow80, 2021-04-01:
> Ok, so if after a while the value is still "null", there is a problem.
> Have you tried the official application to know if there is an information about the battery level, or a low level warning ? If [...]

I don't have the official hub so the official tuya app will not connect to my deCONZ hub unfortunately. Anyway I have review the printed manual and the screen captures of the app doesn't show any battery level.

Regarding the logging, I suppose the Raspberry Pi can do it, right?

Smanar, 2021-04-01:
>Regarding the logging, I suppose the Raspberry Pi can do it, right?

Yes, but hard for him, and not sure it can be usefull.

IDK what to do.
- remove this sensor : I need it for the moment for the device inclusion
- set a value in "battery", but will be a fake one, better to use the "null"

You have no way to ignore it ?
Else idk how deconz will react if you delete it, at next restart it will be not able to use the sensor information to reconize the device, so deconz NEED to use light informations, and not sure they are saved for light.

Aflow80, 2021-04-01:
> > Regarding the logging, I suppose the Raspberry Pi can do it, right?
> 
> Yes, but hard for him, and not sure it can be usefull.
> 
> IDK what to do.
> 
> * remove this sensor : I need it for the moment for the device inclusion
> * set a value in "battery", but will be a fake one, better to use the "null"
> 
> You have no way to ignore it ?
> Else idk how deconz will react if you delete it, at next restart it will be not able to use the sensor information to reconize the device, so deconz NEED to use light informations, and not sure they are saved for light.
> 
> No way to set homebridge detect battery problem when value is low but not = null ?

I have tried to delete the battery resource and after rebooting deconz the window covering resource was no longer recognized. So this approach is not good.

I will try to see if I can do something in HomeBridge Hue.
Thanks.

Aflow80, 2021-04-01:
> I have tried to delete the battery resource and after rebooting deconz the window covering resource was no longer recognized. So this approach is not good.
>
> I will try to see if I can do something [...]

In addition I have tried to change the battery level value but the rest api don’t allow me to change it.

Smanar, 2021-04-02:
Yep it s normal, sate/battery is read only field.

I hope for some information about the battery command, but if you have nothing on the tuya app about it, it s possible the device never send it.

closed 2021-04-23 by manup