dresden-elektronik/deconz-rest-plugin #4664: Direct binding

stigvi, 2021-03-28:
## Describe the question or issue you are having

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26680700)
 - deCONZ version: (2.10.04)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? APC UPS

## deCONZ Logs

## Additional context

ebaauw, 2021-03-28:
Normally the REST API plugin creates a group and sets up the binding when you pair the switch.  You can find out the group, by checking `config.group` on the `/sensors` resource.  I don't know how many bindings the switch supports, but you might need to delete the old binding, before binding to the new cluster.

I think this is the "old" model Hue dimmer (with two endpoints)?  Looks like the right binding to me.  Are you trying the _On_ of _Off_ button?  For _DimUp_ and _DimDown_ you might need to bind the _Level Control_ cluster as well.

If it supports multiple bindings (per cluster), you would get websocket notifications (when pressing a button) of `config.group` changing between the group bound by the API vs the group manually bound by you.  If `config.group` doesn't change (when pressing a button) to the group you bound, the sensor isn't acting on the new binding.

stigvi, 2021-03-29:
>I think this is the "old" model Hue dimmer (with two endpoints)? Looks like the right binding to me. Are you trying the On of Off button? For DimUp and DimDown you might need to bind the Level Control cluster as well.

Yes, this is the old Hue dimmer and it is new out of the box. Currently, I have only tried to get the on/off cluster to work.
After I have added the binding to the switch, I am able to remove this binding and it is confirmed with a success message in the bind dropbox shown in the first post. If I try to unbind a second time, it says that the binding does not exist and as expected. So it seems the binding is stored in the switch, but it is not possible to turn the light on/off.
I have also checked that the light (they are Philips Hue) are in the correct group by query for group information in deconz.

Smanar, 2021-03-29:
Someone remember the Zigbee Group limitation (if there are) ? Like the maximum device in a group, the maximum group for the same light, the maximum remote in a group, ...

ebaauw, 2021-03-29:
There is no limit to the number of lights in a group (other than the number of devices in a network).  Remember that, in Zigbee, you add a group to a light, not the other way round.  Group membership is stored on the light.  The number of groups that a light supports depends on the light firmware.  I've seen anything from 8 to 80, I think.  Same for scenes.

Also, there's no limit to the number of remotes that can be bound to the same group; the binding is stored on the remote.

rpsainio, 2021-04-23:
Well, the answer is yes and no: The binding box works in general, however it seems to me that you are better off using phoscon for configuring the dimmers. The reason is that now days they use indirect binding == the switch is bound to a dummy group the deconz listens to and then will send the command to another group (normally where the lights belong to). This way they can quite easily change the functionality. Took me awhile and zshark/wireshark to figure this out. 
Besides, if you bind the switch to the group lights belong to then the entry in phoscon UI will suddenly disapppear..

stigvi, 2021-04-23:
I suppose you mean like this? 
[image]

But doesn't this require deconz running to turn light on / off? One reason for making direct bindings was to be able to turn on light even if my home automation being down.

rpsainio, 2021-04-23:
Right, with "indirect" binding above mentioned mapping is easy as it happens in deCONZ. Well, I think you need now  days deCONZ running so that this mapping works. I have used deCONZ for some years already and I was sort of surprised see this after I just upgraded to latest and greatest. OTOH my raspi runs behind a UPS and within last years I cannot recall deCONZ stopping unless I wanted to.
As I said, the binding box still works and I think you can do direct bindings also, however you cannot use those devices in  phoscon any longer. Right now I have a mixture of direct-bindings and indirects. OTOH, for emergencies it is possible to have a switch broadcasting to all, so the lights can be switched on. OSRAM Lightify switches have this by default when initialized.

closed 2021-05-22 by github-actions[bot]

xxLeoxx93, 2022-05-02:
@rpsainio 
Did you follow any guide on how to make those bindings? I'm trying to bild a remote to three ikea fyrtur blinds but noi luck so far...

ottelo9, 2022-08-16:
same here, bind dropbox doesnt work for me.
@ebaauw could you help us?

Mimiix, 2022-08-16:
@ottelo9  I think it would be more usefull to post on the forums or join discord :)