dresden-elektronik/deconz-rest-plugin #4671: Attribute productid only delivered in webhook?

wvuyk, 2021-03-29:
## Describe the question or issue you are having
When I receive a webhook report in my plugin I get an attribute 'productid' in the response:

`{"attr":{"id":"23","lastseen":"2021-03-28T22:37Z","manufacturername":"_TYST11_wmcdj3aq","modelid":"mcdj3aq","name":"M1 Off/One light 23",**"productid":"Tuya_COVD ZM25TQ"**,"swversion":"20180727","type":"Window covering device","uniqueid":"bc:33:ac:ff:fe:68:0e:8d-01"},"e":"changed","id":"23","r":"lights","t":"event","uniqueid":"bc:33:ac:ff:fe:68:0e:8d-01"}`

This attribute is not present when getting both the lights and sensors branches. It would be helpfull if the attribute could be added here? I think `productid`  was initially added with #4301 

## Screenshots

No screenshots
## Environment

 - Host system: Any
 - Running method:Any
 - Firmware version: 26390500
 - deCONZ version: 2.10.03
 - Device: Any
 - Do you use an USB extension cable:  no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

Not needed
## Additional context

n/a

ebaauw, 2021-03-29:
Like like it's deliberately excluded from the API: From that PR:
```
    productId->setIsPublic(false); // don't show in REST-API
```
But I would think it shouldn't be visible in websocket notifications either in that case.

wvuyk, 2021-03-29:
Yes, in that case it should not appear in the websocket either. But on the other hand, with the weird modelid's and manufacturer info for Tuya devices, a 'readable' description would be nice to present to users. 
One of my users has 4 Tuya curtain motors, and 4 manufacturers and modelid's Confusing for sure.

ebaauw, 2021-03-29:
I think the API should map these to legible `manufacturername` and `modelid` values, as I do for the LIDL Livarno Lux and SIlvercrest devices.

Smanar, 2021-03-29:
Ha ?
You have it for all devices or only tuya one ?
I haven't it on my side, too old version ?

Smanar, 2021-03-29:
Ha ?
You have it for all devices or only tuya one ?
I haven't it on my side, need to trigger a special event to see them ?

ebaauw, 2021-03-29:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/a8d64006438c67c44ce258dfb16bc3e2bcd4a268/de_web_plugin.cpp#L490-L543

Patched the handling of _Model Identifier_ and _Manufacturer Name_ in the _Read Attributes Reponse_ or _Report Attributes_, search for `isLidlDevice()` or `getLidlDevice()`.

Smanar, 2021-03-29:
For the websoket.
It s disabled for config and state
```
    else if (strncmp(e.what(), "config/", 7) == 0)
    {
        ResourceItem *item = sensor->item(e.what());
        if (item && item->isPublic())
```

But not for attr
```
        ResourceItem *item = lightNode->item(e.what());
        if (item)
        {
```

If I remember the product id need to be insible for the moment, will be used in a future version.

wvuyk, 2021-03-29:
Too bad as the description is a lot better then the very cryptic names on manufacturer and modelid. If it stays in the websocket I might just adjust the plugin to grab it and show it to the user as a workaround for now?

Smanar, 2021-03-29:
As you want, I think it will not provoke problem if we let it, but not sure it will not move in the future, it s not made for that yet ^^, you can see it by error.

wvuyk, 2021-03-29:
I can use it as long it lasts, if it errors, I will put back the original part. Looking at some of my users, even if temporary it will make it easier for them to explain the name. Question closed.

Thanks!

closed 2021-03-29 by wvuyk