dresden-elektronik/deconz-rest-plugin #4679: Aqara Opple switch controls Trådfri/Osram plugs without explicit connection

stuomas, 2021-04-01:
## Describe the question or issue you are having
I just added a 6-button Aqara Opple switch in phoscon. Right after successful pairing, the first row buttons turn my Trådfri and Osram (1 each) smart plugs on/off. It is a bit worrying that it happens automatically. Can I disable that connection somewhere in phoscon?

## Environment
 - Host system: Raspberry Pi 3
 - Running method: Raspbian
 - Firmware version: 26330500
 - deCONZ version: 2.10.04
 - Device: RaspBee I

closed 2021-04-01 by close-issue-app[bot]

DJF3, 2021-04-01:
@stuomas  I have the same issue. I think it happens when you pair the switch, it also "pairs" with the nearest zigbee device it can find. Now I have a working switch but one of the buttons always controls another lamp and I don't know how to get rid of it :-)

stuomas, 2021-04-01:
@DJF3 someone instructed me in the other issue (which was not closed by a bot because of incorrect template...) that you just need to reset the switch after pairing in phoscon. Worked for me but had to pair and reset a few times.

Mimiix, 2021-04-01:
@stuomas You not using / reading the template isn't the bot's fault ;)

stuomas, 2021-04-01:
Of course not the bot's fault ;) 
But I do blame the creator of the bot. Having to include empty titles benefits nobody. Also it wasn't written anywhere that you need to include empty titles, the bot told it only afterwards. The comments in the template said I can add those paragraphs if applicable

Mimiix, 2021-04-01:
I created it ;)

However, you are stating a valid point there: to my memory it always said not to remove the headers. I'll add that :)

Mimiix, 2021-04-01:
Done
https://github.com/dresden-elektronik/deconz-rest-plugin/commit/747165f7531c6608797c48d7341b8a75d949ca28