dresden-elektronik/deconz-rest-plugin #4680: Aqara Opple switch controls Trådfri/Osram plugs without explicit connection

stuomas, 2021-04-01:
## Describe the question or issue you are having
I just added a 6-button Aqara Opple switch in phoscon. Right after successful pairing, the first row buttons turn my Trådfri and Osram (1 each) smart plugs on/off. It is a bit worrying that it happens automatically. Can I disable that connection somewhere in phoscon?

## Screenshots

## Environment

 - Host system: Raspberry Pi 3
 - Running method: Raspbian
 - Firmware version: 26330500
 - deCONZ version: 2.10.04
 - Device: RaspBee I
 - Do you use an USB extension cable: (no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which? no

## deCONZ Logs

## Additional context

Smanar, 2021-04-01:
Have you tried to re-include it ?
This feature is disabled during the device inclusion.

senilio, 2021-04-01:
This is the default behaviour of the Opples. AFAIK there are 2 ways to get rid of it.

1. Press the C button on the back of the switch.
2. Issue a device reset from deCONZ. See screenshot:
 
[image]

stuomas, 2021-04-01:
Ok, thanks! Had to reset it few times and restart gateway before it works. Interesting default action...

closed 2021-04-01 by stuomas