dresden-elektronik/deconz-rest-plugin #4681: ikea trådfri battery remotes suddenly not working in deconz in home assistant.

geltzer, 2021-04-01:
## Describe the bug
sudenly from one day to the next zigbee devices just wen unavialable, I then restarted the deconz add-in, which brought all xiaomi deveices back online, as well as all ikea bulbs and mains outlets. but not the battery remotes.

## Steps to reproduce the behavior

I have updated the deconz add-in to 6,8,0 but this has no effect, 
if I open the phoscon app all buttons are there, and seems to be avialable.
if I try to re-pair them, paring goes staight throug , but nothin works.
If i open the deconz app and view the connection map, all button are present and connected, and the deconz is claimed to have contact with them, allthoug home assistant sees nothing when the button is pressed.

so everything claims to be working but nothing is working in real life. 
everything worked until now, when it suddenly just decided to go belly up. but only the battery devices not the mains powered ikea stuff.

## Expected behavior

not loosing all ikea buttons overnight.

## Screenshots

## Environment

deconz: 6,8,0
ikea dimmer FW: 2,2,010
 

 - Host system: Raspberry Pi
 - Running method:  Home Assistent deCONZ Add-on
 - Firmware version: (26660700)
 - deCONZ version: (6,8,0)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? : NO

## deCONZ Logs

see attached

Mimiix, 2021-04-01:
Can you provide the logs?

geltzer, 2021-04-01:
The log is attached as txt file

Mimiix, 2021-04-01:
There's no file attached...

closed 2021-05-01 by github-actions[bot]