dresden-elektronik/deconz-rest-plugin #4684: Gledopto GL-B-008ZS bulb - deconz bug - brightness below 13% = off

capstan1, 2021-04-01:
## Describe the bug
Reopening issue #2640 as it didn't get resolved by updating deconz. The same issue still applies.
A summary:

- brightness <13% will turn off the bulb
- when turning the light on to for example 12% it flashes shortly, then switches off
- when dimming down to a value <13% it starts dimming and then turns off

Debugging with @ebaauw  :
- Move to Level: All brightnesses (1-255) work with all transition times (0-5s), none turns it off
- Move to Level (with On/Off): Brightness from 2-255 work with transition time 0, brightness 0 and 1 turns the light off. Brightness 0-32 turns the light off when using transition time >0
- ebaauw did a PR, sadly, nothing changed on my side

## Steps to reproduce the behavior
Try to dim these spedific gledopto bulbs to brightness below 13%

## Expected behavior

Turn it to desired brightness, instead of turning off.

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26390500)
 - deCONZ version: (2.10.04)
 - Device: RaspBee I 
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
Please see #2640

## Additional context

Mimiix, 2021-04-01:
@ebaauw Is this due to the bulb or something in deconz?

ebaauw, 2021-04-01:
As mentioned above, there seems to be a firmware bug in the bulb:
> Move to Level (with On/Off): Brightness from 2-255 work with transition time 0, brightness 0 and 1 turns the light off. Brightness 0-32 turns the light off when using transition time >0

Mimiix, 2021-04-01:
@ebaauw Alright, so there's nothing we can do here.

Sorry, but i guess we have to close it @capstan1.

capstan1, 2021-04-01:
Hi!
In the old bug report you mentioned a PR for this issue. Probably trying a workaround. Isn't anything like this possibe?

closed 2021-05-01 by github-actions[bot]