dresden-elektronik/deconz-rest-plugin #4687: Danfoss Ally temperature and offset

luka6000, 2021-04-01:
## Describe the bug

Danfoss Ally has it's own offset setting 
 0x404B Regulation SetPoint Offset on cluster 0201
This is used instead of standard zigbee 0010 Local Temperature Calibration.
The issue with this is that this non standard offset gives a bit different behavior:
1. you set 21
2. trv confirms 21 on attribute 0012
3. but the actual temperature setpoint is 21+offset 0x404B
4. current temperature reported on 0x0000 is real measured temperature and at some point in time it goes up to 21+offset

I have trvs pretty much covered and need to set offset to 25 (+2.5K) and my reported temperatures are +2.5 Celcius. So I set trv to 21, my room temperature goes up to 21 (measured externaly) but trv reports 23.5
And offset is not reported by the API sensor call - there is 0 reported since this probably is coming from attribute 0010.
## Steps to reproduce the behavior

please see above 

## Expected behavior

I expect this will work like with standard offset 0010.
1. Offset is reported by API from attribute 404B
2. temperature is reported like this: temperature = 0x0000 - 0x404B which replicates standard zigbee behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistant deCONZ Add-on
 - deCONZ version: 2.10.4
 - Device: ConBee II
 - Do you use an USB extension cable: yes

## deCONZ Logs

## Additional context

Mimiix, 2021-04-02:
@SwoopX Did you implement this one?

SwoopX, 2021-04-04:
> @SwoopX Did you implement this one?

Integrate you mean? Yes.

luka6000, 2021-04-06:
@SwoopX , I'm pretty sure this was the same with older firmware. Please mind this is about offset calibration and not external temperature or external window sensor - those are possibilities introduced with 1.08. Zigbee attribute 0010 "Local Temperature Calibration" is just a different name for temperature offset. It's "local" trv temperature not external temperature.

I can see where you are going with Device Improvement and kinda agree but deconz is returning offset: 0 while the trv is at offset 2.5. This can be considered a bug, right?

luka6000, 2021-04-06:
[image]

SwoopX, 2021-04-12:
> I'm pretty sure this was the same with older firmware.

No, it wasn't.

> Please mind this is about offset calibration and not external temperature or external window sensor - those are possibilities introduced with 1.08.

So is Regulation SetPoint Offset.

> Zigbee attribute 0010 "Local Temperature Calibration" is just a different name for temperature offset. It's "local" trv temperature not external temperature.

Yes, I know how the zigbee standard attribute offset works. However, 0x404B does not work like it, otherwise they would have presumably taken it. They both share the same offset limits, but 0x0010 adds the offset to the local temperature, which 0x404B does not.

luka6000, 2021-04-13:
> No, it wasn't.
> 
ok, you are of course right. I had a previous deconz version and modified my general.xml to allow editing 0x404B which my ha addon deconz would not allow yet. Sorry.

> Yes, I know how the zigbee standard attribute works. However, 0x404B does not work like it, otherwise they would have presumably taken it. They both share the same offset limits, but 0x0010 adds the offset to the local temperature, which 0x404B does not.

sure. It's probably a matter of different approach to "local temperature" measurement so they offset setpoint instead. There is some algorithm that modify local temperature measurement depending on "eTRV orientation" attribute. I can see this has a huge impact.
But regardless of you decision on the setpoint offset correction on local temp, can you please make it visible under "offset" API attribute? If you want to make it mean exactly the same as 0x0010 you should probably make it with minus sign.

SwoopX, 2021-04-13:
No worriesm I'm already on it. I "only" see some strange effects when reading the attribute from the GUI (the value is odd) and I currently have no idea where that comes from. Setting it already works correctly. Once that's done, you can use the offset as desired but if that attribute doesn't amend the local temperature, I'm afraid you need to do that in your middleware. My Danfoss doesn't correct any values.

luka6000, 2021-04-13:
ok, great. Thank you!
And do you plan to make the value as 0-0x404B (zero minus 0x404B) to make it a 'temperature offset" rather than original "setpoint offset"?

closed 2021-05-22 by manup