dresden-elektronik/deconz-rest-plugin #4688: Set the "permit join" is no more possible using API

Smanar, 2021-04-02:
## Describe the bug
No more possible to use the GUI to set the permit join

## Steps to reproduce the behavior
- Go on deconz
- Edit/permit join
- Set a value
- Press "set"

>22:12:25:250 send permit join, duration: 0

Is displayed on logs

## Expected behavior
A valid value for permit join duration

## Screenshots

## Environment
 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: (26350500)
 - deCONZ version: (2.10.04 / 14/03/2021)
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
>22:12:25:250 send permit join, duration: 0

## Additional context

Mimiix, 2021-04-02:
@SwoopX Confirmed it.

@manup  can you adress this asap?

manup, 2021-04-02:
I think this is broken since a few years now. The dialog itself does set Permit Join, but this is not communicated to the plugin which will overwrite the changed done from the GUI (the dialog predates the mechanism in the plugin).

I think the best fix will be the dialog sending the event to start/stop Permit/Join to the plugin and let the plugin handle the rest, since the mechanism is already there and working.

I schedule this for  2.11.x series but not 2.11.0 (which is already behind).

ebaauw, 2021-04-02:
Does that mean that setting _Permit Join_ in the GUI will trigger the API to create new resources?

Smanar, 2021-04-02:
And if the user disable the api plugin ?
"Few years" ? I m sure having already use it, to include device without using the API.

manup, 2021-04-02:
On  occasions it worked but the Permit Join timer in the plugin was always free to trigger and close the network again.

>Does that mean that setting *Permit Join* in the GUI will trigger the API to create new resources?

Good question the `permitJoin.cpp` could be triggered standalone without creating new resources, but I think it would be more useful to also allow creating resources. At least in the past it was often confusing for users enable joining and wondering why they did see new nodes in the GUI but no new devices via API.

ebaauw, 2021-04-02:
> I think it would be more useful to also allow creating resources

Agreed, that was the reason for my question, sorry if that wasn't clear.

solvisit, 2021-07-07:
Any news on this bug? I cant use permitjoin from RESTAPI or from GUI, while I can "Add new sensor" from phoscon. It is caused by this bug?

rasssta, 2021-08-08:
I'm having this issue when using v2.12.3-beta but it works fine in 2.11.5.

manup, 2022-10-07:
The *Edit &rarr; Permit Join* dialog was replaced in [v2.13.0-beta](https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.13.0-beta) with the *Panels &rarr; Control* panel.

[image]

closed 2022-10-07 by manup