dresden-elektronik/deconz-rest-plugin #4693: Device name

andwari, 2021-04-03:
### Device
    Product name: Woox R7049 Smoke Alarm
    Manufacturer: _TZE200_aycxwiau
    Model identifier: TS0601
    Device type:
        Other: Smoke Alarm

Product page: https://wooxhome.com/products-c10/security-c6/r7049-smoke-alarm-single-unit-p70
### Screenshots

### Node Info
[image]

### Basic Cluster
[image]
[image]

### Groups
[image]

### Scenes
[image]

### Tuya specific Cluster
[image]

### OTAU
[image]
[image]

### Time
[image]

valentingc, 2021-04-19:
I'd also love to see this device being supported. Seems this is the only Zigbee based smoke detector that has great availability where I live, and it is quite affordable.

air2, 2021-07-02:
Could this issue be reopened, I would love to see support for this device

Mimiix, 2021-07-02:
Sure

valentingc, 2021-08-29:
@Mimiix Could this issue please be reopened?

Mimiix, 2021-08-29:
@valentingc sure.

Please keep it open yourselves by bumping it

Smanar, 2021-08-29:
Hello, do you have a linux machine with full OS to test code modifications ?

andwari, 2021-09-01:
@Smanar I have a linux machine and can test some code modifications

Smanar, 2021-09-02:
So it s the first tuya smoke sensor, so I m not sure for result

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch woox_smoke https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Then re include the device.

andwari, 2021-09-02:
@Smanar Thank you very much. It seems that the sensor is now recognized by deconz as fire sensor. Though it does not appear in the phoscon webapp, it seems that the sensor is accessible via the REST API (which is more or less all I need). Once again, thank you very much

aaronerhart, 2021-09-03:
@Smanar 
Sorry I am new here. Will it be possible in the future to integrate the R7049 also via the phoscon app?

Smanar, 2021-09-03:
@andwari so it work on first try ? I can make the PR ?

@aaronerhart ATM we are realy late on PR validation, so IDK when it will be available on deconz, and make phoscon use it is another thing.

You can ask on phoscon github when the device will be in deconz.

valentingc, 2021-09-05:
@Smanar Thanks for your awesome work! Looking forward to the PR, I was waiting to buy the device until there was deconz support.

@andwari Could you maybe if you get to it test-trigger the smoke detector to check whether the fire state is also correctly mapped in the REST API? Seems to me like we should test that too for such kind of a device.

andwari, 2021-09-05:
@valentingc You are absolutely right. I will test it as soon as possible but maybe not today (or my neighbors will kill me ;) ).

closed 2021-10-01 by manup

timvdsm, 2021-12-24:
Hi, I can see the status of the Woox R7049 smoke alarm, however I don't see the battery level. Is it possible to add this also, as this is pretty interesting information.

Smanar, 2021-12-24:
This device don't have battery level, but it need to have a lowbattery notification in the json ?

timvdsm, 2021-12-25:
> This device don't have battery level, but it need to have a lowbattery notification in the json ?

Indeed, that’s what I wanted to ask, to know when the battery is low. The information should be there, because it’s available in other zigbee platforms (as dell as link quality). Would be great to have this additional information in deconz.

Smanar, 2021-12-25:
>because it’s available in other zigbee platforms

Wich one ?
My reference is z2m, ant they haven't it too https://www.zigbee2mqtt.io/devices/R7049.html

timvdsm, 2021-12-25:
> Wich one ? My reference is z2m

Indeed z2m, here you have smoke, battery_low, linkquality. It's the battery_low and linkquality which I absolutely want to add in Home Assistant. However it only exposes for me if smoke is detected or not.

Smanar, 2021-12-25:
>Or is there any way to get the battery_low and linquality?

Link quality are not exposed in the API.
But the low battery need to be in the JSON, can you take a look in phoscon / hep / api information to see the complete device json ?

timvdsm, 2021-12-25:
Hi @Smanar 

        "2": {
            "config": {
                "on": true,
                "reachable": true
            },
            "ep": 1,
            "etag": "790b0b696d9d4dc5305aa5d4e7b11d09",
            "lastannounced": "2021-12-23T22:49:20Z",
            "lastseen": "2021-12-25T09:38Z",
            "manufacturername": "_TZE200_aycxwiau",
            "modelid": "TS0601",
            "name": "Fire 2",
            "state": {
                "fire": null,
                "lastupdated": "none",
                "lowbattery": null
            },
            "type": "ZHAFire",
            "uniqueid": "5c:02:72:ff:fe:90:46:9a-01-ef00"
        },

So indeed ... I see  mentioned "lowbattery": null ... which I suppose will be active when the battery is low. Any idea to get this into Home Assistant? I installed the deconz integration, which connects to my Phoscon GW. And it automatically exposes the smoke detector as a binary sensor in Home Assistant. Now I just need an additional one for the battery low indication. 

And by the way ... many thanks for your hard work on this. This really makes a difference for many people like me to have a life saving device locally integrated in Home Assistant! It's absolutely appreciated!

Smanar, 2021-12-25:
Ha, it s perhaps because the value is "null", for HA the value need to be true or false ?
If @Robban can confirm ?

Kane610, 2021-12-26:
Home assistant integration does not generate sensors for low_battery. And I guess the logic would expect it to be True/False rather than null.

Smanar, 2021-12-26:
Ha yes right, there is something bugged
```
            else if (sensor.fingerPrint().hasInCluster(TUYA_CLUSTER_ID))
            {
                clusterId = clusterId ? clusterId : TUYA_CLUSTER_ID;
                sensor.addItem(DataTypeBool, RStateLowBattery)->setValue(false);
            }
```

timvdsm, 2021-12-26:
> Home assistant integration does not generate sensors for low_battery. And I guess the logic would expect it to be True/False rather than null.

Any way to get it in HA through a workaround?

timvdsm, 2021-12-26:
> Right, and there is something bugged
> 
> ```
>             else if (sensor.fingerPrint().hasInCluster(TUYA_CLUSTER_ID))
>             {
>                 clusterId = clusterId ? clusterId : TUYA_CLUSTER_ID;
>                 sensor.addItem(DataTypeBool, RStateLowBattery)->setValue(false);
>             }
> ```
> 
> It need to be false at startup

Does this mean, it could be solved? 😁

Smanar, 2021-12-26:
>Does this mean, it could be solved? 😁

Not realy ^^.
ATM update can be only done with DDF, but it will not change something for you as you need to make something in HA to manage the lowbattery state.

Electronisse, 2022-01-06:
> @valentingc You are absolutely right. I will test it as soon as possible but maybe not today (or my neighbors will kill me ;) ).

Did you get around to testing it?

I tried to add mine to deconz today.
The smoke detector shows up and everything, so i tried to trigger it with some smoke.
The state/fire does never change for me, its false when the device is first added, and becomes null after restart of deconz.
This is the response i get, both with and without smoke:

`
{
    "config": {
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "18829250289984f542f34f4634d21415",
    "lastannounced": null,
    "lastseen": "2022-01-06T22:51Z",
    "manufacturername": "_TZE200_aycxwiau",
    "modelid": "TS0601",
    "name": "Fire 50",
    "state": {
        "fire": null,
        "lastupdated": "none",
        "lowbattery": null
    },
    "type": "ZHAFire",
    "uniqueid": "bc:33:ac:ff:fe:26:e2:0d-01-ef00"
}
`

Kane610, 2022-01-15:
> > Home assistant integration does not generate sensors for low_battery. And I guess the logic would expect it to be True/False rather than null.
> 
> Any way to get it in HA through a workaround?

The issue is that right now all sensor devices would get a low_battery entity if default values is None since its not possible to differentiate between devices supporting reporting low_battery with devices not supporting it

km8742, 2022-01-16:
> > @valentingc You are absolutely right. I will test it as soon as possible but maybe not today (or my neighbors will kill me ;) ).
> 
> @andwari Did you get around to testing it?
> 
> I tried to add mine to deconz today. The smoke detector shows up and everything, so i tried to trigger it with some smoke. The state/fire does never change for me, its false when the device is first added, and becomes null after restart of deconz. This is the response i get, both with and without smoke:
> 
> `{ "config": { "on": true, "reachable": true }, "ep": 1, "etag": "18829250289984f542f34f4634d21415", "lastannounced": null, "lastseen": "2022-01-06T22:51Z", "manufacturername": "_TZE200_aycxwiau", "modelid": "TS0601", "name": "Fire 50", "state": { "fire": null, "lastupdated": "none", "lowbattery": null }, "type": "ZHAFire", "uniqueid": "bc:33:ac:ff:fe:26:e2:0d-01-ef00" }`

Same problem here, I was able to register 3 smoke detectors R7049 - they're showing up in the websocket with updates. Not showing up in Phoscon webfrontend and not showing an alarm on trigger.

One of three showing fire:null - the other fire:false if I go over the webapi /sensors page.

Smanar, 2022-01-16:
Even if you restart deconz ? It seem it need to use a default value for this kind of sensor

```
            item = sensor.addItem(DataTypeBool, RStateFire);
            item->setValue(false);
```

`"lastseen": "2022-01-06T22:51Z"`

It haven't disconnecvted ?

km8742, 2022-01-16:
> Even if you restart deconz ? It seem it need to use a default value for this kind of sensor
> 
> ```
>             item = sensor.addItem(DataTypeBool, RStateFire);
>             item->setValue(false);
> ```
> 
> `"lastseen": "2022-01-06T22:51Z"`
> 
> It haven't disconnected ?

After a reboot all three fire sensors show fire:null and lowbattery:null.

In the websocket I get events, lastseen is updated.

In the API I have config: on: true / reachable: true.

Smanar, 2022-01-16:
>In the websocket I see events, lastseen is updated.

for third ?

km8742, 2022-01-16:
> > In the websocket I see events, lastseen is updated.
> 
> for third ?

For all of them.
I use the websocket to read all events.

All three smoke sensors give events - so they have to be registered, right? But no event is fired when the smoke sensor goes off.

Sample (uniqueID changed):
`{"attr":{"id":"12","lastannounced":null,"lastseen":"2022-01-16T14:10Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 12","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"Sensor01"},"e":"changed","id":"12","r":"sensors","t":"event","uniqueid":"Sensor01"}

{"attr":{"id":"11","lastannounced":"2022-01-15T11:25:00Z","lastseen":"2022-01-16T14:10Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"Sensor02"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"Sensor02"}

{"attr":{"id":"13","lastannounced":null,"lastseen":"2022-01-16T14:10Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 13","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"Sensor03"},"e":"changed","id":"13","r":"sensors","t":"event","uniqueid":"Sensor03"}
`

Smanar, 2022-01-16:
>But no event is fired when the smoke sensor goes off

I think you have something displayed in HA if the device become unreachable (HA probably use config/reachable based on last seen) ?

But after some time, you need to have state/fire = False ? And those devices don't have a "test" button ?

km8742, 2022-01-17:
I'm not using HA - I'm only using deconz installed on a headless PI.
I'm using the websocket service to check for events (fire:true and so on).
I can test the smoke alarms by long pressing a button, will try but I don't think there will be any difference to a real alarm.

Smanar, 2022-01-18:
Just to be sure you haven't a "light" entry for this device ?

I m seing 2 possibles issues.
- the device don't trigger event at all and it need (new firmware with exotic pairing ...)
- the device don't trigger event at all and it's normal (realy lazy one)

If you test the smoke alarm, the device will obligatory send an event, if you still have state/fire = null and lastupdated = null, I think it s the first one. Can you check if "lastseen" update something too.

km8742, 2022-01-18:
I've triggered the manual alarm:
16:40	Testart 
`	API Output: {"config":{"on":true,"reachable":true},"ep":1,"etag":"f8dec922b7a7c034901957e8d70613d9","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-18T16:39Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","state":{"fire":null,"lastupdated":"none","lowbattery":null},"type":"ZHAFire","uniqueid":"Sensor01"}
`
16:46	Testalarm
	`API Output: {"config":{"on":true,"reachable":true},"ep":1,"etag":"2bbb9db03fcdc73b54915132ae2d943a","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-18T16:45Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","state":{"fire":null,"lastupdated":"none","lowbattery":null},"type":"ZHAFire","uniqueid":"Sensor01"}
`

17:29 After
`{"config":{"on":true,"reachable":true},"ep":1,"etag":"e6f0261dd063c099a32ac02fbbcef6aa","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-18T17:28Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","state":{"fire":null,"lastupdated":"none","lowbattery":null},"type":"ZHAFire","uniqueid":"Sensor01"}
`
Events from websocket:
```
{"attr":{"id":"11","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-18T16:39Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"Sensor01"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"Sensor01"}
{"attr":{"id":"11","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-18T16:40Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"Sensor01"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"Sensor01"}
{"attr":{"id":"11","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-18T16:41Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"Sensor01"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"Sensor01"}
{"attr":{"id":"11","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-18T16:42Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"Sensor01"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"Sensor01"}
{"attr":{"id":"11","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-18T16:43Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"Sensor01"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"Sensor01"}
{"attr":{"id":"11","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-18T16:44Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"Sensor01"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"Sensor01"}
{"attr":{"id":"11","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-18T16:45Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"Sensor01"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"Sensor01"}
{"attr":{"id":"11","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-18T16:46Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"Sensor01"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"Sensor01"}
{"attr":{"id":"11","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-18T16:47Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"Sensor01"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"Sensor01"}
{"attr":{"id":"11","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-18T16:48Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"Sensor01"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"Sensor01"}
{"attr":{"id":"11","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-18T16:49Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"Sensor01"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"Sensor01"}
{"attr":{"id":"11","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-18T16:50Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"Sensor01"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"Sensor01"}
{"attr":{"id":"11","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-18T16:51Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"Sensor01"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"Sensor01"}
[... cut ...]
Currently I dont care about lastseen (my code currently does not check this) but this value is getting updated.
The most important thing is that I get an event when the fire alarm goes off (similar to presence:true with ikea motion detectors) so my software can trigger an alarm - this is triggered trough the websocket - because it's much faster, not by pinging the API sensor values.

I could not find any firmware updates on the site of woox.

So do you think this is not possible without the app & gateway of woox or is this a work in progress on the side of deconz?

Thanks a lot for your time!

Smanar, 2022-01-18:
I think you will not like what I m going to say, but I realy need some logs.

With flag "info" and "info_l2".

All tuya request are logged so we will be able to see all requests from the device. It make request because it update the "lastseen", it 's a good news, but it don't update the battery or the fire detection, so I don't see what it can be, and 1 report every minut is a lot for me, I don't think it's good for battery.

km8742, 2022-01-18:
Is it enough to start deconz service with the following flags?
--dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-http=1

Smanar, 2022-01-19:
Just `--dbg-info=2`
And if you are headless `-platform minimal`

km8742, 2022-01-19:
That's a lot of data. What time window of log file would you need? And how can I send it to you?

Smanar, 2022-01-20:
If you use the button to simulate the fire, I just need the log during the procedure (so 1/2 min logs)
If you can filter, I just need log with "Tuya debug"

And you can use the hotst you want, or just make a c/c here if you can find the tuya logs made by this device specially.

km8742, 2022-01-22:
First off I have seen that two of the three smoke sensors have the same mac address - is that possible?
The only difference in the uniqueid is the last 10 digits - and as far I know those are additionally added right?

Triggered a test alarm with the button on 13:40

```
13:35:28:463 No Tuya productId entry found for manufacturername: _TZ3000_nwpihrkp
13:35:28:680 Tuya debug 7 : Missing manufacture name for 0xSENSOR01
13:35:28:682 No Tuya productId entry found for manufacturername: _TZ3000_nwpihrkp
13:35:28:683 No Tuya productId entry found for manufacturername: _TZ3000_nwpihrkp
13:35:28:684 No Tuya productId entry found for manufacturername: _TZ3000_nwpihrkp
13:35:28:686 Tuya debug 7 : Missing manufacture name for 0xSENSOR01
13:35:28:687 No Tuya productId entry found for manufacturername: _TZ3000_nwpihrkp
13:35:28:688 No Tuya productId entry found for manufacturername: _TZ3000_nwpihrkp
13:35:28:689 No Tuya productId entry found for manufacturername: _TZ3000_nwpihrkp
13:35:28:691 Tuya debug 7 : Missing manufacture name for 0xSENSOR02
13:35:28:694 Tuya debug 7 : Missing manufacture name for 0xSENSOR02
13:35:28:703 Tuya debug 7 : Missing manufacture name for 0xSENSOR03
13:35:28:706 Tuya debug 7 : Missing manufacture name for 0xSENSOR03
13:35:28:720 Tuya debug 7 : Missing manufacture name for 0xSENSOR04
13:35:28:723 Tuya debug 7 : Missing manufacture name for 0xSENSOR04
13:36:07:654 Tuya debug Request : Address 0xSENSOR02, Endpoint 0x01, Command 0x01, Payload 001e0801000101
13:36:07:654 Tuya debug 4 : Address 0xSENSOR02 Payload 001e0801000101
13:36:07:654 Tuya debug 5 : Status: 0 Transid: 30 Dp: 264 (0x01,0x08) Fn: 0 Data 1
13:36:15:580 Tuya debug Request : Address 0xSENSOR02, Endpoint 0x01, Command 0x01, Payload 001f0904000101
13:36:15:580 Tuya debug 4 : Address 0xSENSOR02 Payload 001f0904000101
13:36:15:580 Tuya debug 5 : Status: 0 Transid: 31 Dp: 1033 (0x04,0x09) Fn: 0 Data 1
13:36:17:718 Tuya debug Request : Address 0xSENSOR02, Endpoint 0x01, Command 0x01, Payload 00200801000100
13:36:17:718 Tuya debug 4 : Address 0xSENSOR02 Payload 00200801000100
13:36:17:718 Tuya debug 5 : Status: 0 Transid: 32 Dp: 264 (0x01,0x08) Fn: 0 Data 0
13:40:04:694 Tuya debug Request : Address 0xSENSOR02, Endpoint 0x01, Command 0x01, Payload 00210801000101
13:40:04:694 Tuya debug 4 : Address 0xSENSOR02 Payload 00210801000101
13:40:04:694 Tuya debug 5 : Status: 0 Transid: 33 Dp: 264 (0x01,0x08) Fn: 0 Data 1
13:40:12:538 Tuya debug Request : Address 0xSENSOR02, Endpoint 0x01, Command 0x01, Payload 00220904000101
13:40:12:538 Tuya debug 4 : Address 0xSENSOR02 Payload 00220904000101
13:40:12:538 Tuya debug 5 : Status: 0 Transid: 34 Dp: 1033 (0x04,0x09) Fn: 0 Data 1
13:40:14:674 Tuya debug Request : Address 0xSENSOR02, Endpoint 0x01, Command 0x01, Payload 00230801000100
13:40:14:674 Tuya debug 4 : Address 0xSENSOR02 Payload 00230801000100
13:40:14:674 Tuya debug 5 : Status: 0 Transid: 35 Dp: 264 (0x01,0x08) Fn: 0 Data 0
```

Smanar, 2022-01-22:
>First off I have seen that two of the three smoke sensors have the same mac address - is that possible?

No, You are sure of you ?

>The only difference in the uniqueid is the last 10 digits - and as far I know those are additionally added right?

10 ?
The unique ID is composed buy the mac adress - endpoint (2 digit)- cluster (4 digit)

Ok so from logs, it can be
>13:36:07:654 Tuya debug 5 : Status: 0 Transid: 30 Dp: 264 (0x01,0x08) Fn: 0 Data 1

With data = 1 then 0. And this request is missing in code, but it s the "test" feature command

>13:36:15:580 Tuya debug 5 : Status: 0 Transid: 31 Dp: 1033 (0x04,0x09) Fn: 0 Data 1

With data always = 1. And this request is missing too, and have no clue what is it.

On the code (at the base) this device need to use (0x01 0x01) as command for fire detection.

So bad luck, the device use different request for "fire" and "test" ....

On your logs you don't have a periodic "Tuya debug" for this device ? I think you have a 0x01 0x01 but with value 0 (To mean no fire)
If you can see it, I have found the issue and all is normal.

km8742, 2022-01-22:
Sorry - I just saw that the last two chars after the ':' also belong to the mac address - so no dupes there.

Searched for all messages containing 0x01 and this MAC Address in the log:
`14:03:35:591 Tuya debug Request : Address 0xSENSOR01, Endpoint 0x01, Command 0x24, Payload 00ae
15:04:19:962 Tuya debug Request : Address 0xSENSOR01, Endpoint 0x01, Command 0x24, Payload 00af
16:05:04:198 Tuya debug Request : Address 0xSENSOR01, Endpoint 0x01, Command 0x24, Payload 00b0
16:08:46:298 [WARNING] - Missing cluster in sensor fingerprint: 0xSENSOR01 - 0xE36F (TS0601), endpoint: 0x01, cluster: 0x000A, payload: 0700, zclSeq: 95
17:05:48:431 Tuya debug Request : Address 0xSENSOR01, Endpoint 0x01, Command 0x24, Payload 00b1
18:04:40:470 Tuya debug Request : Address 0xSENSOR01, Endpoint 0x01, Command 0x01, Payload 000d0e04000102
18:06:32:617 Tuya debug Request : Address 0xSENSOR01, Endpoint 0x01, Command 0x24, Payload 00b2
19:07:16:866 Tuya debug Request : Address 0xSENSOR01, Endpoint 0x01, Command 0x24, Payload 00b3
20:08:01:135 Tuya debug Request : Address 0xSENSOR01, Endpoint 0x01, Command 0x24, Payload 00b4
`

Smanar, 2022-01-23:
It's not "0xSENSOR02" from your previous log ?

>18:04:40:470 Tuya debug Request : Address 0xSENSOR01, Endpoint 0x01, Command 0x01, Payload 000d0e04000102

Payload 000d0e04000102 > 00 0d 0e04 00 01 02 So it's the Command 0x04 0x0e
It's the battery level
```
0 2 above 2,8v
0 1 above 2.6v
0 0 at 2.6v and below (beeps ocassionally)
```
I m wrong I m using 0x02 so I need to correct that
And I m wro,g too for the fire detection ...

BTW if you have not make mistake with the device (if it s realy the 0xSENSOR01) I can correct the battery level and use it as heartbeat.
So everytime you will see a battery report, it will update those 2 fields
```
            "lastupdated": "none",
            "lowbattery": null
```

The last seen will be corrected by this PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5675 if manup is agree to add the tuya cluster.

and ofc I can correct the fire detection for fire = null (I m using 0x01 0x01 and need to use 0x04 0x01)

Now the big question, have you a linux machine with full OS to test code change ?

km8742, 2022-01-23:
I have not extracted the sensor data of the one which I did the alarm test - but it all are the same sensor.

I'm running the USB Stick with a Raspberry PI - deconz is installed trough the APT Release path. 
I could theoretically set up a test machine to test only the fire detector so my setup won't get lost on the PI.

Smanar, 2022-01-23:
Yes, you can use a test machine, else don't worry, the procedure don't impact your database and change only 1 file, so you can make a backup of it **in another folder.**

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz (The code is based on realy last one):

```
sudo apt install deconz-dev
git clone --branch woox_fix https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry, it can freeze during compilation, my pi 3 do, so remove "-j2"

No need to re-include the device.

km8742, 2022-01-23:
Did it and worked perfectly! The fire test triggered a websocket message!
Still not showing up in the Phoscon app though.

Do you need any logs for further data?

Smanar, 2022-01-23:
For information even the test feature now trigger a fire detection.
If all is fine for you, will make the PR

Kane610, 2022-01-23:
Can you share websocket examples of these different events @Smanar

Smanar, 2022-01-23:
I don't t have the device but it s like all other sensor : state/fire = true

km8742, 2022-01-24:
Here are the websocket events with the fire alarm:
```
{"attr":{"id":"11","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-23T15:21Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","swversion":null,"type":"ZHAFire","uniqueid":"SENSOR"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"SENSOR"}
{"e":"changed","id":"11","r":"sensors","state":{"fire":true,"lastupdated":"2022-01-23T15:22:03.284","lowbattery":null},"t":"event","uniqueid":"SENSOR"}
{"e":"changed","id":"11","r":"sensors","state":{"fire":false,"lastupdated":"2022-01-23T15:22:19.579","lowbattery":null},"t":"event","uniqueid":"SENSOR"}
{"attr":{"id":"11","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-23T15:22Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"SENSOR"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"SENSOR"}
{"attr":{"id":"11","lastannounced":"2022-01-18T16:38:46Z","lastseen":"2022-01-23T15:23Z","manufacturername":"_TZE200_aycxwiau","modelid":"TS0601","name":"Fire 11","productid":"Tuya_OTH R7049 Smoke Alarm","swversion":null,"type":"ZHAFire","uniqueid":"SENSOR"},"e":"changed","id":"11","r":"sensors","t":"event","uniqueid":"SENSOR"}

```

Will the rest changes get discarded if I do an apt update or will these updates be included in the next release?

Smanar, 2022-01-24:
>Will the rest changes get discarded if I do an apt update or will these updates be included in the next release?

Good question, I will ask to mimiix, but you can can check if the PR is validated or not before making an update.

km8742, 2022-01-24:
Will do - thanks!

Electronisse, 2022-01-28:
Downloaded the woox_fix branch today and tested again.
This now works for me as well, all the way through to home assistant and node red.
Thanks for great work :)

BUT the fire state is inverted for me, shows true in postman and home assistant when there is no smoke, goes false when smoke is detected.

Atleast home assistant seems to expect the opposite of this, true when smoke detected.

Smanar, 2022-01-29:
Arf too, late ...
The PR is merged.

The issue happen only on fire detection or the "test" button too ?

Electronisse, 2022-01-30:
Ah damn, didnt have opportunity to try before now.

I checked, the issue is only with fire detection, and not with the test button.
After press test button, the flag is true while alarm/test is going, and flag becomes false after test is finished.
But real smoke sets flag false when smoke detected, and flag true when smoke is gone.

I tried to fix it locally by flipping "bool fire = (data==0) ? false:true;" to "true:false" in tuya.cpp before compiling. That fixed it for me.

Smanar, 2022-01-30:
Have made a short PR