dresden-elektronik/deconz-rest-plugin #4695: Legrand wireless double switch

kurcsics, 2021-04-04:
## Device

- Product name: Legrand 7 522 87    / Legrand ZLE19
- Manufacturer: Legrand
- Model identifier: Double gangs remote switch
- Device type : Switch

Legrand Double Switch is recognized as a Single Switch and is not listed in Switch-Editor

## Screenshots

### Basic
[image]

### Other clusters that are not mentioned above
[image]
[image]
[image]

kurcsics, 2021-04-04:
Log Switch 1 ON
[image]

Log press Switch 2 ON
[image]

Smanar, 2021-04-04:
Hello, I think it s because this device is not coreclty supported in phoscon.
You don't have third app to be sure ?
Or just with looking logs , the line starting by [info] - button X00Y ................................
I think you will have different value depending from the gangs used ?

If yes, this issue can me moved to phoscon github.

kurcsics, 2021-04-05:
Hello, Thank you, it works with ioBroker.
Thanks!

kvanisterdael, 2021-04-07:
Hello, how did you manage to discover the Legrand switch in Deconz?
I have several Zigbee devices of Legrand, but my Raspbee can't seem to find any of them.

I read somewhere that the Zigbee channel needs to be 11; can you confirm this?

Smanar, 2021-04-07:
Yep, for the moment you NEED channel 11.

You need a real long press, around 8/10s, you need to keep pressed at first blink.
Here you have the procedure for the Zigate, but it s the same https://github.com/pipiche38/Domoticz-Zigate-Wiki/blob/master/en-eng/Corner_Legrand-Netatmo.md#pairing-process

kvanisterdael, 2021-04-08:
Ok, so I switched to channel 11 and now I can detect the legrand devices.

I have a wireless switch that internally has 2 buttons (top - bottom), but if I click them, they both give buttonevent "1002"...
Probably it is not supported yet by Deconz?

[image]

Smanar, 2021-04-08:
It s normal, you have the lastest version.
On the last versions, Legrand have removed the "on"/ "off"  and replaced it with a "toogle", so now you have the command for the 2 positions.
So you have the same command for short press up / short press down, but still have 2 differents commands for long press.
In bonus you have the double short press from my memory (it s the old "off")

Smanar, 2021-04-08:
Oups, for the double version, the toogle is not implemented.

On your deconz install, search for the file /usr/share/deCONZ/devices/button_maps.json

Search the part for the device "Double gangs remote switch" (but it s not the one on your last capture)
And add this line

`[1, "0x01", "ONOFF", "TOGGLE", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Toggle"],`

kvanisterdael, 2021-04-22:
> It s normal, you have the lastest version.
> On the last versions, Legrand have removed the "on"/ "off" and replaced it with a "toogle", so now you have the command for the 2 positions.
> So you have [...]

You're right; I have following buttonevents:

1. short press top / bottom: 1002
2. double click top / bottom: 2002
3. long press top: 1001
4. long press top release: 1003
5. long press bottom: 2001
6. long press bottom release: 2003

This means I can't assign different values to the top and the bottom for the short press and double click... Too bad, because this limits the functionalities.

kurcsics, 2021-04-22:
The trouble is I can't connect with a lamp under Phoscon, only with a third part solution.

Smanar, 2021-04-23:
>This means I can't assign different values to the top and the bottom for the short press and double click... Too bad, because this limits the functionalities.

Yeah, you are not the first with this reaction.
BTW have you needed the json modification ?

`[1, "0x01", "ONOFF", "TOGGLE", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Toggle"],`

To know if i can add it in the code ?

@kurcsics try to make a request on phoscon github ? https://github.com/dresden-elektronik/phoscon-app-beta

kvanisterdael, 2021-04-23:
The json modification was not needed.

closed 2021-05-22 by github-actions[bot]