dresden-elektronik/deconz-rest-plugin #4699: NAMRON AS 4512729

tore1980, 2021-04-04:
## Device

- Product name: NAMRON ZIGBEE 2 CHANNEL SWITCH K4.
- Manufacturer: NAMRON AS.
- Model identifier: 4512729.
- Device type : 
  - Switch

## Screenshots
[image]
[image]
[image]

[image]

SwoopX, 2021-04-04:
Already supported. Please add your model ID here in `button_maps.json`

```
"sunricherMap": {
            "vendor": "Sunricher",
            "doc": "Wireless switches from Sunricher, Namron and EcoDim",
            "modelids": ["ED-10010", "ED-10011", "ED-10012", "ED-10013", "ED-10014", "ED-10015", "ZG2833K8_EU05", "ZG2835", "4512700", "4512701", "4512702", "4512703", "4512705", "4512714", "4512719", "4512720", "4512721", "4512722"],
            "map": [
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "Move up (with on/off)"],
                [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "1", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "Move down (with on/off)"],
                [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "1", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x02", "ONOFF", "ON", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x02", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_3", "S_BUTTON_ACTION_HOLD", "Move up (with on/off)"],
                [1, "0x02", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_3", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x02", "ONOFF", "OFF", "0", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x02", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "1", "S_BUTTON_4", "S_BUTTON_ACTION_HOLD", "Move down (with on/off)"],
                [1, "0x02", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "1", "S_BUTTON_4", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x03", "ONOFF", "ON", "0", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x03", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_5", "S_BUTTON_ACTION_HOLD", "Move up (with on/off)"],
                [1, "0x03", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_5", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x03", "ONOFF", "OFF", "0", "S_BUTTON_6", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x03", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "1", "S_BUTTON_6", "S_BUTTON_ACTION_HOLD", "Move down (with on/off)"],
                [1, "0x03", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "1", "S_BUTTON_6", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x04", "ONOFF", "ON", "0", "S_BUTTON_7", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x04", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_7", "S_BUTTON_ACTION_HOLD", "Move up (with on/off)"],
                [1, "0x04", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_7", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x04", "ONOFF", "OFF", "0", "S_BUTTON_8", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x04", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "1", "S_BUTTON_8", "S_BUTTON_ACTION_HOLD", "Move down (with on/off)"],
                [1, "0x04", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "1", "S_BUTTON_8", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"]
            ]
        },
```

tore1980, 2021-04-06:
@SwoopX Thank you for fast response, and great that the device is supported.
Im using pi / homeassistent with deCONZ add-on, do you know where i can find the file or the path for "button_maps.json"

SwoopX, 2021-04-12:
No, sorry. I got no clue of HA.

closed 2021-05-09 by manup