dresden-elektronik/deconz-rest-plugin #4704: Aqara TVOC Air Quality Monitor (VOCKQJK11LM)

redphx, 2021-04-05:
## Device

- Product name: [Aqara TVOC Air Quality Monitor](https://www.aqara.com/cn/productDetail/TVOC-air-quality-monitor)
- Manufacturer: LUMI
- Model identifier: lumi.airmonitor.acn01
- Device type :
  - Sensor

## Screenshots

[image]

[image]

### Basic

[image]

### Identify

[image]

### Temperature measurement

[image]

### Relative humidity measurement

[image]

### Analog Input (Basic)

[image]

### Power Configuration

[image]

apper13, 2021-05-27:
I would highly appreciate re-opening this ticket. This device is really interesting and I would love to see it in deconz! It is newly available directly in Europe (https://www.amazon.de/dp/B094N7X7NH)

redphx, 2021-05-28:
Updated screenshots with values (had to use the Reset Node feature first + press the Pair button)

redphx, 2021-05-28:
Not sure where this `ppb (tvoc)` came from:   
  
[image]

00lex, 2021-06-01:
im interested too! :)

lisanet, 2021-06-01:
I'd love to see this too.

SwoopX, 2021-06-01:
Can anyone please add the required screenshot of the node's clusters?

redphx, 2021-06-02:
> Can anyone please add the required screenshot of the node's clusters?

Sorry I missed that. Added to the first post.

SwoopX, 2021-06-05:
Thanks. Would anybody also be able to compile a test version of the plugin?

rasssta, 2021-06-15:
> Thanks. Would anybody also be able to compile a test version of the plugin?

Could you please elaborate a bit? :-)

rgers, 2021-06-15:
> Thanks. Would anybody also be able to compile a test version of the plugin?

I just got the sensor and I'm fairly confident I'll be able to compile a test version. Which branch?

SwoopX, 2021-06-15:
It's branch `tvoc` in my repo (https://github.com/SwoopX/deconz-rest-plugin). Would be great to collect like 20 mins of debug log, which includes the pairing of the device. You can so do with `/usr/bin/deCONZ --dbg-info=2 --dbg-error=2 > tvoc_debug`

rgers, 2021-06-15:
[tvoc_debug.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6658932/tvoc_debug.log)

Will this help? It looks to me that after pairing there's no communication... 
Let me know if you need anything else.

SwoopX, 2021-06-16:
Thanks!

Hm, there's indeed nothing. Have you tried forcing the values on the device to change (e.g. blow at it) during that time? If not, it would be good if you could take another short debug log where you try to tickle all sensors.

On the other hand, it wouldn't be too surprising if the device is "honest" zigbee and really follow the specs. I haven't whitelisted it for attribute reporting yet, since most Xiaomi devices ignore it anyway and to their own thing. Will amend the code later and let you know for a new try

redphx, 2021-06-16:
@rgers maybe you'll need to use the `Reset Node` feature in the `Edit` menu, then press (not hold) the Pair button on the TVOC. 
I haven't tried the new build yet, but with the stable build it only transfers data to deCONZ after I do those steps.

rgers, 2021-06-17:
@SwoopX here's a new log with your latest updates. This time I can see correct values for temperature, humidity and battery level in the log. Nothing shows up in Phoscon. I don't see air quality in the log.
[tvoc_debug321.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6671150/tvoc_debug321.log)

SwoopX, 2021-06-17:
Looks like we're almost there. Remarkable that it seems to adhere to the specs indeed.

For the air quality, can you try to do the manual binding with the `analog input` cluster as described here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to...#manually-bind-server-clusters and try to configure attribute reporting for 0x0055 with values `60,600,1`? When done, you should enter sensor search once again (without repair) to hopefully have the missing senor being created.

rgers, 2021-06-17:
Will let you know when I try manual binding, probably tomorrow.
Currently temp and humidity works correctly with OpenHAB, although I see it as 2 different devices (one with temp and other with hum) instead 1 device with 2 channels as usual... but maybe that's specific to this device?

SwoopX, 2021-06-20:
In the end, there's 3 sensors exposed via REST API. From what I've seen so far in the logs, all's as expected (apart from the missing air quality sensor). I got no clue what OH makes out of it but it's also nothing we can help you here with.

rgers, 2021-06-28:
I've done the manual binding and in Cluster Info in 0x0055 I can see the correct value changing but no sensor is being created. Maybe I've done the binding incorrectly...

SwoopX, 2021-06-28:
Do you recall if the values have updated even without the bindig? It's always kind of a challenge with Xiaomi to find out if it is required and accepted or not.

The sensor can show up if you enter sensor search, wake up the device by pushing the physical button once and then reading the voc attribute.

jan666, 2021-07-08:
Any news? Anything I can do to help? (I have this device, not yet paired - I'm on the latest stable deCONZ version)

kenge1967, 2021-07-17:
How is this going on, I just bought it so is there anything I can do to help out?

SwoopX, 2021-07-19:
I did some updates on the code for the air quality sensor. Can somebody please compile and test it?

Instructions: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
Repository: https://github.com/swoopx/deconz-rest-plugin.git
Branch: tvoc

balu-, 2021-07-27:
i would love to see the sensor working

Mimiix, 2021-07-27:
@balu- Well someone needs to test the changes @SwoopX made ;)

balu-, 2021-07-27:
> @balu- Well someone needs to test the changes @SwoopX made ;)

I orderd the Aqara device, so if i find a way to temporary enable the changes from @SwoopX without getting my Setup in an unrecoverable state i might help out on that

00lex, 2021-07-27:
OT: would it be possible to sent text to the screen?

FitziCJ, 2021-07-28:
Maybe because i am using Phoscon in an docker environment and have just ConBee II v2.12.01 running i have some issues someone other does not have (no listing in Phoscon webinterface, but Node-Red is working fine).
BUT: I compiled the tvoc fork and could connect the Aqara TVOC Air Quality Monitor (VOCKQJK11LM) with the phoscon app
I could read temperature und humidity, but not AirQuality:

{
    "13": {
        "config": {
            "battery": 100,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "3697e824bc9beeb8fb6dbea98e7e683f",
        "lastseen": "2021-07-28T21:04Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.airmonitor.acn01",
        "name": "Temperature 13",
        "state": {
            "lastupdated": "2021-07-28T21:04:50.632",
            "temperature": 2421
        },
        "swversion": "20210204",
        "type": "ZHATemperature",
        "uniqueid": "54:ef:44:10:00:10:5c:4a-01-0402"
    },
    "14": {
        "config": {
            "battery": 100,
            "offset": 0,
            "on": true,
            "reachable": true,
            "temperature": 0
        },
        "ep": 1,
        "etag": "b54258af6fe54309f2bd1ac4e4dfbc77",
        "lastseen": "2021-07-28T21:04Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.airmonitor.acn01",
        "name": "Humidity 14",
        "state": {
            "humidity": 6238,
            "lastupdated": "2021-07-28T21:03:20.819"
        },
        "swversion": "20210204",
        "type": "ZHAHumidity",
        "uniqueid": "54:ef:44:10:00:10:5c:4a-01-0405"
    },
    "15": {
        "config": {
            "battery": 100,
            "on": true,
            "reachable": true,
            "temperature": 0
        },
        "ep": 1,
        "etag": "89306a469b30c3c4630cd2ec4eb973a3",
        "lastseen": "2021-07-28T21:04Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.airmonitor.acn01",
        "name": "AirQuality 15",
        "state": {
            "airquality": null,
            "airqualityppb": null,
            "lastupdated": "none"
        },
        "swversion": "20210204",
        "type": "ZHAAirQuality",
        "uniqueid": "54:ef:44:10:00:10:5c:4a-01-000c"
    }
}

Does someone else have the same expirience like me regarding the AirQuality sensor?

SwoopX, 2021-07-28:
Hey, thanks for taking the effort to test it. So we're apparently one step further as the air quality sensor is now created. Now the question is why the values are not coming in...

You could wait over night to see if that changes or try setting up a manual binding as mentioned here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4704#issuecomment-863470991 . Meanwhile, I can check the code once again.

SwoopX, 2021-07-29:
@FitziCJ I made some changes on the air quality processing. Could you please download and compile again (no need to re-pair the device)?

jtscott, 2021-07-29:
I have updated to the tvoc branch but I can't get my device to even pair? Debug logs don't show even an attempt for it to join. Any pro tips? I have tapped, held for 8 seconds, pulled batteries, etc. but the thing won't connect?

SwoopX, 2021-07-29:
Do "permit join: 59" messages show up in the logs (hopefully not immediately followed by "permit join: 0")? How do you pair it? Does it show up in deconz GUI?

jtscott, 2021-07-29:
Here are some snippets I see regarding permit join logs; Via webui I did sensor > add new sensor. It appears that it flaps between 0 and 59 pretty quickly.
[tvoc_debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6903163/tvoc_debug.txt)

`	Line 4836: 01:58:11:182 send permit join, duration: 0

	Line 5119: 01:58:27:049 send permit join, duration: 59

	Line 5123: 01:58:27:055 send permit join, duration: 0

	Line 5780: 01:59:28:048 send permit join, duration: 59

	Line 5785: 01:59:28:054 send permit join, duration: 0

	Line 6781: 02:00:29:048 send permit join, duration: 59

	Line 6789: 02:00:29:062 send permit join, duration: 0

	Line 7250: 02:01:23:047 send permit join, duration: 0
`

SwoopX, 2021-07-29:
Could it be that you have 2 tabs opened for Phoscon? Just received note that they could interfere with each other.

jtscott, 2021-08-01:
I've completely knackered my install somehow, will have to let someone else keep testing.

rasssta, 2021-08-02:
I've compiled the tvoc branch and copied the plugin according to the manual. But I'm not able to add the device. Any hints?

FW: 26680700 /  2.12.03

rasssta, 2021-08-02:
Seem to have the same issue as scott.

```
14:19:31:085 send permit join, duration: 59
14:19:31:091 send permit join, duration: 0
14:19:54:085 send permit join, duration: 0
14:29:12:086 send permit join, duration: 59
14:29:12:093 send permit join, duration: 0
14:30:11:086 send permit join, duration: 59
14:30:11:089 send permit join, duration: 0
14:31:10:090 send permit join, duration: 59
14:31:10:101 send permit join, duration: 0

```

Mimiix, 2021-08-02:
As swoop mentioned: make sure you only have 1 tab open of Phoscon.

SwoopX, 2021-08-02:
Better also make sure you've opened Phoscon just once throughout all your devices (PC, mobile, tablet...).

rasssta, 2021-08-05:
Only 1 tab on 1 device. I've tried to reboot the rpi running deconz as well. Still getting the 59 -> 0 stuff :( 

```
09:51:46:856 Load config permitjoin: 0 from db.
09:51:46:856 Load config permitjoinfull: 0 from db.
09:51:46:955 Load config permitjoin: 0 from db.
09:51:46:955 Load config permitjoinfull: 0 from db.
09:54:43:210 send permit join, duration: 59
09:54:43:214 send permit join, duration: 0
09:55:42:211 send permit join, duration: 59
09:55:42:216 send permit join, duration: 0
09:56:41:211 send permit join, duration: 59
09:56:41:220 send permit join, duration: 0

```

rasssta, 2021-08-08:
Just noticed that HASS is using the websocket as well. Is this the cause of my issues?

```
11:01:08:307 New websocket 192.168.x.8:50415 (state: 3) 
11:01:15:349 New websocket 192.168.x.7:37262 (state: 3) 

```

rasssta, 2021-08-08:
I downgraded to deCONZ 2.11.05 and the "permit join" issue is gone == that parts works as intended. But the tvoc plug doesn't work in that version. Catch 22 here...

f04ly, 2021-08-16:
i'm new to conbee2 - is there a prediction when the tvoc devices will be in the firmware?

rasssta, 2021-08-16:
I did another try with 2.12.5-beta and managed to get it added right away. It shows battery level, temperature and humidity, but not air quality.

I get both reading failed and writing failed for attribute 0x0055.

Log from writing attempt ('grep 0x54' from the log):
```
13:23:02:757 [REQUEST] - 0x54EF4410001058DC (0x68E3) - Profile: 0x0104, Cluster: 0x000C, EP: 0x01, DstAddrMode: 02, Configure reporting (0x06), SeqNo.: 4, Mfc: 0x0000
13:23:53:395 Node data 0x54ef4410001058dc profileId: 0x0104, clusterId: 0x0402
13:23:53:396 0x54EF4410001058DC: update ZCL value 0x01/0x0402/0x0000 after 0 s
13:23:53:398 ZCL attribute report 0x54EF4410001058DC for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
13:23:53:399 [INDICATION] - 0x54EF4410001058DC (0x68E3) - Profile: 0x0104, Cluster: 0x0402, EP: 0x01, Unicast to: 0x0000, DstAddrMode: 02, Report attributes (0x0A), SeqNo.: 217, Mfc: 0x0000, Attr: 0x0000, Datatype: i16, Value: 2565 (0xA05)

```

rasssta, 2021-08-16:
Okay. So I re-pair'd it and was able to change the attributes. But 0x0055 still says 0.

Attaching log (grep 0x54) from pairing, attribute changing, etc.
[tvoc.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6992117/tvoc.txt)

SwoopX, 2021-08-16:
I wonder what makes you think that the few lines above have anything to do with any write attempt. I added some additional code to the PR which basically includes a sniffer for maximum transparency of the device communication so just a few random lines is not too helpful.

If the analog input cluster does not give anything back on either read or write, the device apparently uses something different to send TVOC data, which needs to be identified then.

FitziCJ, 2021-08-16:
I needed to reconnected my Aqara TVOC to deconz again and updated afterwards to my docker environment to v2.12.05, included the previous build TVOC-libde_rest_plugin.so and the result was, that the temperature, humdity AND the tvoc-values are shown in API:

{
    "config": {
        "battery": null,
        "on": true,
        "reachable": true,
        "temperature": 0
    },
    "ep": 1,
    "etag": "b751b2992cc3afa7aee7a8feeac88d1c",
    "lastseen": "2021-08-16T15:49Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.airmonitor.acn01",
    "name": "AirQuality 18",
    "state": {
        "airquality": "good",
        "airqualityppb": 209,
        "lastupdated": "2021-08-16T15:49:01.161"
    },
    "swversion": "20210204",
    "type": "ZHAAirQuality",
    "uniqueid": "54:ef:44:10:00:10:5c:4a-01-000c"
}

I don't why these values weren't visible before, maybe it is the update to v2.12.05

rasssta, 2021-08-16:
@FitziCJ Which commit are you referring to as "previous build"? :)

FitziCJ, 2021-08-16:
Indeed :) , i compiled the tvoc branch from SwoopX on my raspberry pi round about 19 days ago and saved the libde_rest_plugin.so before i updated today deconz to 2.12.05.

f04ly, 2021-08-18:
awaiting merge sounds great, thanks for your work!

f04ly, 2021-08-23:
is there a way to get the software before the merge? seems it takes a bit more than a few days :(

leonardpitzu, 2021-08-24:
`git clone --recursive https://github.com/dresden-elektronik/deconz-rest-plugin
cd deconz-rest-plugin/
git fetch origin pull/5213/head
git checkout FETCH_HEAD
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
sudo reboot`

f04ly, 2021-08-30:
compile and copy was fine. i can find the sensor now (search new sensor, green button with "found new sensor") but then there is no new sensor on deconz browser gui and i cant find anything what happend in messages/syslog

what is wrong?

Mimiix, 2021-08-30:
@f04ly That's normal. The Rest API shoudl show it though L)

FitziCJ, 2021-08-30:
Maybe some adaptions must be done to fix this. I still have this behavior in my docker environment too, but can see the reported values in the api. Do you see the three sensors in your api?

f04ly, 2021-08-30:
how can i check that?
in openhab i see 4 new sensors - but i am not absolutely sure, whether they are really new...
humidity, temperature, power, consumption (LUMI) 

is it possible that they are the 4 values from tvoc?

FitziCJ, 2021-08-30:
You should see temperature, humidity and Airquality. You could check your api on your phoscon gateway Unser help - > api. Choose Sensor and you should see at least three sensors.

f04ly, 2021-08-30:
ah ok yeah i see air quality - nice i didn't know that api feature. 
is there a reason why this sensor is not "forwarded" to openhab? oder deconz gui?
this is output of api for air quality

                    {
    "config": {
        "battery": 100,
        "on": true,
        "reachable": true,
        "temperature": 2613
    },
    "ep": 1,
    "etag": "70ddfc74e2db4a2008abc5375a880809",
    "lastannounced": "2021-08-30T08:49:03Z",
    "lastseen": "2021-08-30T10:54Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.airmonitor.acn01",
    "name": "AirQuality 23",
    "state": {
        "airquality": "good",
        "airqualityppb": 100,
        "lastupdated": "2021-08-30T10:39:18.900"
    },
    "swversion": "20210204",
    "type": "ZHAAirQuality",
    "uniqueid": "54:ef:44:10:00:10:61:9f-01-000c"
}

f04ly, 2021-08-30:
according to unique id i can see temperature and humidity in openhab als new things, but air quality is not shown so far

Mimiix, 2021-08-30:
@f04ly That's something that needs to be figured out on Openhab's site. @SwoopX seems good so far :)

[24 more replies]

closed 2021-11-10 by manup