dresden-elektronik/deconz-rest-plugin #4709: Unresponsiveness Zemismart M515EGBZTN

GreyEarl, 2021-04-05:
## Describe the bug

After one or two days tops the Zemismart M515EGBZTN does not respond anymore on input. The hardware buttons on the device itself do work. The only thing to make the device working again is:

1. Power off device
2. Restart DeCONZ service
3. Turn on the device

The unresponsiveness is first noticed when automation rules does not seem to be executed, while it is. The automation is done in combination with Home Assistant and NodeRED. To rule out these instances I executed a direct command with the DeCONZ API:

```
pi@home-automation:~ $ curl -H 'Content-Type: application/json' -X PUT -d '{"bri": 246, "lift": 97}' http://10.0.0.177/api/33******
35 /lights/12/state
[{"success":{"/lights/12/state/lift":97}}]pi@home-automation:~ $
```

The strange thing is I get an 'success message', but nothing happens. It wil only respond again when I follow the above procedure. I see delay events in the logs. See below.

## Steps to reproduce the behavior

So far I cannot reproduce the issue. It will occur after 1 or 2 days.

## Expected behavior

Execute the commands given.

## Screenshots
N/A

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian 
 - Firmware version: 26660700
 - deCONZ version: Self-compiled 2.10.03 with changes in merge #4616 
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

```
10:28:00:545 delay sending request 224 dt 2 ms to 0x5C0272FFFEECB8D7, ep: 0x01 cluster: 0xEF00 onAir: 1
10:28:00:551 delay sending request 224 ep: 0x01 cluster 0xEF00 to 0x5c0272fffeecb8d7 onAir 1
```

Full logs when issue is present and trying to submit various input:
[log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6259374/log.txt)

## Additional context
N/A

Smanar, 2021-04-05:
Hello

>The strange thing is I get an 'success message', but nothing happens. It wil only respond again when I follow the above procedure. I see delay events in the logs. See below.

This is normal, it s  long story but the api is updated before the command is send.

Do you have take a look on the mesh to see if the device is connected direclty to the conbee (so can test with an USB extension) or use a router, and wich one router ?

There is too a PR in waiting list that improve defaut response for tuya device, but as you are the only one for the moment with this issue, I m not sure I can change something.

GreyEarl, 2021-04-11:
Hi Smanar,

> 
> This is normal, it s long story but the api is updated before the command is send.
> 

Ah, that's good to know, thanks!

> Do you have take a look on the mesh to see if the device is connected direclty to the conbee (so can test with an USB extension) or use a router, and wich one router ?
> 

Yeah, it is directly connected to the Conbee. I will test with extension cord.

> There is too a PR in waiting list that improve defaut response for tuya device, but as you are the only one for the moment with this issue, I m not sure I can change something, and your logs doesn't look spammed by request.
> 

Ok, that sounds logical, thanks

GreyEarl, 2021-04-17:
After more thorough testing this week, it appears to be a range / connection issue, although it had a 'green connection line'. I disassembled the device from the wall and placed it very close to the Conbee. I haven't had a dropout all week. For these reasons, I am closing this case.

The initial placement of the Zemismart device was near a Zigbee router, but it never connect to it.
[image]

closed 2021-04-17 by GreyEarl