dresden-elektronik/deconz-rest-plugin #4713: Hue Motion Outdoor Sensor SML002 - how to update firmware?

Ltek, 2021-04-06:
## Describe the question or issue you are having

I've searched and cannot find the firmware download for March 11, 2019, Software version: 6.1.1.27575

I dont have a Hue Hub. Any help is appreciated!

## Screenshots
n/a

## Environment

 - Host system: PC
 - Running method: Homeseer
 - Firmware version: 26390500
 - deCONZ version: 2.10.04
 - Device: ConBee I
 - Do you use an USB extension cable:  no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
n/a

## Additional context
blah, stupid template

Mimiix, 2021-04-06:
Sadly, hue doesn't release their firmware files. That being said: we don't have them either.

ebaauw, 2021-04-06:
What's the _Image Type_ for the outdoor motion sensor?  There's a good change it uses the same firmware as the (indoor, regular) motion sensor, 0x010d.  The link to that in on the repository for the OTAU plugin, https://github.com/dresden-elektronik/deconz-ota-plugin.

closed 2021-05-05 by github-actions[bot]