dresden-elektronik/deconz-rest-plugin #4714: Hue Motion Outdoor Sensor SML002 -- add sensitivity setting control

Ltek, 2021-04-06:
## Feature request type
A used figured out a way to manually do this. Hoping you can add the feature for those of us that are not as technical...

So I have moved to Deconz recently and whilst it’s been working great I have been suffering with too many false positives because the sensitivity of the Hue motion sensors is too high.

So what I did to remedy this was:

Discover the API key which Home Assistant is using to access Deconz. This can be found in the config/.storage/core.config_entries file. Search for "deconz" and then you’ll find the "api_key". You could of course create a new api key in Deconz but for me I only needed to make this change once.

Download a REST tool for Chrome - I am using YARC - Yet Another Rest Client

Open YARC and in the URL field type http://<deconz-ip-address>:<deconz-http-port-number>/api/<api_key>/sensors and make a GET request.

You’ll then be presented with a list of sensors. Search the JSON output for “sensitivity”. Mine were all set to 2. Make note of the id number in the example below it is sensor id 5.

Copy to clipboard
  "5": {
    "config": {
      "alert": "none",
      "battery": 100,
      "delay": 0,
      "ledindication": false,
      "on": true,
      "pending": [],
      "reachable": true,
      "sensitivity": 2,
      "sensitivitymax": 2,
      "usertest": false
    },
    "ep": 2,
    "etag": "<removed>",
    "manufacturername": "Philips",
    "modelid": "SML001",
    "name": "Landing Motion Sensor",
    "state": {
      "lastupdated": "2019-09-17T08:45:54",
      "presence": false
    },
    "swversion": "6.1.1.27575",
    "type": "ZHAPresence",
    "uniqueid": "<removed>"
  },
Change the URL to http://<deconz-ip-address>:<deconz-http-port-number>/api/<api_key>/sensors/<sensor_id>/config , add the JSON payload and choose a PUT request.
Copy to clipboard
{
       "sensitivity": 1,
}
Verify the change has been made by making a GET request to http://<deconz-ip-address>:<deconz-http-port-number>/api/<api_key>/sensors/<sensor_id>
Copy to clipboard
{
  "config": {
    "alert": "none",
    "battery": 100,
    "delay": 0,
    "ledindication": false,
    "on": true,
    "pending": [],
    "reachable": true,
    "sensitivity": 1,
    "sensitivitymax": 2,
    "usertest": false
  },
  "ep": 2,
  "etag": "<removed>",
  "manufacturername": "Philips",
  "modelid": "SML001",
  "name": "Landing Motion Sensor",
  "state": {
    "lastupdated": "2019-09-17T08:55:52",
    "presence": false
  },
  "swversion": "6.1.1.27575",
  "type": "ZHAPresence",
  "uniqueid": "<removed>"
}

## Description
Please expose this setting in the UI so we can easily change it

## Considered alternatives
none possible

## Additional context
n/a

Kane610, 2021-04-06:
You can use the configure service in HASS to issue rest commands to deCONZ. 
https://www.home-assistant.io/integrations/deconz/#service-deconzconfigure

In your case your input would be something like
`{ "entity": "binary_sensor.landing_motion_sensor", "data": { "config": { "sensitivity": 1 } } }`
or
`{ "field": "/sensors/5", "data": { "config": { "sensitivity": 1 } } }`

Mimiix, 2021-04-06:
Hi, thank you for the feature request. However, this isn't something for deconz.

As @Kane610 suggested, that's how you should do it for ha. 

If you feel phoscon needs this : make a feature request here https://github.com/dresden-elektronik/phoscon-app-beta

Closing.

closed 2021-04-06 by Mimiix

Ltek, 2021-04-06:
> You can use the configure service in HASS to issue rest commands to deCONZ.
> https://www.home-assistant.io/integrations/deconz/#service-deconzconfigure
>
> In your case your input would be something [...]

HASS is not relevant - I provided the info an an example to show the ability is already in Deconz and just needs to be exposed in the UI

Ltek, 2021-04-06:
> Hi, thank you for the feature request. However, this isn't something for deconz.
> 
> As @Kane610 suggested, that's how you should do it for ha.
> 
> If you feel phoscon needs this : make a feature request here https://github.com/dresden-elektronik/phoscon-app-beta
> 
> Closing.

Why would this not be for Deconz? 
I provided the info an an example to show the ability is already in the Deconz API and just needs to be exposed in the UI. Why would ypu not want people to be able to change setting of the devices in the Deconz UI, yet you provide that functionality in the API?

Mimiix, 2021-04-06:
@Ltek It does work in deCONZ. A user (@micke, thanks!) just tried it for me:

occupancy Sensor cluster 
Sensitivity Attribute 0x0030:
[image]

Edited:

[image]

And the rest api reflecting the change:
[image]

So it is in deCONZ.