dresden-elektronik/deconz-rest-plugin #4715: Zigbee devices often randomly unavailable

lars-dev, 2021-04-06:
## Describe the bug

Since roughly the last update of the deconz addon to version 6.8.0 my zigbee devices become unavailable every so often. Mostly it seems to be the Aqara Door and window Sensor, and Ikea lights. The Hue stuff wasn't making problems.

It seems like some things just dont get triggered correctly
Sometimes I will trigger a light directly in HomeAssitant and it will toggle. sometimes it will show that its unavailable.
Sometimes the opening of the Aqara door sensor will toogle my light instantly. sometimes it wont. Sometimes it will do so but about up to 20 seconds later.

Similar with the motion sensors that trigger different lights.

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26390500)
 - deCONZ version: (2.10.4)
 - Device: RaspBee I 
 - Do you use an USB extension cable: No
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

22:47:04:638 0x00158D000465797A extract Xiaomi special attribute 0xFF01
22:47:04:639 	01 battery 2995 (0x0BB3)
22:47:04:639 	04 unknown 17320 (0x43A8)
22:47:04:640 	05 RSSI dB (?) 5 (0x0005)
22:47:04:640 	06 LQI (?) 4294967296 (0x0100000000)
22:47:04:641 	64 temperature 1795
22:47:04:642 	65 humidity 5689
22:47:04:642 	66 pressure 95233 (952)
22:47:04:643 	0a Parent NWK 36669 (0x8F3D)
22:47:04:867 Skip idle timer callback, too early: elapsed 815 msec
22:47:05:867 send permit join, duration: 0
22:47:21:436 dev /dev/ttyAMA0
22:47:21:437 GW firmware version: 0x26390500
22:47:21:438 GW firmware version is up to date: 0x26390500
22:47:31:367 Skip idle timer callback, too early: elapsed 499 msec
22:47:38:481 Skip idle timer callback, too early: elapsed 949 msec
22:47:40:381 Skip idle timer callback, too early: elapsed 949 msec
22:47:45:367 Skip idle timer callback, too early: elapsed 514 msec
22:47:49:507 [INFO] - No button handler for: TRADFRI on/off switch, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0001, command: 0x0A, payload: 2100204A, zclSeq: 89

and some more logs from today:

09:03:20:198 APS-DATA.confirm id: 48, status: 0x00 SUCCESS
09:03:20:200 APS-DATA.confirm request id: 48 -> erase from queue
09:03:20:213 aps request id: 48 finished, erase from queue
09:03:20:288 APS-DATA.indication srcAddr: 0x1d5b, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 119, rssi: -76
09:03:20:291 ZCL got data for node=0x1D5B, cl=0x0006, at=0x0000, status=0x00, type=0x10
09:03:46:665 APS-DATA.request id: 53, addrmode: 0x03, addr: 0x0017880108c73e85, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x0B queue: 0 len: 5 tx.options 0x00
09:03:46:775 dev /dev/ttyAMA0
09:03:46:779 GW firmware version: 0x26390500
09:03:46:783 GW firmware version is up to date: 0x26390500
09:03:46:827 APS-DATA.confirm id: 53, status: 0x00 SUCCESS
09:03:46:829 APS-DATA.confirm request id: 53 -> erase from queue
09:03:46:855 APS-DATA.request id: 55, addrmode: 0x03, addr: 0x0017880108c73e85, profile: 0x0104, cluster: 0x0300, ep: 0x01 -> 0x0B queue: 1 len: 17 tx.options 0x00
09:03:46:902 Mgmt_Lqi_req zdpSeq: 90 to 0x0017880108EC5626 start index 0
09:03:46:905 APS-DATA.request id: 56, addrmode: 0x03, addr: 0x0017880108ec5626, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 2 len: 2 tx.options 0x00
09:03:46:908 aps request id: 53 finished, erase from queue
09:03:46:911 APS-DATA.indication srcAddr: 0x1d5b, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 151, rssi: -72
09:03:46:913 ZCL got data for node=0x1D5B, cl=0x0008, at=0x0000, status=0x00, type=0x20
09:03:46:981 APS-DATA.confirm id: 55, status: 0x00 SUCCESS
09:03:46:984 APS-DATA.confirm request id: 55 -> erase from queue
09:03:47:028 APS-DATA.confirm id: 56, status: 0x00 SUCCESS
[... 170 lines cut ...]
09:08:09:323 APS-DATA.indication srcAddr: 0x714a, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 207, rssi: -65
09:08:09:327 ZCL attribute report 0x086BD7FFFE207A06 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
09:08:09:330 APS-DATA.request id: 26, addrmode: 0x02, addr: 0x714a, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
09:08:09:422 APS-DATA.indication srcAddr: 0x137d, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 191, rssi: -67
09:08:09:427 ZCL attribute report 0x086BD7FFFE328CB6 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
09:08:09:430 APS-DATA.request id: 29, addrmode: 0x02, addr: 0x137d, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
09:08:09:483 APS-DATA.confirm id: 26, status: 0x00 SUCCESS
09:08:09:486 APS-DATA.confirm request id: 26 -> erase from queue
09:08:09:534 APS-DATA.indication srcAddr: 0xa891, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 71, rssi: -82
09:08:09:536 APS-DATA.request id: 26 erase from queue
09:08:09:538 ZCL attribute report 0x000D6FFFFEFD89A8 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 

  Debug logs can help to investigate certain problems like device pairing and communication issues.
  Logging can be enabled on the command line as described in:
  https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-flags
  Secondly, for versions above 2.05.87: In the deCONZ GUI: click Help > debug view. Output of APS,ZDP, ZCL and INFO can be handy. 

## Additional context
30 nodes
4 Aqara multi sensor
1 Aqara door/window
9 Hue Lights
8 IKEA Lights
4 Ikea Switches
2 Ikea motion sensor

Mimiix, 2021-04-06:
Hi!

Can you check what device belongs to macadress 0x0017880108c75992

it seems that it stops routing for some reason (0xE9).

lars-dev, 2021-04-06:
0x0017880108c75992 seems to be a hue spot. 
[image]

I didn't notice any issues with that light specifically. But I also can't say that there where none.

Mimiix, 2021-04-06:
> 0x0017880108c75992 seems to be a hue spot.
> [image]
> 
> I didn't notice any issues with that light specifically. But I also can't say that there where none.

Can you make a screenshot of the deconz map?

lars-dev, 2021-04-06:
[image]

Mimiix, 2021-04-06:
That seems to be a fine network. 

Some more questions:
- What Case are you using for your RPI?
- What PowerSupply do you have? how much Amps does it provide?
- Are all devices not working per home-assistant or phoscon as wel?(as this issue describes: https://github.com/home-assistant/core/issues/37372) ?

lars-dev, 2021-04-06:
Standard Plastic Case.
[image]

[image]

i just testet triggering some lights directly in phoscon. They reacted instantly. I cant say if they would have been slower when triggered via a motion sensor or not.

Smanar, 2021-04-06:
And you have them sicne your software update ?
No firmware updated, no SSD added ?

lars-dev, 2021-04-06:
I had some promblems every once in a while but nothing major as this.
I haven’t changed anything on the hardware side in about a year or longer.
It all runs on an SD Card bit i havent changed that since last year either.

i have updated the Hass OS, the supervisor, the core and the deconz addon. I always do that within roughly a week after the update goes live.

msex67, 2021-04-07:
I am experiencing the same issues, mainly with ikea and xiaomi devices. Devices can be loost for weeks, and suddenly start working again.  Sometimes a reboot of the device helps.

Harry-1976, 2021-04-10:
Having same issue. 

When in a HA-group often 1 lights refuse to react and drops connection. It reconnects shortly thereafter.

gilbert-grape, 2021-04-11:
Same problem.
- using aqara door and smoke detecors, tradfri plug and lights, mueller licht and hue lights.
- after a restart of home assistant, connection randonly get lost. Sometimes nothing, sometimes a bulb
- in a group of bulbs with same power, sometimes just one light disconnects. Next time another one.
- have the problem with all manufacturs
- when it happens, in home assistant it is unavailable, same in phoscon and deconz
- the device never comes back, need a re-pairing or power off/on for lights

It is very annoying.

morfei1, 2021-04-11:
I've experienced similar problem with a Sonoff Basiczbr3 switch. It's working OK, and at some random time it just refuse to turn on the fan. No respond. Going to deCONZ and trying to read basic descriptor of it at the time it does not respond, does nothing. The red light on the right side of the node it blinking and all the lines connected to it disappear. Then after a while it reconnects.

I have this problem since the last summer of 2020. When I installed it. I have the problem with all of the deCONZ versions since then.

I found out the when it happens (it's completely random) a reboot of deCONZ sloves the problem.

I started to think that it's related to deCONZ not being reboot for about a month since I'm running only stable releases.

I'm not running HA-addon but my setup is standalone rpi3+ with raspbee I and deCONZ on SDcard. No USB3 devices near it.

As I mentioned, a reboot of deCONZ always slove this little problem.

gilbert-grape, 2021-04-11:
Then it is a different problem, i think.
I am running it in home assistant with addon. A reboot of ha (i guess this is like your reboot) does not solve my disconnected devices.

peltsippi, 2021-04-14:
I am just setting up deconz with Ikea lights and probably have this same issue too.

Remote control works just fine with the light, but light is unavailable in Phoscon and homeassistant. Turning mains off and on again did not help. Remote battery level is available in Phoscon & ha also. So somehow remote is connected just fine but light is missing. Did also reboot and waited 30 minutes and still unavailable.

KennethLavrsen, 2021-04-16:
@Mimiix why did you turn a bug report into a user question? It is not the first time you do that and it has upset people before. Is Dresden Elektronik trying to hide the obvious stability problems Deconz has gotten.

I also see devices randomly going unavailable. Not battery devices but also mains powered devices. And I am a professionel RF engineer and have done everything to put my Deconz device on extension cable away from noise sources. And all the “me too” replies in this report? It does not trigger Dresden Elektronik developers to think “maybe we do have a problem”?

Mimiix, 2021-04-16:
@KennethLavrsen 
> @Mimiix why did you turn a bug report into a user question?

Because i believe it's a individual issue. We tend to get a lot of people joining this issue and saying "metoo" while it's not the same "bug" they are after. Until we can rule all other causes (interference, firmware bugs on devices) or  user error it's a user question/request. Having 1 issue with people saying "unavailable" but different sympthoms (Like you just mentioned: You have also mains powered device issues, not sensor issues) aren't the same bug. Having multiple issues is easier to find out what the common factor is. I often send a set of issues to manup and point out the common factor there and that makes it way easier to figure out what goes on. If we have 1 issue with small comments saying "devices don't work" that really makes it hard to figure out.  On this particular case: There's over 100k installations of deCONZ. There's only a handful of people having issues. I don't say there couldn't be a bug, but i really want to rule out user error/environment error before going on the bug road. Having seperate issues just helps us see the bigger picture. 

> It is not the first time you do that and it has upset people before. 

Correct. Not sure why people would be upset about it as It's just a label and having it as a user question or bug report doesn't matter really. Responses would just be the same and handling would be the same.  For devs it's easier to see what to attend to as it filters for them. It's a matter of resources here really. 

> Is Dresden Elektronik trying to hide the obvious stability problems Deconz has gotten.

Why would they do that? I am not acting on their behalf and to be fair: Manup never asked to handle stuff a "certain" way. I really  just want issues to be streamlined.  I am acting on what i believe is better. 

Given this insight, i really hope that you understand that i have the best interest here but to achieve things fixed i like to have things seperated. As i can't really "unmerge" github issues when things "suddenly" are fixed. Great example is the SHP13 issue: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3788 . Some people still experience issues, while others have it solved. 

That all being said: Github isn't that great for having end users in here. Github is meant to submit a issue and already point what needs to be fixed. That's why we are setting up a forum so we can help users there and once we figured out what the bug/problem is, we can submit a github issue to get it fixed.

Feel free to reach out to me if you have any further questions :) Happy to talk with you and see what i can improve to make the user experience better.

peltsippi, 2021-04-16:
> There's only a handful of people having issues

How can you verify this? Requiring logs and similar and everything reported as a separate issue raises the bar for reporting significantly. It is just easier to figure out a way to live with the issues. 

Unfortunately my couple first days with deconz + ha have not been that nice. Instant reliability related issues when all that was changed is switching ikea gateway that stopped all communication to ConBee II.

Smanar, 2021-04-16:
How many percent users are for you, able to run deconz and use a menu to check log or able to access log using VNC or SSH, or able to find procedure on wiki or deconz, then multiply this value by the total user than use deconz, you realy can imagine it s possible to don't see return and ignore them ?

I m agree with you, just the template for issue is an horror, but problems can have so much origins, and it s not possible without a minimum of informations.

KennethLavrsen, 2021-04-17:
@Mimiix I have been the main maintainer on the Motion project (camera) and release manager for TWiki, and later Foswiki. I have dealt with 1000s of bug reports in those roles. And one thing I learned (by mistakes and getting critique and eventually success) was to treat reports with respect while still being able to say "sorry, no". And yes, there were many bug we could not reproduce or would cost too much to fix while affecting only few.  That is how it has to be. We are all limited by time and hands.

But putting a "user question" on an obvious bug report is like labelling it "go to heck" or "sod off". That label is OK to have when people use the git hub for support.

Here is some practical advice.

When you get many bug reports that you cannot reproduce (happens rarely or requires the hardware available, etc) then create a common place holder bug report. E.g. make one called "Devices randomly unavailable" and explain in it that you are collecting bug reports but that you cannot at the moment find the root cause and cannot reproduce it. Each time a new report comes up with the same familiar pattern you mark it as a duplicate and link it to the master bug and close the new report. 
For the reporter the message is - someone has taken ownership and cared. And when they read the master report they will understand that it is a difficult bug to fix. And the advantage for you is that you get less open defects in the Github. All these open unresolved user questions with angry messages mud the picture and spread bad mood. And as the many duplicate bugs arrive - they may eventually - in unison - give the hint to what could be the root cause.  And if not then it will act like a Pandoras box. In addition you can edit the report with advice to how people can work around the issue.

That is the advice from an old open source release manager. I hope it is useful.

Mimiix, 2021-04-20:
@KennethLavrsen  I've replied in the designated issue where this belongs. Let's keep this issue on topic , shall we 🙂 ?

KennethLavrsen, 2021-05-12:
I have seen a Philips hue light go missing lately and the way to get it to connect again is to power cycle it. I see battery sensors unavailable often but they join again when you trigger them or just wait long enough. That does not concern me. But why does mains powered devices go offline? There have been certain Osram plugs and Ikea early bulbs with firmware bugs and they ended in the bin. But left are some random cases that can be any device in the house that become unavailable. I would say once every week or second week. And it started beginning of the year. Last year it never happened after I tossed the old Ikea bulbs and replaced all Osram plugs by Hue smart plugs. There is something that can bring one individual device in limbo. I have around 100 devices and plenty of coverage in the mesh. Naturally the more stuff you have the more probability that you see device failures. But why will a Philips device suddenly be unreachable until power cycle. When it happens sometimes the device still reacts to group messages. But ignore individual on/off and appears unavailable in Phoscon. It is strange

tgalizia, 2021-05-12:
> I have seen a Philips hue light go missing lately and the way to get it to connect again is to power cycle it. I see battery sensors unavailable often but they join again when you trigger them or [...]

I have exactly the same behavior as you describe. Some bulbs disappear until paired again, some need a power cycle, some battery-powered switches became unresponsive, randomly. No luck with Phoscon because the device is unavailable or doesn't respond to commands until it decides to start working again. Frustrating.
Also, the problem of the ghost devices, I have about 10 devices that have been removed but still appear in Deconz and if you delete them they appear again.

peltsippi, 2021-05-12:
> Some bulbs disappear until paired again, some need a power cycle, some battery-powered switches became unresponsive, randomly. No luck with Phoscon because the device is unavailable or doesn't respond to commands until it decides to start working again. Frustrating.

Having similar issues also. Restarting deconz pc helps with some issues but not with all.

KennethLavrsen, 2021-06-03:
Go away stale bot

MacAce1, 2021-06-08:
Only my 2 Hue LTW010 1.50.2_30933 (White Ambience bulbs next to each other) ever have this problem of being unresponsive very occasionally, most often only one, sometimes both at the same time, instantly fixed via power cycle. Never any other surrounding LWB010s (White & Color) or LST002s (Lightstrip).

closed 2021-07-08 by github-actions[bot]

nin9s, 2022-01-03:
is that still an issue with BASICZBR3 ?