dresden-elektronik/deconz-rest-plugin #4718: ConBee II: GW update firmware not found

mjaensch, 2021-04-06:
## Describe the question or issue you are having

I bought a brand new ConBee II, attached it to my home server (Ubuntu Server 18.04) and started adding my Zigbee devices and naming them in Phoscon. However, a few Aqara devices didn't show up in the Phoscon App and I wasn't able to get an overview of my zigbee network, because I couldn't use deCONZ application without xserver. After trying a few more things like using a 2m USB cable, I ended up switching over to my Windows Client.

In deCONZ I was now able to see my network (see screenshot below). I was still not able to add the last sensors so I spoke to a friend of mine who recommended ConBee II. He was surprised that the zigbee devices didn't show up with names / labels in deCONZ, but as hex values only. So he suggested to do a firmware update. I've done several firmware updates with different firmwares, but I still have the following problems:

1. Log entry "*GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF*" every 10 seconds in deCONZ debug window
2. *Firmware: Not connected* in Phoscon (never saw somthing other there)
3. Two Aqara devices are not shown in deCONZ / I am unable to connect them

## Screenshots

[image]
[image]

## Environment

**first environment**
 - Host system: PC
 - Running method: Ubuntu Server 18.04 LTS
 - Firmware version: Not connected
 - deCONZ version: - (no xserver)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

**second environment**
 - Host system: PC
 - Running method: Windows
 - Firmware version: Not connected
 - deCONZ version: v2.10.3
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? yes, mouse + keyboard, no serial devices

## deCONZ Logs

```
20:24:36:074 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:24:46:081 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:24:56:079 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:25:06:078 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:25:16:081 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:25:16:205 new node - ext: 0x00158d0003a23087, nwk: 0x72DB
20:25:20:050 Current channel 15
20:25:20:065 CTRL got nwk update id 2
20:25:20:081 Device TTL 3542 s flags: 0x7
20:25:26:080 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:25:26:227 saved node state in 0 ms
20:25:36:078 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:25:46:082 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:25:56:085 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:26:06:088 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:26:16:082 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:26:20:069 Current channel 15
20:26:20:085 CTRL got nwk update id 2
20:26:20:088 Device TTL 3482 s flags: 0x7
20:26:26:084 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:26:36:089 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:26:46:101 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:26:56:095 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:27:06:105 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:27:16:112 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:27:20:070 Current channel 15
20:27:20:086 CTRL got nwk update id 2
20:27:20:103 Device TTL 3422 s flags: 0x7
20:27:26:112 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:27:36:126 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:27:46:126 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:27:56:136 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:28:06:148 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:28:16:150 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:28:20:068 Current channel 15
20:28:20:084 CTRL got nwk update id 2
20:28:20:100 Device TTL 3362 s flags: 0x7
20:28:26:163 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:28:36:164 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
20:28:46:162 GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF
```

## firmware flash log
```
GCFFlasher -d COM3 -t 60 -f ../firmware/deCONZ_ConBeeII_0x26680700.bin.GCF
GCFFlasher V3_14 (c) dresden elektronik ingenieurtechnik gmbh
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
flashing 164373 bytes: |==============================|
verify: .
SUCCESS
Wait 10 seconds until application starts
```

Mimiix, 2021-04-06:
Can you add some more logs?

mjaensch, 2021-04-06:
sure, how can i access the raw log files?
I've started deCONZ and selected all debug levels in debug view and then connected the ConBee II.
However doing so, it took a while until the network appeared in deCONZ.
Here is the log:
```
20:57:29:114 Announced to internet https://phoscon.de/discover
20:57:29:925 enqueue event config/localtime for /config/
20:57:29:972 COM: \\.\COM1 : Kommunikationsanschluss (0x0000/0x0000)
20:57:30:930 enqueue event config/localtime for /config/
20:57:30:978 COM: \\.\COM1 : Kommunikationsanschluss (0x0000/0x0000)
20:57:31:941 enqueue event config/localtime for /config/
20:57:31:973 COM: \\.\COM1 : Kommunikationsanschluss (0x0000/0x0000)
20:57:32:920 Daylight now: dusk, status: 210, daylight: 0, dark: 1
20:57:32:935 enqueue event config/localtime for /config/
20:57:32:966 COM: \\.\COM1 : Kommunikationsanschluss (0x0000/0x0000)
20:57:33:944 enqueue event config/localtime for /config/
20:57:33:976 COM: \\.\COM1 : Kommunikationsanschluss (0x0000/0x0000)
20:57:34:936 enqueue event config/localtime for /config/
20:57:34:983 COM: \\.\COM1 : Kommunikationsanschluss (0x0000/0x0000)
20:57:35:946 enqueue event config/localtime for /config/
20:57:35:995 COM: \\.\COM1 : Kommunikationsanschluss (0x0000/0x0000)
20:57:36:934 enqueue event config/localtime for /config/
20:57:37:004 COM: \\.\COM1 : Kommunikationsanschluss (0x0000/0x0000)
20:57:37:906 COM: \\.\COM1 : Kommunikationsanschluss (0x0000/0x0000)
20:57:37:937 enqueue event config/localtime for /config/
20:57:38:001 COM: \\.\COM1 : Kommunikationsanschluss (0x0000/0x0000)
20:57:38:945 enqueue event config/localtime for /config/
20:57:39:008 COM: \\.\COM1 : Kommunikationsanschluss (0x0000/0x0000)
20:57:39:936 enqueue event config/localtime for /config/
20:57:39:997 COM: \\.\COM1 : Kommunikationsanschluss (0x0000/0x0000)
20:57:40:938 enqueue event config/localtime for /config/
20:57:41:001 COM: \\.\COM1 : Kommunikationsanschluss (0x0000/0x0000)
20:57:41:001 COM: \\.\COM3 : Serielles USB-Ger�t (0x1CF1/0x0030)
20:57:41:005 auto connect com \\.\COM3
20:57:41:017 COM check bootloader
20:57:41:032 
R21B18 Bootloader
20:57:41:032 Vers: 2.07
20:57:41:032 build: Jun 17 2019
20:57:41:032 , 08:48:53
20:57:41:033 COM: \\.\COM1 : Kommunikationsanschluss (0x0000/0x0000)
20:57:41:033 COM: \\.\COM3 : Serielles USB-Ger�t (0x1CF1/0x0030)
20:57:41:932 enqueue event config/localtime for /config/
20:57:42:921 Daylight now: dusk, status: 210, daylight: 0, dark: 1
20:57:42:936 enqueue event config/localtime for /config/
20:57:43:029 COM check bootloader
20:57:43:044 
R21B18 Bootloader
20:57:43:044 Vers: 2.07
20:57:43:044 build: Jun 17 2019
20:57:43:044 , 08:48:53
20:57:43:933 enqueue event config/localtime for /config/
20:57:43:934 networkState: CC_ReconnectNetwork
20:57:43:934 start reconnect to network
20:57:44:913 NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=100
LOCATION: http://192.168.2.37:80/description.xml
SERVER: Linux/3.14.0 UPnP/1.0 IpBridge/1.26.0
GWID.phoscon.de: 00212EFFFF06C0BF
hue-bridgeid: 00212EFFFF06C0BF
NTS: ssdp:alive
NT: upnp:rootdevice
USN: uuid:{c9174fd2-733d-4998-adcc-85e4ddb5fa3a}::upnp:rootdevice

20:57:44:914 NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=100
LOCATION: http://192.168.2.37:80/description.xml
SERVER: Linux/3.14.0 UPnP/1.0 IpBridge/1.26.0
[... 1243 lines cut ...]
21:00:11:431 Node 0xD85DEF11A10042C6 is known by 4 neighbors, last seen 4 s
21:00:11:980 Node 0x14B457FFFEFAF0CE is known by 4 neighbors, last seen 1 s
21:00:12:042 enqueue event config/localtime for /config/
21:00:12:544 Node 0xD85DEF11A10042CC is known by 4 neighbors, last seen 38 s
21:00:12:973 Daylight now: dusk, status: 210, daylight: 0, dark: 1
21:00:13:052 enqueue event config/localtime for /config/
21:00:13:114 Node 0x00158D0001E802AB is known by 0 neighbors, last seen -1 s
21:00:13:393 Mgmt_Lqi_req zdpSeq: 81 to 0x14B457FFFEFAF0CE start index 3
21:00:13:393 APS-DATA.request id: 123, addrmode: 0x03, addr: 0x14b457fffefaf0ce, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
21:00:13:441 APS-DATA.confirm id: 123, status: 0x00 SUCCESS
21:00:13:441 APS-DATA.confirm request id: 123 -> confirmed, timeout 1617735613
21:00:13:472 APS-DATA.indication srcAddr: 0xae7a, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -43
21:00:13:472 	asdu: 5100050302bfc006ffff2e21007ce0affeffccccccfd4825020f61bfc006ffff2e2100cc4200a111ef5dd847bd25020fe1
21:00:13:472 APS-DATA.indication request id: 123 -> finished
21:00:13:473 APS-DATA.request id: 123 erase from queue
21:00:13:473 ZDP status = 0x00 -> SUCCESS
21:00:13:473 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0xAE7A
21:00:13:473 ZDP Mgmt_Lqi_rsp zdpSeq: 81 from 0x14B457FFFEFAF0CE total: 5, startIndex: 3, listCount: 2
21:00:13:473     * neighbor: 0xCCCCCCFFFEAFE07C (0x48FD), LQI: 97, relation: 0x02 rxOnWHenIdle: 1
21:00:13:474     * neighbor: 0xD85DEF11A10042CC (0xBD47), LQI: 225, relation: 0x02 rxOnWHenIdle: 1
21:00:13:474 MS CORE_APS_IND (processPacked:980) took 2 ms
21:00:13:672 Node 0x00158D0002735DEA is known by 1 neighbors, last seen 11 s
21:00:14:043 enqueue event config/localtime for /config/
21:00:14:231 Node 0x00158D000271FE56 is known by 1 neighbors, last seen 7 s
```

Smanar, 2021-04-06:
If you have include your device on a machine, and making test on another it s normal you haven't name, you need to migrate too the database (using the phoscon backup/retore feature)
Setting are stored on the device, but devices data on the host.

When you have made the try on windows (a real machine ?) have you listen the usb disconnect sound ?

```
20:57:47:919 COM: \\.\COM1 : Kommunikationsanschluss (0x0000/0x0000)
20:57:47:920 COM: \\.\COM3 : Serielles USB-Ger�t (0x1CF1/0x0030)
```

It s the conbee on com3 ????

mjaensch, 2021-04-06:
I still have the same device / sensor names in Phoscon Web GUI and I definitely didn't use any backup/restore feature. I initially set up all devices names and settings using the Phoscon on my Ubuntu Server and now I have the same names / settings in Phoscon on my Windows Client.

Yes, my Windows PC is a real, normal Desktop PC. When I connect / disconnect the device I can hear the corresponding sound.
ConBee II is on COM3.
That's the GCFFlasher output:
```
GCFFlasher -l
GCFFlasher V3_14 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
\\.\COM3         | 0x1CF1 | 0x0030  | DE2411714  | ConBee II
```

Smanar, 2021-04-07:
>I still have the same device / sensor names in Phoscon Web GUI and I definitely didn't use any backup/restore feature. I initially set up all devices names and settings using the Phoscon on my Ubuntu Server and now I have the same names / settings in Phoscon on my Windows Client.

 Ubuntu Server / Windows Client, so you have only 1 deconz instance on the Ubuntu machine ? I was talking about when you did the tests on windows for exemple.
Are you sure you haven't 2 deconz instance, the headless + the GUI one ?

```
sudo systemctl status deconz
sudo systemctl status deconz-gui
```

Only one of thoses command need to work.

mjaensch, 2021-04-08:
Initially I set up the ConBee II on my Home Server (running Ubuntu Server 18.04 LTS). On this server the service deconz is active and running. deconz-gui is inactive (no xserver on Ubuntu Server).
Because of my problems I moved over and connected ConBee II to my normal Windows Computer, because I can use the deconz Software with GUI here and it's easier to debug having the network view.
Having connected ConBee to my Windows Computer I totally ignored the Ubuntu Server. I now use the local Phoscon App / web server that is running on Windows.

However, as soon as I get my ConBee II running, I will connect it to the Ubuntu Server again, because that one is running 24/7.

My main problem is, that I have those periodically log entries "GW update firmware not found: deCONZ_ConBeeII_0x26660700.bin.GCF" and "Firmware: Not connected" in Phoscon.

Smanar, 2021-04-09:
>Having connected ConBee to my Windows Computer I totally ignored the Ubuntu Server. I now use the local Phoscon App / web server that is running on Windows.

Ok, so you are fine, not possible doing better

```
GCFFlasher -d COM3 -t 60 -f ../firmware/deCONZ_ConBeeII_0x26680700.bin.GCF
GCFFlasher V3_14 (c) dresden elektronik ingenieurtechnik gmbh
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
flashing 164373 bytes: |==============================|
verify: .
SUCCESS
Wait 10 seconds until application starts
```

Seem perfect too for me ....

And still "Firmware: Not connected" in Phoscon after 5mn ?  (the device can need 1 complete minute to be ready)

mjaensch, 2021-04-11:
Okay, I finally got it running.
I played around with the ConBee II and flashed it several times. I always had a successful flash process and in total I flashed the device at least 8 times with different firmware versions. However, after the last flash with the newest version, suddenly Phoscon asked  for a new password and showed the firmware version in its settings.
I have no clue how to reproduce this issue. I've flashed exactly the same firmware version at least three times and it somehow worked only after my last attempt.
I will close this issue now.
Thank you @Smanar for your help! 😃

closed 2021-04-11 by mjaensch

Smanar, 2021-04-12:
And no difference during the firmware update ? it take same time on every test ? no different messages ?