dresden-elektronik/deconz-rest-plugin #4723: Device name

NMGAndy, 2021-04-07:
## Device

- Product name: TUYA CURTAIN MOTOR
- Manufacturer: _TZE200_zpzndjez
- Model identifier: TS0601
- Device type :  Window Covering 

Absolutely same motor as: 
[https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4288](url)

The only difference is Manufacturer (_TZE200_zpzndjez).

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2021-04-07:
We need to have the clusters. They change stuff some times.

Please add all requested screenshots.

gmitch64, 2021-04-07:
I'm having a similar issue with some Tuya Tubular motors - all are sold as ZM25TQ, but their basic clusters are showing differently.

I have 2 of these ZM25TQ motors which show as a model TS0601
[image]

I also have 2 of these ZM25TQ motors which show as a model mcdj3aq
[image]

I don't know if this is the same issue, or whether it's a separate Tuya weirdness....

Graham

NMGAndy, 2021-04-08:
> We need to have the clusters. They change stuff some times.
> Please add all requested screenshots.

Added.

Mimiix, 2021-04-08:
@NMGAndy Can you add _all_ clusters?

Smanar, 2021-04-08:
_TZE200_zpzndjez is already in the code _TZE200_wmcdj3aq too.

Wich one deconz have you ?
Your devices are routers or End devices, missing on your captures.

NMGAndy, 2021-04-08:
> _TZE200_zpzndjez is already in the code _TZE200_wmcdj3aq too.
> Wich one deconz version have you ?

[image]
[image]
[image]

_TZE200_zpzndjez is added as outlet and doesn't work.

> Your devices are routers or End devices, missing on your captures.

Routers.

Smanar, 2021-04-08:
You need the version v2.10.5
The outlet is normal, we have some problem ATM to reconize tuya device, so we need 2 inclusions:

At first inclusion the device is reconized as outlet, and the code write data on the database.
On the second inclusion, the code use data to reconize it as covering.

gmitch64, 2021-04-09:
> You need the version v2.10.5

Any idea when that's going to be available?
Or what do I need to pull down the code and compile it locally?

Graham

gmitch64, 2021-04-09:
Here's the clusters for my ZM25TQ/_TYST11_wmcdj3aq after a double inclusion.

[image]

[image]

[image]

[image]

[image]

gmitch64, 2021-04-09:
Here's the clusters for my ZM25TQ/_TZE200_wmcdj3aq after a double inclusion.

[image]

[image]

[image]

[image]

[image]

[image]

NMGAndy, 2021-04-09:
> @NMGAndy Can you add _all_ clusters?

Added all clusters.

Mimiix, 2021-04-09:
> > You need the version v2.10.5
> 
> Any idea when that's going to be available?
> Or what do I need to pull down the code and compile it locally?
> 
> Graham

There isn't. Afaik it will be 2.11.0 as we have a stable on 2.10 already. 

Not sure when they release Wil be.

gmitch64, 2021-04-09:
Ah. I was just going off Smanar comment about needing v2.10.5 earlier in the thread.

The release cadence says that the next beta should be on the 15th.

Smanar, 2021-04-09:
Yep sorry ^^
I m using this page to know on wich one version the PR was merged https://github.com/dresden-elektronik/deconz-rest-plugin/releases
But it don't mean all versions are published.

gmitch64, 2021-04-09:
>  if you have a linux machine with a real OS

You mean Windows isn't a real OS? :)

I have a Pi, and I've done some deCONZ compilations there in the past, so that's not an issue - I just need to know which git to use to pull the correct code down...

Smanar, 2021-04-09:
lol
But on linux you can have light OS, totaly unusable.

You can use the official code, official branch, nothing, to change like https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

closed 2021-05-08 by github-actions[bot]