dresden-elektronik/deconz-rest-plugin #4728: Terncy Smart Dial SD01

migo-dev, 2021-04-07:
## Device

- Product name: Terncy Smart Dial SD01
- Manufacturer: 0x1228
- Model identifier: TERNCY-SD01
- Device type :
  - (Wireless) Switch

https://www.terncy.com/en-us/smartdial/

[image]

## Screenshots

### Node

[image]

### Basic

[image]

### Identify

[image]

### Power configuration

[image]

### Poll control

[image]

### OTAU

[image]

-----------

Thank you very much in advance - I'd really appreciate if you could implement this device!

Please let me know if I forgot something and/or you need more information.

PS: I saw that there is already support in ZHA and Zigbee2MQTT, if that helps with the implementation.

Mimiix, 2021-04-08:
Nice device!  Can you perhaps read the clusters prior screenshotting? Without the model id / manufacturer name it can't be added.

migo-dev, 2021-04-08:
@Mimiix it is! 😃 
Please excuse my inexperience/ignorance, I never really used this view to check all those details and assumed it got filled, after I pressed read.

I hope this helps:

[image]
(the manufacturer name) stays empty and greyed out, no matter how often I press read and interact with the device - so I attached the below screenshot of the OTAU cluster, which contains the manufacturer id

[image]

there are also some more information in this OTAU screenshot, compared to the one in my original post.

eggertm, 2021-05-15:
I have the same device, and it seems that the manufacturer name is always grayed out. Is there some way to read that again?

migo-dev, 2021-05-16:
@eggertm there is a 'read' button above the attributes, but I could not get it to be filled/not greyed out.

m1ke-hub, 2021-05-18:
@Mimiix I saw you mentioning in another device request that some devs might be interested in getting a device sent to implement it. Unfortunately, I can't remember the thread nor the names of those devs - so I'd really appreciate it if you could ask/link them, if this interest is still applicable.

BTW, do you know whether they return the devices afterwards (shipping covered ofc) or keep them as a 'fee'? 😄

Mimiix, 2021-05-18:
@SwoopX are you interested?

eggertm, 2021-05-18:
Actually, due to logistics issues (shipment took more than 3 months to deliver), I have effectively received a free device.  If a dev wants one to implement, I'll be happy to donate one and pay for shipping (from Iceland).

SwoopX, 2021-05-20:
Well, the most challenging part I see here is the missing manufacturer name. Would be interesting to see what the sniffer tells us.

So, if you're up to it, feel free to ping me on Discord. I'm happy to have a look at the device and figure something out.

m1ke-hub, 2021-06-11:
Bump

May I take this opportunity to ask whether there is something to report yet? 😃

SwoopX, 2021-06-11:
You mean except for lots of pain as it is horrible from a zigbee perspective? 🙃

I guess I'll have the code and the required instructions ready over the weekend.

SwoopX, 2021-06-11:
So far so good...

[image]

SwoopX, 2021-06-14:
Ok, I guess the work is finished. First, some technical observations:

- The device indeed has no attribute for manufacturer name (0x0004)
- It doesn't really support attribute reporting in the sense that any request for configuration gets a success, but is ignored in the end. Consequently, all reports are predefined by the device.
- On each button action, it either sends a cluster specific attribute report (0xfccc) or a manufacturer specific command (mfc: 1228, cluster 0xfccc)
- Almost all communication from the device to the coordinator goes to Endpoint 110, which requires some changes on the coordinator via deconz GUI to make it usable. This can currently lead to a conflict in case also Green Power devices are used (as the corresponding endpoint needs to be reconfigured)
- On a longer press, the device emits the command `0x14` with a payload incrementing each second (1 sec = 01, 2 sec = 02, etc.). However, after releasing the button, no additional command or report is sent which pretty much makes it impossible to leverage these.. Regardless, a long press for 1 sec will be available, but timing is key for the event to get generated.
- The numer of short presses is distinguishable from the 2nd byte of the corresponding command payload. The first byte is apparently a counter which hopefully resets to `00` after `FF` has been reached. In worst case, the payload length would increase to 3 resulting in a button event not being emitted anymore.

The device will have the following capabilities:

**Presses**
- Single (buttonevent `1002`), double(buttonevent `1004`), tripple (buttonevent `1005`) and quadruple (buttonevent `1006`) press
- long press for 1 second (timing is key, buttonevent `1003`)

**Rotation**
- One rotation tick clockwise will give a positive increase of state `angle` by 12, counter-clockwise a negative decrease of -12
- On a clockwise rotation, button event `1030` will be send, counter-clockwise `1031`
- For each rotation, the time in miliseconds is exposed as `eventduration`

closed 2021-06-30 by manup

m1ke-hub, 2021-08-20:
Hey @SwoopX, please excuse my late reply. Since I'm using the Home Assistant add-on, I was unable to test the beta. Yesterday, the update was released and I could finally test the device - after changing the network settings as instructed above, everything worked perfectly!! Thank you so much for your work and detailed instructions!

SwoopX, 2021-08-21:
Thanks for having sent it over. Enjoy!

m1ke-hub, 2021-08-21:
Can't take credit for that. It was @eggertm